openwrt/openwrt #16109: Firmware selector failed to build 23.05.4 image for Cudy WR3000 v1

V10lator, 2024-08-08:
### Describe the bug

This happens with the default selection (`ppp` and `ppp-mod-pppoe` removed to trigger a real build) :
```
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package busybox.
 * pkg_hash_check_unresolved: cannot find dependency cgi-io for luci-base
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	cgi-io
 * opkg_install_cmd: Cannot install package luci.
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package busybox.
 ```
 
 Not sure if this is related to #16089

### OpenWrt version

23.05.4 (r24012-d8dd03c46f)

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy WR3000 v1

### Image kind

Self-built image

### Steps to reproduce

Slightly change the default package selection, like removing `ppp` and `ppp-mod-pppoe`, then request to build it.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

V10lator, 2024-08-08:
> Is this from a clean repository?

[image]

V10lator, 2024-08-09:
Looks like it was just a temporary hiccup or something as it works today.

closed 2024-08-09 by V10lator

pparent76, 2024-08-12:
I have the same problem today, with freshly downloaded image builder  on target mt7621