openwrt/openwrt #16114: RT-AC3100 with BCM4366/3 has issues with the 5ghz band

seatracker, 2024-08-08:
### Describe the bug

The RT-AC3100 has 2 versions, one with BCM4366/3 and the other with BCM4366/4.  As a result, the image comes with the brcmfmac4366b and brcmfmac4366c firmwares.  Their version is 10.28.2.

I was trying the image on a device with BCM4366/3, but clients could not stay connected to the 5Ghz band.  They would connect, get a DHCP response and nothing else.  Usually just disconnect.

Found out the issue is caused by the brcmfmac4366b firmware.  It is not working correctly.  I switched to an older version (10.10.122.301) and clients were able to connect.

The firmware in the openwrt image is:
BCM4366/3 wl0: Nov 5 2018 03:11:08 version 10.28.2 (r769115) FWID 01-801fb449

Replaced firmware that works:
BCM4366/3 wl0: May 31 2017 20:43:28 version 10.10.122.301 (r658909)

I found this documented in another device that uses the same chipset:
https://openwrt.org/toh/d-link/dir-885l#hw_vera1_50_ghz_issues

I tried a few images (since it was first supported) and all have the same issue.  It might affect only BCM4366/3

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

bcm53xx/generic

### Device

ASUS RT-AC3100

### Image kind

Official downloaded image

### Steps to reproduce

Install OpenWRT, connect to the 5Ghz band and try to send any traffic.

No configuration is needed.  It does not work.

### Actual behaviour

Client connects to the 5Ghz band, is able to receive DHCP response, but nothing else works

### Expected behaviour

Traffic works with 5Ghz

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MasterKeyxda, 2024-08-16:
I sometimes have this problem. I extracted an old firmware out of official Asus firmware for bcm4366-c0 and the problem seems to disappear.

seatracker, 2024-08-19:
Interesting.  Thanks for the comment.  So, it might be possible that the firmware that is coming with the OpenWRT image (10.28.2) has some issues with both the bcm4366-b1 and bcm4366-c0.
I looked at the lastest Asus image and got the current firmwares.  I upgraded mine to 10.10.69.69042 and it is working fine.

bcm4366-b1:  Version: 10.10.69.69042 (r798414) CRC: 15b8a7d7 Date: Fri 2021-08-27 10:27:27 CST Ucode Ver: 1073.579 FWID: 01-f537b78e

bcm4366-c0: Version: 10.10.122.303 (r666429) CRC: 3ba207f7 Date: Fri 2021-08-27 18:38:19 CST Ucode Ver: 1128.13905 FWID: 01-ccd91e51

Attaching both files here, if anyone needs them.
[brcmfmac4366.zip](https://github.com/user-attachments/files/16665291/brcmfmac4366.zip)

MasterKeyxda, 2024-08-19:
Thank you for confirming it

tbrautaset, 2025-07-22:
> [@tbrautaset](https://github.com/tbrautaset)
> 
> Can you go through this page and try to get dhd. Ko then extract firmware https://www.perfacilis.com/blog/systeembeheer/linux/install-asus-pce-ac88-on-linux.html#extracting
> 
> Upload them using scp.

Hey @MasterKeyxda – just wanted to say **thanks a ton** for pointing me to that link 🙌
It was exactly what I needed.

After some digging around, I managed to extract both the `brcmfmac43602-pcie.bin` **and** the embedded `.txt` NVRAM config from the RT-AC3200 firmware. Took a bit of effort to locate the right offsets, but everything's working now.

Next up is the RT-AC5300 – which will probably be easier, since it's much closer to the AC3100 layout.

Really appreciate the nudge in the right direction – gold stuff 🔧😄

seatracker, 2025-07-26:
@MasterKeyxda 
Thanks for taking the lead here!

Very nice the guide you used, thanks for sharing.  Here is the one I used, to give credit:
https://serialize.wordpress.com/2017/02/12/extract-4366c0-firmware/

My router is running very stable, but i mostly have IoT devices connected to it.  I remember having issues with WPA3, and I think 802.11.w.  If I remember correctly, the radio would not go up when those were enabled.

mohd-akram, 2025-09-18:
Ran into the same thing with my AC88U which used the brcmfmac4366b-pcie.bin that came with OpenWrt. The file provided by @seatracker (with MD5 sum bbd41d50d40b4b0dcf8b6810f88bcd7f) matches the one in the last firmware available for the AC88U, 3.0.0.4.386_48322. It would be great if OpenWrt carried this version of the firmware by default. Thank you all!

Abdoymani, 2025-10-23:
Despite having `brcmfmac-firmware-43602a1-pcie` installed, the driver reports missing `.txt`, `.clm_blob`, and `.txcap_blob` files.

### What I have tried

- Installed all related packages: `kmod-brcmfmac`, `brcmfmac-firmware-43602a1-pcie`.
- Extracted multiple Netgear stock firmwares (`R8000-V1.0.4.xx.chk`) using `binwalk` — **no CLM or TXCAP blobs were found**.
- Tested firmwares from Asus RT-AC88U and Linksys EA8500 (same BCM43602 chipset) — only partial success.
- Attempted to download blobs from `linux-firmware.git` and several mirrors, but these specific files are missing for BCM43602 PCIe.
- Created a custom NVRAM `.txt` file to extend channel support; partial improvement, but TX queue timeouts persist.

### What I need

Could anyone please:
1. Share working versions of the following files (for BCM43602 PCIe):
   - `brcmfmac43602-pcie.txt`
   - `brcmfmac43602-pcie.clm_blob`
   - `brcmfmac43602-pcie.txcap_blob`
2. Or confirm whether these blobs are embedded in the proprietary Broadcom firmware and not distributed publicly?

Any help or pointers would be greatly appreciated.
This issue is reproducible on OpenWrt (latest snapshot, kernel 6.6.x) and affects Wi-Fi channel availability and TX stability.

### References
- Similar discussions on OpenWrt and linux-firmware mailing lists mention the same missing blobs for BCM43602.
- Device tree and dmesg logs available upon request.

endoftheline818, 2026-09-07:
Confirming this on a different chip revision, in case it's useful for narrowing the scope.

**TP-Link Archer C2300 v1** — `BCM43666/4`, so the **c0** variant and `brcmfmac4366c-pcie.bin`, where the original report was a `BCM4366/3` on the *b* variant. Same symptom exactly: associate, four-way handshake completes, DHCP lease obtained, then traffic stops and the client disconnects.

The affected build is the only `4366c` image in linux-firmware (added in `211de1679a68`, so there's no older revision to roll back to):

```
Firmware: BCM43666/4 wl0: Nov  5 2018 03:19:56 version 10.28.2 (r769115) FWID 01-d2cbb8fd
```

Switching to the `4366c0` file @seatracker attached above resolved it:

```
Firmware: BCM43666/4 wl0: Aug 27 2021 18:31:04 version 10.10.122.303 (r666429) FWID 01-ccd91e51
```

| | `AP-STA-CONNECTED` | `AP-STA-DISCONNECTED` | window |
| --- | --- | --- | --- |
| 10.28.2 | 8 | 8 | 460 s |
| 10.10.122.303 | 1 | **0** | 414 s |

Longest continuous association went from seconds to 364 s, and the link now sustains 346 Mbps down / 263 Mbps up with `tx failed` at 0.

One incidental observation: on 10.28.2, beacon RSSI measured from a fixed reference receiver returned *exactly* `-85.00` on every scan. On 10.10.122.303 it varies normally between reads (-74 to -80). I don't know what that indicates, but a perfectly constant value across repeated scans seemed worth mentioning.

For anyone testing this safely — the driver tries the device-specific filename first and falls back to the generic file if it's absent, so you can drop the alternate build in as e.g. `brcmfmac4366c-pcie.tplink,archer-c2300-v1.bin` and revert by deleting that one file. No need to overwrite the shipped image.

So this affects at least `BCM4366/3` (b) and `BCM43666/4` (c0) — consistent with both files coming out of the same November 2018 build run at `r769115`.

galyadoma, 2026-09-30:
Confirming this on an ASUS RT-AC88U (bcm53xx) with OpenWrt 24.10.4, and adding
measurements — replacing the firmware blob fixes it completely.

## Affected firmware

Shipped by `brcmfmac-firmware-4366b1-pcie 20241110-r2`:
Version: 10.28.2 (r769115) Date: Mon 2018-11-05
FWID: 01-801fb449
sha256: ab4d5012393aed59dc76e27d6e11d1c8f821ab24c3005436dee8d799b6298224

Note the RT-AC88U has two BCM4366 chips (2.4 GHz and 5 GHz) that both load
`brcmfmac4366b-pcie.bin`, so the bug degrades both radios at once.

## Symptoms with 10.28.2

- Clients associate, complete the 4-way handshake and get a DHCP lease, then
  drop 55–90 s later. From the client's point of view: "connected, no internet".
- TX rate is pinned at 6.5 Mbit/s regardless of signal strength — even with the
  client one metre from the router.
- `brcmf_run_escan: error (-52)` on every scan, which also breaks hostapd's
  OBSS scan (`noscan=0`).
- `brcmf_cfg80211_start_ap: Set Channel failed: chspec=0xd892, -52` on most
  channel/width combinations; the AP silently stays on the previous channel.
- Hundreds of `brcmf_fweh_event_worker: PSM's watchdog has fired!` on **both**
  phys — a couple of hundred over two weeks of uptime.

## After replacing the blob

Using the 2021 build posted earlier in this thread:
Version: 10.10.69.69042 (r798414) Date: Fri 2021-08-27
FWID: 01-f537b78e
sha256: 7bbed5f05c9b026c5dd35d7491c4135620030bcfa4d96bb3672b737a6c5c4ab8
size: 1038627 bytes

- Per-client TX rates of **526.5 Mbit/s** and **390 Mbit/s** (VHT80, ch 36),
  vs 6.5 Mbit/s before.
- `dmesg | grep -c "PSM's watchdog"` → **0** over 45+ hours of uptime.
- No more disconnects; scanning works again.
- 2.4 GHz throughput improved as well (~75 → ~100 Mbit/s measured), which is
  consistent with both chips sharing the blob.
- End-to-end: fast.com went from 75 Mbit/s to 170–180 Mbit/s on a 200 Mbit line.

Nothing else was changed — same OpenWrt build, same wireless config, single
`cp` of the .bin plus a reboot.

## Remaining limitation

Upper 5 GHz channels are still rejected after the swap:
brcmf_cfg80211_start_ap: Set Channel failed: chspec=0xe05b, -52

for channel 149, while 36–48 work. This looks unrelated to the version
regression and rather a consequence of the missing CLM blob —
`brcmf_c_process_clm_blob: no clm_blob available (err=-2)` is logged with both
firmware versions, and `opkg files brcmfmac-firmware-4366b1-pcie` shows the
package ships only the `.bin`, no `brcmfmac4366b-pcie.clm_blob`.

Happy to provide full dmesg/logread from either firmware version if that helps.

rofra, 2026-10-12:
I successfilly tested the firmware backport on Asus RT-AC88U in command line on OpenWrt 25.12.5 r33051-f5dae5ece4 / LuCI (HEAD detached at 128a7812) branch 26.180.75667~128a781 with:

```bash
cd /tmp/
cp -p /lib/firmware/brcm/brcmfmac4366b-pcie.bin /root/brcmfmac4366b-pcie.bin.original

wget https://github.com/user-attachments/files/16665291/brcmfmac4366.zip -O brcmfmac4366.zip

apk add unzip
unzip brcmfmac4366.zip

cp /tmp/brcmfmac4366b-pcie.bin /lib/firmware/brcm/brcmfmac4366b-pcie.asus,rt-ac88u.bin

reboot
```