openwrt/openwrt #16115: Can I Build the old version of `iptables` package?

WenisOK, 2024-08-09:
### Describe the bug

the latest package `iptables`'s verion is `1.8.8`, It runs in my self -compiled OpenWRT firmware that does not run smoothly. so i change the folder `packages/network/utils/iptable` to commit `30c15d0`, it's version is `1.8.7`, when i changed and build the firmware, the console has throw error an make is stop, the console throw log:
```log
Applying /home/wenisok/openwrt/package/network/utils/iptables/patches/001-xtables-Call-init_extensions6-for-static-builds.patch using plaintext: 
patching file iptables/xtables-monitor.c
patching file iptables/xtables-restore.c
patching file iptables/xtables-save.c
patching file iptables/xtables-standalone.c
patching file iptables/xtables-translate.c

Applying /home/wenisok/openwrt/package/network/utils/iptables/patches/002-xtables-Call-init_extensions_a_b.patch using plaintext: 
patching file iptables/xtables-monitor.c
patching file iptables/xtables-restore.c
patching file iptables/xtables-save.c
patching file iptables/xtables-standalone.c
patching file iptables/xtables-translate.c
patching file iptables/xtables-arp.c
patching file iptables/xtables-eb.c

Applying /home/wenisok/openwrt/package/network/utils/iptables/patches/010-add-set-dscpmark-support.patch using plaintext: 
patching file extensions/libxt_CONNMARK.c
patching file include/linux/netfilter/xt_connmark.h

Applying /home/wenisok/openwrt/package/network/utils/iptables/patches/020-treewide-use-uint-instead-of-u_int.patch using plaintext: 
patching file extensions/libxt_conntrack.c
patching file include/libipq/libipq.h
patching file include/libiptc/libxtc.h
patching file include/linux/netfilter_arp/arpt_mangle.h
patching file iptables/xshared.c
Hunk #1 FAILED at 1025.
1 out of 1 hunk FAILED -- saving rejects to file iptables/xshared.c.rej
patching file iptables/xshared.h
Hunk #1 FAILED at 80.
1 out of 1 hunk FAILED -- saving rejects to file iptables/xshared.h.rej
patching file libipq/ipq_create_handle.3
patching file libipq/ipq_set_mode.3
Patch failed!  Please fix /home/wenisok/openwrt/package/network/utils/iptables/patches/020-treewide-use-uint-instead-of-u_int.patch!
make[3]: *** [Makefile:774: /home/wenisok/openwrt/build_dir/target-x86_64_musl/linux-x86_64/iptables-1.8.7/.prepared_a7686cd9aba23eb4895fb41c1e40eefd_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/wenisok/openwrt/package/network/utils/iptables'
time: package/network/utils/iptables/compile#0.70#0.33#0.90
    ERROR: package/network/utils/iptables failed to build.
make[2]: *** [package/Makefile:173: package/network/utils/iptables/compile] Error 1
make[2]: Leaving directory '/home/wenisok/openwrt'
make[1]: *** [package/Makefile:167: /home/wenisok/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wenisok/openwrt'
make: *** [/home/wenisok/openwrt/include/toplevel.mk:248: world] Error 2
```

### OpenWrt version

7455353

### OpenWrt release

7455353

### OpenWrt target/subtarget

x86

### Device

x86_64

### Image kind

Self-built image

### Steps to reproduce

1. change folder‘s commit `packages/network/utils/iptables` to commit `30c15d0`.
2. build the firmware: make -j1 V=s

### Expected behaviour

build openwrt firmware with `iptables-1.8.7`

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
# CONFIG_GRUB_CONSOLE is not set
# CONFIG_GRUB_IMAGES is not set
CONFIG_HTOP_LMSENSORS=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LUCI_LANG_zh_Hans=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_firewall=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_htop=y
# CONFIG_PACKAGE_jansson is not set
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-nat=y
CONFIG_PACKAGE_kmod-nf-ipt=y
# CONFIG_PACKAGE_kmod-nft-fib is not set
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libip4tc=y
CONFIG_PACKAGE_libip6tc=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo=y
CONFIG_PACKAGE_libiwinfo-data=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
# CONFIG_PACKAGE_libmnl is not set
CONFIG_PACKAGE_libncurses=y
# CONFIG_PACKAGE_libnftnl is not set
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-i18n-base-zh-cn=y
CONFIG_PACKAGE_luci-i18n-firewall-zh-cn=y
CONFIG_PACKAGE_luci-i18n-opkg-zh-cn=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_vim-full=y
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=384
# CONFIG_TARGET_ROOTFS_TARGZ is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-09:
What problem are you trying to solve?
Your exhibits present 1.8.7 being defective after your patches but no problem with 1.8.8
Best place to seek help with your development is some forum, as this post while lengthy does not present any bug in OpenWRT whatsoever.

WenisOK, 2024-08-09:
I want to compile an OpenWRT firmware, and I need to built a package in the firmware.
The address of the package is https://github.com/vernesong/openclash
But this package does not seem to work normally at `iptables 1.8.8` and it is normal after being tested `1.8.7` in `iptables`.
Then I replaced the folder `package/network/utils/iptables` to commit 30c15d06e872c3170367fbf3c33da3d2b7b639cd. but it is not successfuly.
My issues is how to build the firmware with `iptables 1.8.7` not `iptables 1.8.8`

brada4, 2024-08-10:
The page you refer to mentions `kmod-ipt-nat(iptables)`
With default firewall setup that module is not employed, you will need to remove kmod-nft-nat kmod-nf-nat after installing it,

closed 2024-08-22 by WenisOK