openwrt/openwrt #16117: When using nginx to reverse proxy uHTTPd with an SSL certificate, the response is often very slow.

VergilGao, 2024-08-09:
### Describe the bug

My uHTTPd instance uses a self-signed SSL certificate to provide HTTPS connections. At the same time, I have an Nginx service to reverse proxy all web services in my internal network.

Assuming my OpenWrt’s IP address is 192.168.16.1, uHTTPd’s HTTP port is 4001, and HTTPS is 4002.

When I use Nginx to reverse proxy uHTTPd over HTTP, everything works fine. When I directly access https://192.168.16.1:4002, everything is also fine. However, when I use Nginx to reverse proxy https://192.168.16.1:4002, the problem occurs, and the webpage often takes 10 to 20 seconds to load.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

N100

### Image kind

Self-built image

### Steps to reproduce

the `self-signed-cert.sh`:

```
#!/bin/sh
mkdir -p "$CN" && cd "$CN"

openssl ecparam -genkey -name prime256v1 -out cakey.pem
openssl req -x509 -sha256 -new -nodes -key cakey.pem -days 3650 -out "${CN}.ca.pem" -subj "/CN=vergilgao.home" -addext "subjectAltName=DNS:vergilgao.home"

openssl ecparam -genkey -name prime256v1 -out "${CN}.key"
openssl req -sha256 -new -key "${CN}.key" -out "${CN}.csr" -subj "/CN=${CN}" -addext "subjectAltName=DNS:${CN}"
openssl x509 -sha256 -req -in "${CN}.csr" -CA "${CN}.ca.pem" -CAkey cakey.pem -CAcreateserial -out "${CN}.crt" -days 3650 -extensions v3_req -extfile /etc/ssl/openssl.cnf

rm cakey.pem

cat "${CN}.crt" "${CN}.key" > "${CN}.bundle.pem"

openssl pkcs12 -export -in "${CN}.crt" -inkey "${CN}.key" -out "${CN}.p12" -passout pass:
```

use `./self-signed-cert.sh 192.168.16.1` get `192.168.16.1.csr` for `uhttpd.csr` `192.168.16.1.key` for `uhttpd.key` `192.168.16.1.ca.pem` for nginx

the `/etc/config/uhttpd`:

```
config uhttpd 'main'                                                                                                                                                                                option redirect_https '0'
        option home '/www'
        option rfc1918_filter '1'                                                                                                                                                                   option max_requests '20'
        option max_connections '20'
        option cgi_prefix '/cgi-bin'
        list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
        option script_timeout '60'
        option network_timeout '30'
        option http_keepalive '50'
        option tcp_keepalive '20'
        option ubus_prefix '/ubus'
        option cert '/etc/uhttpd.crt'
        option key '/etc/uhttpd.key'                                                                                                                                                                
        list listen_https '192.168.16.1:4002'
```
[... 178 lines cut ...]
192.168.99.101 [09/Aug/2024:15:39:22 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 22.362s/22.362s) <- "https://router.example.com/cgi-bin/luci/admin/system/mounts" "-"
192.168.99.101 [09/Aug/2024:15:41:07 +0800] GET https://router.example.com/luci-static/argon/js/jquery.min.js?v=3.5.1 => 200 (response 89391B in 11.242s/11.242s) <- "https://router.example.com/cgi-bin/luci/admin/services/homeproxy/server" "-"
192.168.99.101 [09/Aug/2024:16:16:55 +0800] GET https://router.example.com/luci-static/argon/js/jquery.min.js?v=3.5.1 => 200 (response 89391B in 5.331s/5.330s) <- "https://router.example.com/cgi-bin/luci/admin/status/logs" "-"
192.168.99.101 [09/Aug/2024:16:16:55 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 5.333s/5.333s) <- "https://router.example.com/cgi-bin/luci/admin/status/logs" "-"
192.168.99.101 [09/Aug/2024:16:22:56 +0800] GET https://router.example.com/luci-static/argon/js/jquery.min.js?v=3.5.1 => 200 (response 89391B in 5.203s/5.203s) <- "https://router.example.com/cgi-bin/luci/admin/status/logs" "-"
192.168.99.101 [09/Aug/2024:16:23:01 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 10.708s/10.708s) <- "https://router.example.com/cgi-bin/luci/admin/status/logs" "-"
192.168.99.101 [09/Aug/2024:16:24:18 +0800] GET https://router.example.com/luci-static/argon/js/jquery.min.js?v=3.5.1 => 200 (response 89391B in 5.100s/5.100s) <- "https://router.example.com/cgi-bin/luci/admin/system/system" "-"
192.168.99.101 [09/Aug/2024:16:24:23 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 10.602s/10.601s) <- "https://router.example.com/cgi-bin/luci/admin/system/system" "-"
192.168.99.101 [09/Aug/2024:16:25:31 +0800] POST https://router.example.com/cgi-bin/cgi-exec?1723191909350 => 200 (response 215B in 21.067s/21.074s) <- "https://router.example.com/cgi-bin/luci/admin/system/opkg" "-"
192.168.99.101 [09/Aug/2024:16:26:00 +0800] POST https://router.example.com/cgi-bin/cgi-exec?1723191948711 => 200 (response 4448031B in 10.691s/10.691s) <- "https://router.example.com/cgi-bin/luci/admin/system/opkg" "-"
192.168.31.154 [09/Aug/2024:17:06:17 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 5.478s/5.478s) <- "https://router.example.com/cgi-bin/luci/" "-"
192.168.20.4 [09/Aug/2024:17:19:09 +0800] GET https://router.example.com/luci-static/argon/js/jquery.min.js?v=3.5.1 => 200 (response 89391B in 10.518s/10.518s) <- "https://router.example.com/cgi-bin/luci/admin/status/logs" "-"
192.168.20.4 [09/Aug/2024:17:19:09 +0800] GET https://router.example.com/luci-static/argon/css/cascade.css?v=2.3.1 => 200 (response 154305B in 10.519s/10.519s) <- "https://router.example.com/cgi-bin/luci/admin/status/logs" "-"
192.168.20.4 [09/Aug/2024:17:19:10 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 11.030s/11.030s) <- "https://router.example.com/cgi-bin/luci/admin/status/logs" "-"
192.168.20.4 [09/Aug/2024:17:21:29 +0800] GET https://router.example.com/luci-static/argon/css/cascade.css?v=2.3.1 => 200 (response 154305B in 11.022s/11.022s) <- "https://router.example.com/cgi-bin/luci/admin/network/routes" "-"
192.168.20.4 [09/Aug/2024:17:24:44 +0800] GET https://router.example.com/luci-static/argon/js/jquery.min.js?v=3.5.1 => 200 (response 89391B in 10.977s/10.977s) <- "https://router.example.com/cgi-bin/luci/admin/system/system" "-"
192.168.20.4 [09/Aug/2024:17:25:08 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 10.889s/10.889s) <- "https://router.example.com/cgi-bin/luci/admin/services/homeproxy" "-"
192.168.20.4 [09/Aug/2024:17:57:03 +0800] GET https://router.example.com/luci-static/argon/js/jquery.min.js?v=3.5.1 => 200 (response 89391B in 5.197s/5.197s) <- "https://router.example.com/cgi-bin/luci/admin/network/network" "-"
192.168.20.4 [09/Aug/2024:17:57:08 +0800] GET https://router.example.com/luci-static/argon/css/cascade.css?v=2.3.1 => 200 (response 154305B in 10.698s/10.698s) <- "https://router.example.com/cgi-bin/luci/admin/network/network" "-"
192.168.20.4 [09/Aug/2024:17:57:08 +0800] GET https://router.example.com/cgi-bin/luci/admin/translations/zh-cn?v=git-23.233.52805-dae2684 => 200 (response 159912B in 10.698s/10.698s) <- "https://router.example.com/cgi-bin/luci/admin/network/network" "-"
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

VergilGao, 2024-08-09:
here is a script to get the slow connection info in nginx logs:

```
#!/bin/sh

# 定义日志文件路径
log_file="$1"

grep "response" "$log_file" | while read -r line; do
    response_time=$(echo "$line" | sed -n 's/.*response [0-9]*B in \([0-9.]*\)s.*/\1/p')
    if [ $response_time ] ; then
        response_time=$(printf "%.0f" "$response_time")
        # 检查响应时间是否大于1秒
        if [ "$response_time" -gt 1 ]; then
            echo "$line"
        fi
    fi
done
```

brada4, 2024-08-09:
Why dont you use nginx in place of uhttpd?

VergilGao, 2024-08-09:
> Why dont you use nginx in place of uhttpd?

Because I have another access point (AP), it’s struggling a bit to run Nginx

brada4, 2024-08-10:
256MB sized requests are not needed for OpenWRT, it is clear mistake in config and you have to ask where you scratched that from the internet.

VergilGao, 2024-08-10:
> 256MB sized requests are not needed for OpenWRT, it is clear mistake in config and you have to ask where you scratched that from the internet.

I modified it myself because I had previously compiled a firmware over 150MB (for testing purposes). I changed the client_max_body_size to make it easier to upload the firmware from the web.

brada4, 2024-08-10:
You have to seek help with nginx tuning somewhere else. It is not an OpenWRT bug as you claim.

jow-, 2024-08-10:
When you terminate HTTPS on nginx, why speak HTTPS on the backend as well? Maybe it already helps to simply forward as plain HTTP.

VergilGao, 2024-08-10:
> When you terminate SSL on nginx, why speak HTTPS on the backend as well? Maybe it already helps to simply forward as plain HTTP.

Because my nginx instance and openwrt are not on the same device, the http communication between nginx and openwrt is still in plain text and cannot be protected.

brada4, 2024-08-10:
20s response is reasonably expected with likes of `opkg update`
You have to accomodate that in your non-openwrt RP