openwrt/openwrt #16126: Wi-Fi interfaces do not want to start

castillofrancodamian, 2024-08-09:
### Describe the bug

Wireless interfaces do not start even after restarting the router.

### OpenWrt version

r27100-21a5b3b540

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

MERCUSYS MR70X v1

### Image kind

Official downloaded image

### Additional info

<details><summary>/etc/config/wireless</summary>
<p>

```
root@mr70x:~# cat /etc/config/wireless

config wifi-device '2g'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option band '2g'
        option htmode 'HE20'
        option country 'AR'
        option cell_density '1'
        option channel '1'

config wifi-iface '2g_lag'
        option device '2g'
        option network 'lan'
        option mode 'ap'
        option ssid 'LAG'
        option encryption 'psk2+ccmp'
        option ifname '2g_lag'
        option key 'password'
        list maclist 'xx:xx:xx:xx:xx:xx'
        option macfilter 'allow'
        option ieee80211r '1'
        option ft_over_ds '0'
        option ft_psk_generate_local '1'
        option ieee80211k '1'
        option time_advertisement '2'
        option time_zone '<-03>3'
        option wnm_sleep_mode '1'
        option wnm_sleep_mode_no_keys '1'
        option bss_transition '1'
        option proxy_arp '1'
        option ieee80211w '1'

config wifi-device '5g'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option channel '100'
        option channels '100-144'
        option band '5g'
        option htmode 'HE80'
[... 583 lines cut ...]
Fri Aug  9 19:45:53 2024 kern.info kernel: [  114.435498] br-lan: port 3(5g_lag) entered learning state
Fri Aug  9 19:46:02 2024 kern.info kernel: [  122.755417] br-lan: port 3(5g_lag) entered forwarding state
Fri Aug  9 19:46:02 2024 kern.info kernel: [  122.761018] br-lan: topology change detected, propagating
Fri Aug  9 20:14:37 2024 daemon.notice netifd: wan (3728): udhcpc: sending renew to server 172.20.2.95
Fri Aug  9 20:14:37 2024 daemon.notice netifd: wan (3728): udhcpc: lease of 24.232.82.34 obtained from 172.20.2.95, lease time 21600
Fri Aug  9 20:14:38 2024 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Fri Aug  9 20:23:09 2024 authpriv.info dropbear[10539]: Child connection from 192.168.20.100:47178
Fri Aug  9 20:23:09 2024 authpriv.notice dropbear[10539]: Password auth succeeded for 'root' from 192.168.20.100:47178
Fri Aug  9 20:24:32 2024 authpriv.info dropbear[10584]: Child connection from 192.168.20.100:49624
Fri Aug  9 20:24:34 2024 authpriv.notice dropbear[10584]: Password auth succeeded for 'root' from 192.168.20.100:49624
Fri Aug  9 20:26:10 2024 authpriv.info dropbear[10539]: Exit (root) from <192.168.20.100:47178>: Exited normally
```

</p>
</details>

### Diffconfig

```text
make image PROFILE=mercusys_mr70x-v1 PACKAGES="wpa-supplicant-mbedtls hostapd-mbedtls hostapd-utils luci-ssl htop nano iperf3 luci-i18n-firewall-es luci-i18n-opkg-es luci-theme-openwrt-2020 luci-i18n-base-es luci-i18n-commands-es luci-lua-runtime luci-proto-wireguard qrencode luci-proto-relay luci-i18n-sqm-es openssh-sftp-server avahi-dbus-daemon curl luci-i18n-mwan3-es luci-i18n-nlbwmon-es luci-i18n-ddns-es -wpad-basic-mbedtls"
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-10:
Since br-invitados has no permanent ports you have to check the box saying to bring up empty bridge.

castillofrancodamian, 2024-08-10:
And why do wireless networks appear after restarting without making any changes?