openwrt/openwrt #16143: "Port shadow" vulnerability doesn't seem to be addressed

waruiji, 2024-08-11:
### Describe the bug

The vulnerability affects both OpenVPN and Wireguard servers.
For example see [Vulnerabilities in VPNs: Paper presented at the Privacy Enhancing Technologies Symposium 2024 - The Citizen Lab](https://citizenlab.ca/2024/07/vulnerabilities-in-vpns-paper-presented-at-the-privacy-enhancing-technologies-symposium-2024/).

Although [CVE-2021-3773 - Netfilter OpenVPN Information Disclosure Vulnerability](https://cvefeed.io/vuln/detail/CVE-2021-3773) mentions OpenVPN, the fix seem to affect Wireguard as well, as it's in the `netfilter nf_tables`.

The fix seems to be available in kernel ver. 5.17, but not ver. 5.15.162 that latest OpenWRT comes with (see  [netfilter: nf_tables: disable register tracking](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.17)

If the kernel can't be updated, a clear workaround for both OpenVPN and Wireguard would be highly appreciated.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ath79/generic

### Device

Netgear WNDR3700 v2

### Image kind

Self-built image

### Additional info

The reason I chose "Self-built image" as image kind is because I used "Attended Sysupgrade" to keep my packages and configs.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2024-08-11:
Kernel 6.6 is already in testing and will be [made default](https://github.com/openwrt/openwrt/pull/15712) soonish

rany2, 2024-08-11:
v5.15 doesn't appear to be impacted anyway. `netfilter: nf_tables: add register tracking infrastructure` isn't in 5.15 kernel so there is nothing to disable.

waruiji, 2024-08-11:
@rany2 It seems at least Red Hat addressed it in 5.15, at least for OpenVPN (see [2004949 – (CVE-2021-3773) CVE-2021-3773 kernel: lack of port sanity checking in natd and netfilter leads to exploit of OpenVPN clients](https://bugzilla.redhat.com/show_bug.cgi?id=2004949)), maybe there were more than one fix for it.

rany2, 2024-08-11:
@waruiji [This article](https://citizenlab.ca/2024/07/vulnerabilities-in-vpns-paper-presented-at-the-privacy-enhancing-technologies-symposium-2024/) does mention the following:

> Netfilter [did](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=4a6fbdd801e882ee6ca5cdfdc3374f0ae263174c) [commit](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=878aed8db324bec64f3c3f956e64d5ae7375a5de) and [then](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=a82c25c366b0963d33ddf699196e6cf57f6d89b1) [revert](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=ee0a4dc9f317fb9a97f20037d219802ca8de939b) a mitigation because it affected backwards compatibility with other use cases, offering an nftables [rule](https://lore.kernel.org/all/20220312220315.64531-1-pablo@netfilter.org/) as an alternative.

So a workaround is provided but it might be difficult to make use of in OpenWRT if you're using firewall4. I think you should open an issue with fw4 so the issue is mitigated there as it can't be dealt with in the kernel it seems.

waruiji, 2024-08-11:
The [paper itself](https://petsymposium.org/popets/2024/popets-2024-0070.pdf) mentions attack results (Table 1).
Then in Table 2 they mention possible mitigations.
I don't have enough context, but my first impression is that a firewall rule might address only one or two of possible issues identified.

brada4, 2024-08-11:
@jow- ?
Like this but selectable via conf?

cat /etc/nftables.d/masq-random.nft
```
chain srcnat_wan {
        meta nfproto ipv4 meta l4proto { tcp , udp , dccp , sctp } masquerade to :1024-65535 random
}
```