openwrt/openwrt #16148: mvebu: sysupgrade borked

anomeome, 2024-08-11:
### Describe the bug

sysupgrade does not upgrade image

Just going from images on hand, oldest that upgrades, to borked

r26868 - r26967

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

mvebu

### Device

mamba, rango

### Image kind

Self-built image

### Steps to reproduce

GUI or CLI sysupgrade

### Actual behaviour

no upgrade

### Expected behaviour

an upgrade

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ghoffman9, 2024-08-12:
yes from clean repository downloads.openwrt.org
'factory' image: r27110
https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/openwrt-mvebu-cortexa9-linksys_wrt32x-squashfs-factory.img flashed to both partitions successfully  using mtd write kerenl1 and kernel2
sysupgrade image r27110:
https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/openwrt-mvebu-cortexa9-linksys_wrt32x-squashfs-sysupgrade.bin

on my wrt32x:
this will work from partion 1-> partition 2
but from partition 2-> partion 1 it fails

fails go back at least to r26954

thx

i have some past builds and last factory and sysupgrade files that i have that worked were r26300 but these findings are lmitied to what i have saved.

dynasticorpheus, 2024-08-17:
I am also having the same issue on my wrt32x

ghoffman9, 2024-08-18:
i've doen more testing of old build versions i could recover.
builds (and date downloaded) in which sysupgrade works correctly both from part 1->2 and 2->1:
r26300, 6/2/24
26657, 6/8/24
26608, 6/12/24
26912, 7/10/24
26928, 7/12/24
26955, 7/17/24

builds which fail 20>1:
26775, 6/24/24
26993, 7/25/24
27110, 8/11/24

this interval from 6/24- 7/25 corresponds to kernel version commits, but also to the shift from auc to owut, and come changes to ubus calls were made as part of this. specifically,

I'm planning to deprecate auc once owut is fully fit for the task. I anyway wanted to replace auc with a ucode implementation, and now [@efahl](https://forum.openwrt.org/u/efahl) was quick to implement it with even more features than expected cake It will still need some minor fixes here and there, see comments in https://github.com/openwrt/packages/pull/24324, but I expect this to be doable before the next stable release which will then no longer offer auc.
in which [@daniel](https://forum.openwrt.org/u/daniel) and [@psherman](https://forum.openwrt.org/u/psherman) are discussing plans for ucode changes for sysupgrade, and

Testers invited! owut has made it into the SNAPSHOT package feed, so opkg update && opkg install owut Documentation is at [owut wiki page](https://openwrt.org/docs/guide-user/installation/sysupgrade.owut) Source at [owut github repo](https://github.com/efahl/owut/) Report issues, questions, comments in this thread (or as [owut github issues](https://github.com/efahl/owut/issues) if you're more comfortable there) Suggested first steps: owut check - Just to see what your installation looks like and get comfortable with things. Does the output match your expected values wrt correct target and arch, correct fs type, versions, et …
in which owut made it into snapshot on 6/25/24.
there was subsequent discussion about sysupgrade failing in

Sure! If you need I can list my init.d scripts, but nothing out of the ordinary (nor did I customize anything): root@AX3000T_Router:/etc/init.d# ls adblock collectd done gpio_switch luci_statistics odhcpd rpcd sysfixtime ucitrack urandom_seed boot cron dropbear led network packet_steering sqm sysntpd uhttpd urngd bootcount dnsmasq …
curiously, luci-advanced-reboot stoped being able to decode the openwrt version in the other partition during this time interval also.

so i wonder if 'something' the complex sysupgrade-ucode logic is borked for the wrt32x/venom series?

ghoffman9, 2024-08-21:
to clarify, bump, and get triaged again:
Describe the bug
sysupgrade does not upgrade image

Just going from images on hand, oldest that upgrades, to borked

r26868 - r26967

OpenWrt version
master

OpenWrt release
master

OpenWrt target/subtarget
mvebu

Device
**venom**

Image kind
from https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/
https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/openwrt-mvebu-cortexa9-linksys_wrt32x-squashfs-sysupgrade.bin

Steps to reproduce
GUI or CLI sysupgrade

Actual behaviour
no upgrade - sysupgrade fails

ghoffman9, 2024-09-16:
this is not invalid. sysupgrade is borked for wrt1900 and for wrt32x since about r26300.

Z10N0110, 2024-09-19:
I got similar result that sysupgrade command does not work on another mvebu target `synology ds213j`. Just found out few days ago when do the new fresh installation by follow the steps described in commit 3f7637b05047badc15a58b1581c071892bc6e340.

Here is the UART serial console output when i do the `sysupgrade -n openwrt-mvebu-cortexa9-synology_ds213j-squashfs-sysupgrade.bin` in another SSH session shell.

```shell
root@OpenWrt:~# Watchdog handover: fd=3
- watchdog -
Watchdog does not have CARDRESET support
Mon Sep 16 15:36:39 UTC 2024 upgrade: Sending TERM to remaining processes ...
Mon Sep 16 15:36:43 UTC 2024 upgrade: Sending KILL to remaining processes ...
[  130.042087] stage2 (7699): drop_caches: 3
Mon Sep 16 15:36:49 UTC 2024 upgrade: Switching to ramdisk...
Mon Sep 16 15:36:52 UTC 2024 upgrade: Performing system upgrade...
[  132.402326] do_stage2 (7699): drop_caches: 3
Could not open mtd device: firmware
Can't open device for writing!
cat: write [  132.428279] reboot: Restarting system
���������������������������������
BootROM 1.08
Booting from SPI flash
High speed PHY - Version: 2.1.2 (COM-PHY-V20) 
Update PEX Device ID 0x67100
High speed PHY - Ended Successfully
                                   DDR3 Training Sequence - Ver 4.5.0 
DDR3 Training Sequence - Ended Successfully 
Status = MV_OK
BootROM: Image checksum verification PASSED

 __   __                      _ _
|  \/  | __ _ _ ____   _____| | |
| |\/| |/ _` | '__\ \ / / _ \ | |
| |  | | (_| | |   \ V /  __/ | |
|_|  |_|\__,_|_|    \_/ \___|_|_|
         _   _     ____              _
        | | | |   | __ )  ___   ___ | |_ 
        | | | |___|  _ \ / _ \ / _ \| __| 
        | |_| |___| |_) | (_) | (_) | |_ 
         \___/    |____/ \___/ \___/ \__| 
 ** LOADER **

U-Boot 2011.12-SYNO-add48c3-2013-07-18 (Jul 18 2013 - 16:45:54) Marvell version: v2011.12 2013_Q1.2

Board: SYNO-DS213j-BP
SoC:   MV6710 A1
CPU:   Marvell PJ4B v7 UP (Rev 1) LE
       CPU    @ 1200 [MHz]
       L2     @ 600 [MHz]
       TClock @ 200 [MHz]
       DDR    @ 600 [MHz]
       DDR 16Bit Width, FastPath Memory Access
DRAM:  512 MiB

Map:   Code:		0x1ff0a000:0x1ffa1ed8
       BSS:		0x1ffefca0
       Stack:		0x1f9f9ef8
       Heap:		0x1f9fa000:0x1ff0a000

MMC:   MRVL_MMC: 0
SF: Detected M25P64 with page size 64 KiB, total 8 MiB
PEX 0.0(0): Detected No Link.
PEX 0.1(1): Detected No Link.
FPU not initialized
USB 0: Host Mode
USB 1: Host Mode
Modules/Interfaces Detected:
       PEX0 (Lane 0)
       SATA0 (Lane 2)
       SATA1 (Lane 3)
       SGMII0 Phy module (Lane 1)

Synology Model: DS213j
Fan Status: Good

Net:   egiga0 [PRIME]
Press Ctrl+C to abort autoboot in 3 second
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Marvell>> printenv
CASset=min
MALLOC_len=5
autoload=no
baudrate=115200
bootargs=console=ttyS0,115200 ip=off initrd=0x8000040,8M root=/dev/md0 rw syno_hw_version=DS213jv10 ihd_num=2 netif_num=1 flash_size=8 root=/dev/md0 rw syno_hw_version=DS213jv10 ihd_num=2 netif_num=1 flash_size=8
bootargs_end=:10.4.50.254:255.255.255.0:KW40:eth0:none
bootargs_root=root=/dev/nfs rw
bootcmd=bootm f4110000
bootdelay=3
cacheShare=no
console=console=ttyS0,115200
disL2Cache=yes
disaMvPnp=no
eeeEnable=no
enaAutoRecovery=yes
enaClockGating=no
enaFPU=no
enaWrAllo=no
eth1addr=00:50:43:02:00:00
eth1mtu=1500
ethact=egiga0
ethaddr=00:50:43:02:02:00
ethmtu=1500
ethprime=egiga0
image_name=uImage
initrd_name=uInitrd
ipaddr=10.4.50.136
loadaddr=0x02000000
loads_echo=0
mvNetConfig=mv_net_config=1,(00:50:43:11:11:11,0:1:2:3:4),mtu=1500
mv_pon_addr=00:50:43:00:00:02
netbsd_en=no
netmask=255.255.255.0
netretry=no
pcieTune=no
pexMode=rc
pxe_files_load=:default.arm-armada370-db:default.arm-armadaxp:default.arm
pxefile_addr_r=3100000
rcvrip=169.254.100.100
rootpath=/srv/oneiric
sata_delay_reset=0
sata_dma_mode=yes
serverip=10.4.50.68
setL2CacheWT=no
standalone=fsload 0x2000000 $image_name;setenv bootargs $console $mtdparts root=/dev/mtdblock0 rw ip=$ipaddr:$serverip$bootargs_end; bootm 0x2000000;
stderr=serial
stdin=serial
stdout=serial
usb0Mode=host
usb1Mode=host
usb2Mode=device
usbActive=1
vxworks_en=no
yuk_ethaddr=00:00:00:EE:51:81

Environment size: 1511/65532 bytes
Marvell>> 
```

In the above output, you can clearly see the following lines: `Cannot open mtd device: firmware`, `Cannot open device for writing!`, which obviously means that no image is written to mtd, and u-boot is unable to boot the image which confirms this.

After a quick look at the code, i can tell this was printed by the [mtd](https://github.com/openwrt/openwrt/blob/d78dec3e19e3f2ab0f14ba6819357d50156b30b8/package/system/mtd/src/mtd.c#L134) command.

Don't know if this is related to owut.

SkewedZeppelin, 2024-10-01:
1XMR bounty to anyone who fixes this.

aleks-mariusz, 2024-10-01:
just to put my hand up as well, i've also had a possibly related issue w/ sysupgrade since roughly around mid-summer.  details [here](https://forum.openwrt.org/t/divested-wrt-no-nonsense-hardened-builds-for-linksys-wrt-series/80522/1512) but the gist of it is, i had also issues updating with custom builds on only one partition (not sure which exactly but i always had to use sysupgrade twice since the first flash resulted in a version frozen (w/ kernel 6.6.35 and other packages from that time) - i traced it to having occured between r26679 and r26844 (but by "trace", i just mean that r26844 had kernel 6.6.35 which seemed to continually "re-appear" after the first sysupgrade, and r26679 is the last version before the weirdness started).

borkra, 2024-10-12:
`Could not open mtd device: firmware`

linksys devices do not have mtd device called `firmware`. But Fortinet devices do, probably one of the Fortinet support commits broke this.

borkra, 2024-10-12:
Actually my suspect this commit: https://github.com/openwrt/openwrt/commit/6ff970bb51f53c731b0612152734cec82ca92543

it defined PART_NAME=firmware as global, which it should not be, except commit is from 2022...

borkra, 2024-10-12:
@rmilecki It looks like your commit https://github.com/openwrt/openwrt/commit/715634e6d1443eacdcb84b04d1028c1564b08dbf broke this
```
-       local board_dir="$(tar t${gz}f "$tar_file" | grep -m 1 '^sysupgrade-.*/$')"
+       local board_dir="$($cmd < "$tar_file" | tar tf - | grep -m 1 '^sysupgrade-.*/$')"
```
I am having hard time understanding how above code supposed to work when `$cmd` is empty, and what are you trying to achieve, could you please fix tit.

borkra, 2024-10-12:
The patch below fixes the problem, not sure if it was intended operation or not.

If you want to apply it keep in mind that nand.sh is installed on the router, so you need to change it with vi, or update router 2 or 3 times, before the change takes an effect.

```
diff --git a/package/base-files/files/lib/upgrade/nand.sh b/package/base-files/files/lib/upgrade/nand.sh
index 9fa3cd2ddd..ca302557c8 100644
--- a/package/base-files/files/lib/upgrade/nand.sh
+++ b/package/base-files/files/lib/upgrade/nand.sh
@@ -300,7 +300,7 @@ nand_upgrade_fit() {
 # Write images in the TAR file to MTD partitions and/or UBI volumes as required
 nand_upgrade_tar() {
 	local tar_file="$1"
-	local cmd="$2"
+	local cmd="${2:-cat}"
 	local jffs2_markers="${CI_JFFS2_CLEAN_MARKERS:-0}"
 
 	# WARNING: This fails if tar contains more than one 'sysupgrade-*' directory.  
```

closed 2024-10-13 by openwrt-bot

mbrinkmann, 2024-10-14:
Thanks a lot, problem solved on my acs1900 <3

ghoffman9, 2024-10-14:
@borkra @robimarko @anomeome  -
this issue has been solved after the work done by @bokra to find the offending code.
sysupgrade  now works correctly on my wrt32x and wrt1900v1. 
thank you

commit a9402a83 (2024-10-13): mvebu: fix sysupgrade failure

Linksys devices do no pass $cmd at all.

Fixes: #16148
Fixes: 715634e6d144 ("base-files: upgrade: nand: use "cmd" argument for extracting command")

Link: https://github.com/openwrt/openwrt/pull/16690
commit d21720fa (2024-10-13): mvebu: fix default partition name

The firmware partition name is specifc to ipTIME NAS1dual and should not be
set globally.

Fixes: #16148
Fixes: 6ff970bb51f5 ("mvebu: add support for ipTIME NAS1dual")

Link: https://github.com/openwrt/openwrt/pull/16690