openwrt/openwrt #16157: Openwrt does not boot into console

Olli1080, 2024-08-12:
### Describe the bug

With recent Releases my installation does not boot into console and refuses any input after the bootloader.
The same behavior occurs when booting into failsafe mode.

I did already try the latest stable release.

Any help is greatly appreciated.

[image]

### OpenWrt version

r27123-8cf9a932fa

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

HUNSN Micro Firewall Appliance

### Image kind

Official downloaded image

### Steps to reproduce

wget https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-x86-64-generic-ext4-combined-efi.img.gz

gunzip openwrt-x86-64-generic-ext4-combined-efi.img.gz
 
lsblk
 
dd if=openwrt-x86-64-generic-ext4-combined-efi.img bs=1M of=/dev/#some_nvme

shutdown -h now

##

Then just starting the machine.

### Actual behaviour

Does not show console and does not accept input like instructed in the picture.

### Expected behaviour

Openwrt console screen and input from keyboard.

### Additional info

Don't know if it has to do with it but the log tells me to press Enter to activate console but this doesn't help either.
Maybe related to #15615?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

markmentovai, 2024-08-13:
x86 images don’t have USB keyboard support enabled by default. The serial console on ttyS0 works correctly, but there’s also a virtual console configured on on tty1, and all you get is output. You’ve probably found that the keyboard can drive GRUB while booting, but then stops working once the kernel boots.

I build with `CONFIG_HID`, `CONFIG_HID_GENERIC`, `CONFIG_HID_SUPPORT`, `CONFIG_USB_HID`, and `CONFIG_USB_HIDDEV` to overcome this. You can almost certainly get away with less, but I use this to match my armsr builds.

Olli1080, 2024-08-13:
Thank you for the quick response. I will try out your suggestion and close the issue if it works.

Olli1080, 2024-08-19:
Works! Thank you very much.
Don't know why firefox opened the zip as plain text in browser but chrome downloaded it.

ynezz, 2024-08-20:
> I did already try the latest stable release.

@Olli1080 so that means 23.05.4, right?

Olli1080, 2024-08-20:
Yes that's right.
Adding  kmod-usb-hid and kmod-usb3 got it working.

markmentovai, 2024-08-20:
https://github.com/openwrt/openwrt/issues/16157#issuecomment-2298048458
> > add kmod-usb-hid kmod-usb3 to packages list
> 
> IMO this should work out of the box, so lets keep this ticket open so it will get fixed eventually.

→ #16208

markmentovai, 2024-08-20:
Many of these config symbols were previously set but were lost in the config refresh at 2a86425de1076ca014f6a9c57980ae3112d50008. This may have happened inadvertently because before that change, [`CONFIG_HID`](https://github.com/openwrt/openwrt/blob/2a86425de1076ca014f6a9c57980ae3112d50008%5E/target/linux/x86/config-6.6#L163), [`CONFIG_USB_HID`](https://github.com/openwrt/openwrt/blob/2a86425de1076ca014f6a9c57980ae3112d50008%5E/target/linux/x86/config-6.6#L380), and [`CONFIG_USB_HIDDEV`](https://github.com/openwrt/openwrt/blob/2a86425de1076ca014f6a9c57980ae3112d50008%5E/target/linux/x86/config-6.6#L381) were set at the target (not subtarget) level in target/linux/x86/config-6.6, but `CONFIG_HID_GENERIC` was also required to make this work, and it was only present at the subtarget level ([x86/generic](https://github.com/openwrt/openwrt/blob/2a86425de1076ca014f6a9c57980ae3112d50008%5E/target/linux/x86/generic/config-6.6#L172) and [x86/64](https://github.com/openwrt/openwrt/blob/2a86425de1076ca014f6a9c57980ae3112d50008%5E/target/linux/x86/64/config-6.6#L233)).

closed 2024-08-21 by openwrt-bot