openwrt/openwrt #16162: NanoPI R6S VLAN configuration using eth0 card

kmarzic, 2024-08-13:
### Describe the bug

Hi,

I am running OpenWRT snapshot on FriendlyElec NanoPI R6S.

I have noted that `eth0` card (RTL8211F Gigabit Ethernet) is not working properly with VLAN configuration.
Maybe I have some error in the configuration, or there is some bug.

Not all VLANs related `eth0` are created (created are first 3 VLANs from `/etc/config/network` configuration, 4 VLANs are missing).

Ethernet Cards:
eth0 - LAN2 - Native Gigabit Ethernet, RTL8211F-CG chip
eth1 - WAN - PCIe 2.5G Ethernet Ethernet, RTL8125BG chip
eth2 - LAN1 - PCIe 2.5G Ethernet Ethernet, RTL8125BG chip

### OpenWrt version

r27137-f51cb74473

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R6S

### Image kind

Self-built image

### Steps to reproduce

VLAN configuration:

| VID  | eth0 | eth2 | eth1 | br-eth |
|------|------|------|------|--------|
| 200  | u    | u    |      | u      |
| 201  | t    | t    |      | t      |
| 202  | t    | t    |      | t      |
| 203  | t    | t    |      | t      |
| 204  | t    | t    |      | t      |
| 205  | t    | t    |      | t      |
| 206  | t    | t    |      | t      |
| 207  | t    | t    |      | t      |
| 100  |      |      | t    | t      |
| 101  |      |      | t    | t      |
| 103  |      |      | t    | t      |
| 1500 |      |      | t    | t      |

```
root@np2:~# cat /etc/config/network
...
config device
        option name 'eth0'
        option macaddr 'b2:be:02:75:52:f1'

config device
        option name 'eth2'
        option macaddr 'b2:be:02:75:52:f1'

config device
        option name 'eth1'
        option macaddr 'b2:be:02:75:52:f0'

[... 186 lines cut ...]
                  202
root@np2:~#
```

### Expected behaviour

Those  VLANs should be created:

```
root@np2:~# bridge vlan show dev eth0
port              vlan-id
eth0              200 PVID Egress Untagged
                  201
                  202
                  203
                  204
                  205
                  206
                  207
root@np2:~#
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kmarzic, 2024-08-14:
Hi,

I have tested quicky ImmortalWRT fix: https://github.com/immortalwrt/immortalwrt/discussions/1407.
Maybe you can take a look and see if you can (or can not) use it in OpenWRT snapshot.

Regards

BWhitten, 2024-08-14:
@1715173329 do you want to take the synopsys patch via your PR #16149?

1715173329, 2024-08-15:
#16149 is irrelevant with this issue. Feel free to take over.

peterwillcn, 2024-08-21:
> #16149 is irrelevant with this issue. Feel free to take over (as I don't have this board).

Tested and repaired on nanopc-T6 device.

kmarzic, 2024-08-21:
Hi,
It seems there are still problems with NanoPI R6S with latest snapshot.
Still first 3 VLANs created on `eth0`, other 4 VLANs are missing. 
```
root@np2:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r27194-a2662309aa'
DISTRIB_TARGET='rockchip/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r27194-a2662309aa'
DISTRIB_TAINTS=''
root@np2:~#
```
```
root@np2:~# dmesg | grep eth0
[   11.070874] rk_gmac-dwmac fe1c0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   11.071857] rk_gmac-dwmac fe1c0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-1
[   11.304525] rk_gmac-dwmac fe1c0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   11.314997] rk_gmac-dwmac fe1c0000.ethernet eth0: No Safety Features support found
[   11.315664] rk_gmac-dwmac fe1c0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   11.316474] rk_gmac-dwmac fe1c0000.ethernet eth0: registered PTP clock
[   11.317419] rk_gmac-dwmac fe1c0000.ethernet eth0: FPE workqueue start
[   11.317983] rk_gmac-dwmac fe1c0000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[   11.354819] 8021q: adding VLAN 0 to HW filter on device eth0
[   13.895634] rk_gmac-dwmac fe1c0000.ethernet eth0: FPE workqueue stop
[   13.896269] rk_gmac-dwmac fe1c0000.ethernet eth0: Timeout accessing MAC_VLAN_Tag_Filter
[   13.896966] rk_gmac-dwmac fe1c0000.ethernet eth0: failed to kill vid 0081/0
[   19.247387] rk_gmac-dwmac fe1c0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   19.248406] rk_gmac-dwmac fe1c0000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-1
[   19.494529] rk_gmac-dwmac fe1c0000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   19.505057] rk_gmac-dwmac fe1c0000.ethernet eth0: No Safety Features support found
[   19.505724] rk_gmac-dwmac fe1c0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   19.506532] rk_gmac-dwmac fe1c0000.ethernet eth0: registered PTP clock
[   19.507325] rk_gmac-dwmac fe1c0000.ethernet eth0: FPE workqueue start
[   19.507890] rk_gmac-dwmac fe1c0000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[   19.544687] 8021q: adding VLAN 0 to HW filter on device eth0
[   19.546353] br-eth: port 1(eth0) entered blocking state
[   19.546813] br-eth: port 1(eth0) entered disabled state
[   19.547278] rk_gmac-dwmac fe1c0000.ethernet eth0: entered allmulticast mode
[   19.549483] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   19.550184] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   19.550881] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   19.551577] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   19.552274] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   20.037237] rk_gmac-dwmac fe1c0000.ethernet eth0: entered promiscuous mode
[   28.877491] rk_gmac-dwmac fe1c0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   28.877560] br-eth: port 1(eth0) entered blocking state
[   28.878689] br-eth: port 1(eth0) entered forwarding state
[   29.879308] rk_gmac-dwmac fe1c0000.ethernet eth0: left allmulticast mode
[   29.879954] rk_gmac-dwmac fe1c0000.ethernet eth0: left promiscuous mode
[   29.880607] br-eth: port 1(eth0) entered disabled state
[   29.945405] br-eth: port 1(eth0) entered blocking state
[   29.945870] br-eth: port 1(eth0) entered disabled state
[   29.946340] rk_gmac-dwmac fe1c0000.ethernet eth0: entered allmulticast mode
[   29.947073] rk_gmac-dwmac fe1c0000.ethernet eth0: entered promiscuous mode
[   29.947931] br-eth: port 1(eth0) entered blocking state
[   29.948388] br-eth: port 1(eth0) entered forwarding state
[   29.949028] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   29.949720] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   29.950408] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   29.951095] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
[   29.951782] rk_gmac-dwmac fe1c0000.ethernet eth0: MAC_VLAN_Tag_Filter full (size: 4)
root@np2:~#
```
```
root@np2:~# bridge vlan show dev eth0
port              vlan-id
eth0              200 PVID Egress Untagged
                  201
                  202
root@np2:~#
```

peterwillcn, 2024-08-27:
ping @mj22226

peterwillcn, 2024-08-27:
@kmarzic Try r8125 Ethernet controller 9.013.02-NAPI-RSS driver

1715173329, 2024-08-27:
The issue comes from the GMAC, not PCIe network card.

peterwillcn, 2024-08-27:
> The issue comes from the GMAC, not PCIe network card.

Yes, you're right.

Madguy93, 2025-01-10:
I'm also struggling with getting vlans working for eth0 on latest OpenWrt 24.10.0-rc5 r28304-6dacba30a7.
While I haven't tested vlans on the other interfaces yet, the config I'm trying to setup worked great when running friendlyElec so I believe it's a software issue.

dmsza, 2025-02-14:
I am not using VLANs with the R6S, but I am getting the error below in the log (24.10.0):

`Fri Feb 14 10:34:57 2025 kern.err kernel: [   15.746282] rk_gmac-dwmac fe1c0000.ethernet eth0: Timeout accessing MAC_VLAN_Tag_Filter`