openwrt/openwrt #16163: ppp-mod-passwordfd is broken

beelze, 2024-08-13:
### Describe the bug

passwordfd.so failed to load with 
```
/usr/sbin/pppd: Error relocating /usr/lib/pppd/2.4.9/passwordfd.so: eaptls_passwd_hook: symbol not found
/usr/sbin/pppd: Couldn't load plugin passwordfd.so
```

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (stock layout)

### Image kind

Official downloaded image

### Steps to reproduce

Providing `plugin passwordfd.so` option to `pppd` leads to an error. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

beelze, 2024-08-13:
sysupgraded from the ofiicial [release](https://downloads.openwrt.org/releases/23.05.4/targets/mediatek/filogic/openwrt-23.05.4-mediatek-filogic-xiaomi_redmi-router-ax6000-stock-squashfs-sysupgrade.bin)

Neustradamus, 2024-08-20:
It is possible to test with PPP 2.5.0 and PPP master code?

beelze, 2024-08-20:
any clues how to build ppp 2.5.0 using openwrt build system?

Neustradamus, 2024-08-21:
@beelze: Go here:
- https://github.com/openwrt/openwrt/pull/12848

DragonBluep, 2024-10-06:
It's not a bug, we didn't enable EAP-TLS at all.
https://github.com/openwrt/openwrt/blob/b62e6f5beb2a574c67ba5159f0049e54cdf25d69/package/network/services/ppp/patches/200-makefile.patch#L33