openwrt/openwrt #16165: qmi-wwan driver not loaded with Quectel EM060K modem

ASoTNetworks, 2024-08-14:
### Describe the bug

I am using a self compiled OpenWRT 23.05.4 on Gl.iNet E750v2 with the `EM060K-GL` modem.

I noticed that there was no wwan interface or `/dev/cdc-wdm0` device file but I fixed the driver issue by applying this patch by adding the device ID for this modem (`2c7c:030b Quectel EM060K-GL`):
```
Index: linux-5.15.162/drivers/net/usb/qmi_wwan.c
===================================================================
--- linux-5.15.162.orig/drivers/net/usb/qmi_wwan.c
+++ linux-5.15.162/drivers/net/usb/qmi_wwan.c
@@ -1082,6 +1082,7 @@ static const struct usb_device_id produc
 	},
 	{QMI_MATCH_FF_FF_FF(0x2c7c, 0x0125)},	/* Quectel EC25, EC20 R2.0  Mini PCIe */
 	{QMI_MATCH_FF_FF_FF(0x2c7c, 0x0306)},	/* Quectel EP06/EG06/EM06 */
+	{QMI_MATCH_FF_FF_FF(0x2c7c, 0x030b)},	/* Quectel EM060K */
 	{QMI_MATCH_FF_FF_FF(0x2c7c, 0x0512)},	/* Quectel EG12/EM12 */
 	{QMI_MATCH_FF_FF_FF(0x2c7c, 0x0620)},	/* Quectel EM160R-GL */
 	{QMI_MATCH_FF_FF_FF(0x2c7c, 0x0800)},	/* Quectel RM500Q-GL */
```

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ath79/nand

### Device

GL.iNet GL-E750

### Image kind

Self-built image

### Steps to reproduce

Build the image and flash it to the device. While configuring wwan0 and /dev/cdc-wdm0 is not available.

### Actual behaviour

wwan0 driver not loaded

### Expected behaviour

Driver loaded with /dev/cdc-wdm0 and wwan0 interface available without patching the driver file.

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_nand=y
CONFIG_TARGET_ath79_nand_DEVICE_glinet_gl-e750=y
CONFIG_ATH10K_LEDS=y
CONFIG_ATH10K_THERMAL=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
[... 221 lines cut ...]
CONFIG_PACKAGE_uqmi=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_vnstat2=y
CONFIG_PACKAGE_vnstati2=y
CONFIG_PACKAGE_wget-ssl=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mbedtls=y
CONFIG_PACKAGE_wwan=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_PING_LEGACY_SYMLINKS=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TRACEPATH_LEGACY_SYMLINKS=y
CONFIG_WPA_MBO_SUPPORT=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-08-14:
This is not an OpenWrt bug, but rather feature missing in the upstream kernel.
So, a proper solution would be to send the patch adding the device to the upstream kernel and then backporting it to OpenWrt.

marshall-lee, 2026-09-27:
Seems to be added in https://github.com/torvalds/linux/commit/0f2fd31f63c65c409bd336af3a42d478a9207c1f