openwrt/openwrt #16169: pkg_hash_check_unresolved: cannot find dependency kmod-crypto-kpp for kmod-crypto-lib-curve25519

peterwillcn, 2024-08-14:
### Describe the bug

ubuntu 22.04LTS
openwrt master latest
mediatek/filogic 
bananapi_bpi-r3-mini

```
CCollected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-crypto-kpp for kmod-crypto-lib-curve25519
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-curve25519 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-curve25519.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-wireguard:
 *      kmod-crypto-kpp
 * opkg_install_cmd: Cannot install package kmod-wireguard.
 ```

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

mediatek/filogic 

### Device

bananapi_bpi-r3-mini

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3-mini=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_AUTOREMOVE=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BIND_LIBJSON=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
# CONFIG_GDB is not set
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_HTOP_LMSENSORS=y
CONFIG_IFSTAT_SNMP=y
CONFIG_IPERF_ENABLE_MULTICAST=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
[... 363 lines cut ...]
CONFIG_PACKAGE_zyxel-bootconfig=m
CONFIG_PING_LEGACY_SYMLINKS=y
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_TRACEPATH_LEGACY_SYMLINKS=y
CONFIG_USE_LLVM_BUILD=y
CONFIG_WOLFSSL_HAS_DEVCRYPTO=y
CONFIG_WOLFSSL_HAS_DEVCRYPTO_FULL=y
CONFIG_ZSTD_OPTIMIZE_O3=y
# CONFIG_IB is not set
# CONFIG_MAKE_TOOLCHAIN is not set
CONFIG_PACKAGE_kmod-crypto-kpp=m
# CONFIG_SDK is not set
# CONFIG_SDK_LLVM_BPF is not set
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

peterwillcn, 2024-08-14:
> Invalid Release reported. `master` Is this from a clean repository?

SNAPSHOT

peterwillcn, 2024-08-14:
add fixed
```
CONFIG_KERNEL_BIG_KEYS=y
CONFIG_KERNEL_KEYS_REQUEST_CACHE=y
CONFIG_KERNEL_PERSISTENT_KEYRINGS=y
```

Neustradamus, 2024-08-21:
@peterwillcn: It has been fixed upstream?

peterwillcn, 2024-08-21:
> @peterwillcn: It has been fixed upstream?

Upstream configuration seed has not been fixed yet.

nroach44, 2024-09-26:
Can confirm that the three config changes are still required for the build to complete on 2023.05.05.

valkodev, 2024-11-03:
For me this didn't helped. Still getting the error during packgage installation. Treid snapshot as well as latest new tag 24.10. Without Wireguard the compliation is working fine. Tried to do distclean, same result. Any idea what to try? Thanks in advance.

Target architecture is ATH79

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-crypto-kpp for kmod-crypto-lib-curve25519
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-curve25519 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-curve25519.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-wireguard:
 * 	kmod-crypto-kpp
 * opkg_install_cmd: Cannot install package kmod-wireguard.