openwrt/openwrt #16180: Unable to get IPv6 on DHCPv6 client "Network device is not present" after PPPoE session resets (disconnects/reconnects)

vincejv, 2024-08-16:
### Describe the bug

When PPPoE interface or session is disconnected and reconnected, DHCPv6 client which uses a PPPoE device as the underlying interface no longer is able to retrieve IPv6 via DHCPv6, error is: `Network device is not present`, a full system reboot resolves the issue or `service network restart`

---
Previously posted on discourse: https://forum.openwrt.org/t/dhcpv6-client-network-device-is-not-present-after-pppoe-session-resets-disconnects-reconnects-system-restart-restores-the-ipv6-netifd-issue/206097/3

I applied the workaround stated here https://forum.openwrt.org/t/network-device-in-not-present/102844/17

the only way to fix this issue on the current netifd codebase is to restart the network service by `service network restart`... it isn't the ideal solution to me as this messes up the dynamic interfaces like wireguard and veth docker interfaces, idk how to restore it after the network service restart? restart docker daemon service? restart wireguard? might as well reboot the entire system

although the problem in the reference thread was DS-LITE, this applies to PPPoE DHCPv6 PD as well...

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

Dell Inc. OptiPlex 790

### Image kind

Official downloaded image

### Steps to reproduce

Pre-condition: You get IPv6 addresses/PD through PPPoE like this
[image]

1. Restart PPPoE using Luci

or

1. Get a random disconnect and reconnect with PPPoE, maybe perhaps unstable internet or power loss from ONT

### Actual behaviour

DHCPv6 client that uses the PPPoE interface is unable to get any IPv6 address even if PPPoE session is re-established and link local address is provisioned, it reports "Network device is not present"

### Expected behaviour

No error and DHCPv6 client should retrieve an IPv6 address and PD after PPPoE session is re-established

### Additional info

<details>
  <summary>/etc/config/network (redacted wireguard interfaces/clients as the keys are there)</summary>
  
  ```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd46:5d36:7a68::/48'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth1'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '172.16.0.1'
        option netmask '255.255.255.128'
        option ip6assign '60'

config interface 'wan'
        option device 'eth0.1020'
        option proto 'pppoe'
        option username 'pppoe@creds.com'
        option password 'xxxx'
        option ipv6 '1'
        option pppd_options 'debug logfile /var/log/pppd.log'
        option peerdns '0'
        option metric '10'
        option mtu '1500'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '1030'
        option name 'eth0.1030'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '2000'
        option name 'eth0.2000'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '1020'
        option name 'eth0.1020'

config interface 'intstbr'
        option device 'br-fa28d09b34a9'
        option proto 'none'
        option auto '0'

config device
        option type 'bridge'
        option name 'br-fa28d09b34a9'

config interface 'extstbr'
        option device 'br-a1986e704674'
        option proto 'none'
        option auto '0'

config device
        option type 'bridge'
        option name 'br-a1986e704674'

config interface 'ontgwacces'
        option proto 'static'
        option device 'eth0.1020'
        option ipaddr '192.168.254.100'
        option netmask '255.255.255.0'
        option defaultroute '0'

config interface 'mtkgwaccess'
        option proto 'dhcp'
        option device 'eth0.2000'
        option peerdns '0'
        option metric '25'

config device
        option type 'bridge'
        option name 'lxcbr0'
        option macaddr '00:16:3e:b2:00:00'
        option bridge_empty '1'

config interface 'lxcbr0'
        option proto 'static'
        option device 'lxcbr0'
        option ipaddr '172.23.0.1'
        option netmask '255.255.255.0'
        list ip6addr 'fd20:c766:2c36::1/64'

config interface 'ipsecbr0'
        option proto 'static'
        option device 'ipsec0'
        option ipaddr '172.25.0.1'
        option netmask '255.255.255.0'
        option delegate '0'
        option sourcefilter '0'
        list ip6addr 'fd6e:5c05:d5be::1/64'

config interface 'wan6'
        option proto 'dhcpv6'
        option device 'pppoe-wan'
        option reqaddress 'try'
        option reqprefix '56'
        option peerdns '0'
        option norelease '1'
        option sourcefilter '0'
        option metric '11'

config device
        option name 'eth0'
        option mtu '1508'

config interface 'wanb'
        option proto 'dhcp'
        option device 'eth0.1030'
        option hostname '*'
        option peerdns '0'
        option metric '12'

config interface 'wanb6'
        option proto 'dhcpv6'
        option device 'eth0.1030'
        option reqaddress 'try'
        option reqprefix '56'
        option peerdns '0'
        option metric '13'

config interface 'docker'
        option device 'docker0'
        option proto 'none'
        option auto '0'

config device
        option type 'bridge'
        option name 'docker0'

config device
        option type '8021q'
        option ifname 'eth0'
        option vid '1050'
        option name 'eth0.1050'

config interface 'wanlte'
        option proto 'dhcp'
        option device 'eth0.1050'
        option hostname '*'
        option peerdns '0'
        option metric '30'
  ```
</details>

<details>
  <summary>ubus call system board</summary>
  
  ```
{
        "kernel": "5.15.162",
        "hostname": "Pictor",
        "system": "Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz",
        "model": "Dell Inc. OptiPlex 790",
        "board_name": "dell-inc-optiplex-790",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.4",
                "revision": "r24012-d8dd03c46f",
                "target": "x86/64",
                "description": "OpenWrt 23.05.4 r24012-d8dd03c46f"
        }
}
  ```
</details>

### Diffconfig

```text
n/a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vincejv, 2024-08-16:
<details>
  <summary>Logs including pppd debug logs when Network device is not present issue exists</summary>
  
  ```
Sat Aug  3 04:01:02 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 172.16.0.5 c4:70:ab:98:52:bd EW1200G-PRO-9852BD
Sat Aug  3 04:49:01 2024 daemon.debug pppd[7409]: rcvd [LCP TermReq id=0x1c]
Sat Aug  3 04:49:01 2024 daemon.info pppd[7409]: LCP terminated by peer
Sat Aug  3 04:49:01 2024 daemon.info pppd[7409]: Connect time 1882.0 minutes.
Sat Aug  3 04:49:01 2024 daemon.info pppd[7409]: Sent 3155104696 bytes, received 3841838230 bytes.
Sat Aug  3 04:49:01 2024 daemon.info avahi-daemon[5075]: Withdrawing address record for 180.190.x.x on pppoe-wan.
Sat Aug  3 04:49:01 2024 daemon.info avahi-daemon[5075]: Leaving mDNS multicast group on interface pppoe-wan.IPv4 with address 180.190.x.x.
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 12[KNL] 180.190.x.x disappeared from pppoe-wan
Sat Aug  3 04:49:01 2024 daemon.info avahi-daemon[5075]: Interface pppoe-wan.IPv4 no longer relevant for mDNS.
Sat Aug  3 04:49:01 2024 daemon.debug pppd[7409]: Script /lib/netifd/ppp-down started (pid 134579)
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 14[KNL] fe80::e943:28ac:8d04:385f disappeared from pppoe-wan
Sat Aug  3 04:49:01 2024 daemon.info avahi-daemon[5075]: Interface pppoe-wan.IPv6 no longer relevant for mDNS.
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Interface 'wgcf' has lost the connection
Sat Aug  3 04:49:01 2024 daemon.info avahi-daemon[5075]: Leaving mDNS multicast group on interface pppoe-wan.IPv6 with address 2001:fd8:f421:5094:e943:28ac:8d04:385f.
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 12[KNL] interface pppoe-wan deactivated
Sat Aug  3 04:49:01 2024 daemon.info avahi-daemon[5075]: Withdrawing address record for 2001:fd8:f421:5094:e943:28ac:8d04:385f on pppoe-wan.
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 04[KNL] 2001:fd8:f421:5094:e943:28ac:8d04:385f disappeared from pppoe-wan
Sat Aug  3 04:49:01 2024 daemon.warn miniupnpd[39126]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sat Aug  3 04:49:01 2024 daemon.debug pppd[7409]: Script /lib/netifd/ppp-down started (pid 134581)
Sat Aug  3 04:49:01 2024 daemon.debug pppd[7409]: sent [LCP TermAck id=0x1c]
Sat Aug  3 04:49:01 2024 daemon.notice pppd[7409]: Modem hangup
Sat Aug  3 04:49:01 2024 daemon.notice pppd[7409]: Connection terminated.
Sat Aug  3 04:49:01 2024 daemon.info pppd[7409]: Connect time 1882.0 minutes.
Sat Aug  3 04:49:01 2024 daemon.info pppd[7409]: Sent 3155104696 bytes, received 3841838230 bytes.
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 15[KNL] interface pppoe-wan deleted
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 12[KNL] 172.16.0.2 disappeared from wgcf
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 14[KNL] 2606:4700:110:8f31:2b6b:6092:bdd0:545c disappeared from wgcf
Sat Aug  3 04:49:01 2024 user.info mwan3track[872988]: Check (ping) failed for target "1.0.0.1" on interface wgcf (wgcf). Current score: 10
Sat Aug  3 04:49:01 2024 user.info mwan3track[872988]: Check (ping) failed for target "8.8.4.4" on interface wgcf (wgcf). Current score: 10
Sat Aug  3 04:49:01 2024 user.notice mwan3track[872988]: Interface wgcf (wgcf) is disconnecting
Sat Aug  3 04:49:01 2024 user.notice mwan3-hotplug[134582]: Execute ifdown event on interface wgcf (unknown)
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 12[KNL] interface wgcf deactivated
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Interface 'wan' has lost the connection
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 15[NET] using forecast interface br-lan
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 15[CFG] joining forecast multicast groups: 224.0.0.1,224.0.0.22,224.0.0.251,224.0.0.252,239.255.255.250
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Interface 'wan6' is disabled
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Network device 'wgcf' link is down
Sat Aug  3 04:49:01 2024 daemon.notice netifd: wan6 (9180): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan6" } (Permission denied)
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 13[KNL] interface wgcf deleted
Sat Aug  3 04:49:01 2024 daemon.debug pppd[7409]: Script /lib/netifd/ppp-down finished (pid 134579), status = 0x1
Sat Aug  3 04:49:01 2024 daemon.debug pppd[7409]: Script /lib/netifd/ppp-down finished (pid 134581), status = 0x1
Sat Aug  3 04:49:01 2024 daemon.info pppd[7409]: Exit.
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Interface 'wan' is now down
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Interface 'wan' is setting up now
Sat Aug  3 04:49:01 2024 daemon.info pppd[134727]: Plugin pppoe.so loaded.
Sat Aug  3 04:49:01 2024 daemon.info pppd[134727]: PPPoE plugin from pppd 2.4.9
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 09[NET] using forecast interface br-lan
Sat Aug  3 04:49:01 2024 daemon.info ipsec: 09[CFG] joining forecast multicast groups: 224.0.0.1,224.0.0.22,224.0.0.251,224.0.0.252,239.255.255.250
Sat Aug  3 04:49:01 2024 daemon.notice pppd[134727]: pppd 2.4.9 started by root, uid 0
Sat Aug  3 04:49:01 2024 daemon.err odhcp6c[9180]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Sat Aug  3 04:49:01 2024 daemon.notice netifd: Interface 'wan6' is now down
[... cut ...]
Sat Aug  3 04:51:11 2024 user.info banIP-1.0.0-5[145269]: start banIP domain lookup
Sat Aug  3 04:51:11 2024 user.info banIP-1.0.0-5[145269]: domain lookup finished in 0m 0s (blocklist, 0 domains, 0 IPs)
Sat Aug  3 04:51:11 2024 user.info banIP-1.0.0-5[145269]: domain lookup finished in 0m 0s (allowlist, 0 domains, 0 IPs)
Sat Aug  3 04:51:11 2024 user.info banIP-1.0.0-5[145269]: start detached banIP log service (/sbin/logread)
Sat Aug  3 05:15:00 2024 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 172.16.0.11 04:68:65:f1:d8:f9
  ```
</details>

vincejv, 2024-08-16:
<details>
  <summary>Logs when DHCPv6 is successful</summary>
  
  ```
Sat Aug  3 09:30:20 2024 daemon.info pppd[7171]: Plugin pppoe.so loaded.
Sat Aug  3 09:30:20 2024 daemon.info pppd[7171]: PPPoE plugin from pppd 2.4.9
Sat Aug  3 09:30:20 2024 daemon.notice netifd: wanlte (7150): udhcpc: broadcasting discover
Sat Aug  3 09:30:20 2024 daemon.notice netifd: wanb (7147): udhcpc: broadcasting discover
Sat Aug  3 09:30:20 2024 daemon.notice netifd: mtkgwaccess (7155): udhcpc: broadcasting discover
Sat Aug  3 09:30:20 2024 daemon.err modprobe: failed to find a module named aufs
Sat Aug  3 09:30:20 2024 daemon.err dockerd[6427]: time="2024-08-03T01:30:20.176082044Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sat Aug  3 09:30:20 2024 daemon.err dockerd[6427]: time="2024-08-03T01:30:20.190526426Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sat Aug  3 09:30:20 2024 daemon.info ipsec: 12[KNL] fe80::a2f3:c1ff:fe03:7301 appeared on br-lan
Sat Aug  3 09:30:20 2024 daemon.notice pppd[7171]: pppd 2.4.9 started by root, uid 0
Sat Aug  3 09:30:20 2024 daemon.debug pppd[7171]: Send PPPOE Discovery V1T1 PADI session 0x0 length 18
Sat Aug  3 09:30:20 2024 daemon.debug pppd[7171]:  dst ff:ff:ff:ff:ff:ff  src 5c:f9:dd:6f:79:b6
Sat Aug  3 09:30:20 2024 daemon.debug pppd[7171]:  [service-name] [host-uniq  03 1c 00 00] [PPP-max-payload  05 dc]
Sat Aug  3 09:30:20 2024 daemon.info ipsec: 13[NET] using forecast interface br-lan
Sat Aug  3 09:30:20 2024 daemon.info ipsec: 13[CFG] joining forecast multicast groups: 224.0.0.1,224.0.0.22,224.0.0.251,224.0.0.252,239.255.255.250
Sat Aug  3 09:30:20 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sat Aug  3 09:30:20 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 6 names
Sat Aug  3 09:30:20 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c.5840 - 6 names
Sat Aug  3 09:30:20 2024 daemon.err odhcp6c[7152]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sat Aug  3 09:30:20 2024 daemon.info ipsec: 05[KNL] flags changed for fd46:5d36:7a68::1 on br-lan
Sat Aug  3 09:30:20 2024 daemon.info ipsec: 10[NET] using forecast interface br-lan
Sat Aug  3 09:30:20 2024 daemon.info ipsec: 10[CFG] joining forecast multicast groups: 224.0.0.1,224.0.0.22,224.0.0.251,224.0.0.252,239.255.255.250
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Joining mDNS multicast group on interface eth0.1050.IPv6 with address fe80::5ef9:ddff:fe6f:79b6.
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 12[KNL] fe80::5ef9:ddff:fe6f:79b6 appeared on eth0.1050
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: New relevant interface eth0.1050.IPv6 for mDNS.
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Registering new address record for fe80::5ef9:ddff:fe6f:79b6 on eth0.1050.*.
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 13[NET] using forecast interface br-lan
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 13[CFG] joining forecast multicast groups: 224.0.0.1,224.0.0.22,224.0.0.251,224.0.0.252,239.255.255.250
Sat Aug  3 09:30:21 2024 daemon.err odhcp6c[7152]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Joining mDNS multicast group on interface eth0.1020.IPv6 with address fe80::5ef9:ddff:fe6f:79b6.
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 12[KNL] fe80::5ef9:ddff:fe6f:79b6 appeared on eth0.1020
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: New relevant interface eth0.1020.IPv6 for mDNS.
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Registering new address record for fe80::5ef9:ddff:fe6f:79b6 on eth0.1020.*.
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Joining mDNS multicast group on interface eth0.1030.IPv6 with address fe80::5ef9:ddff:fe6f:79b6.
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 11[KNL] fe80::5ef9:ddff:fe6f:79b6 appeared on eth0.1030
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: New relevant interface eth0.1030.IPv6 for mDNS.
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Registering new address record for fe80::5ef9:ddff:fe6f:79b6 on eth0.1030.*.
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 13[NET] using forecast interface br-lan
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 13[CFG] joining forecast multicast groups: 224.0.0.1,224.0.0.22,224.0.0.251,224.0.0.252,239.255.255.250
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Joining mDNS multicast group on interface eth0.2000.IPv6 with address fe80::5ef9:ddff:fe6f:79b6.
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 15[KNL] fe80::5ef9:ddff:fe6f:79b6 appeared on eth0.2000
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: New relevant interface eth0.2000.IPv6 for mDNS.
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Registering new address record for fe80::5ef9:ddff:fe6f:79b6 on eth0.2000.*.
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::5ef9:ddff:fe6f:79b6.
Sat Aug  3 09:30:21 2024 daemon.info ipsec: 12[KNL] fe80::5ef9:ddff:fe6f:79b6 appeared on eth0
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: New relevant interface eth0.IPv6 for mDNS.
Sat Aug  3 09:30:21 2024 daemon.info avahi-daemon[4756]: Registering new address record for fe80::5ef9:ddff:fe6f:79b6 on eth0.*.
[... cut ...]
Sat Aug  3 09:30:29 2024 user.notice nft-qos-dynamic: ACTION=update, MACADDR=02:81:d3:dc:19:2c, IPADDR=172.16.0.3, HOSTNAME=orangepione
Sat Aug  3 09:30:29 2024 daemon.info ipsec: 14[KNL] interface veth5a7c35f activated
Sat Aug  3 09:30:29 2024 user.notice nft-qos-monitor: ACTION=update, MACADDR=04:68:65:f1:d8:f9, IPADDR=172.16.0.11, HOSTNAME=Fornax
Sat Aug  3 09:30:29 2024 user.notice nft-qos-dynamic: ACTION=update, MACADDR=1c:d1:07:ce:e9:93, IPADDR=172.16.0.52, HOSTNAME=realme-8-5G
  ```
</details>

vincejv, 2024-08-16:
Abnormal operation:
[image]

Normal operation (IPv6 is okay):
[image]

e9hack, 2024-08-16:
How did you configure wan6? Usually /lib/netifd/ppp6-up generates the device wan_6.

vincejv, 2024-08-16:
@e9hack thanks for the reply, i posted the `/etc/config/network` in additional info section, it's in the accordion/colapsible, just click on it

ppp6-up is stock, didn't modify it

<details>
  <summary>/lib/netifd/ppp6-up</summary>

```
#!/bin/sh
PPP_IPPARAM="$6"

. /lib/netifd/netifd-proto.sh
proto_init_update "$IFNAME" 1 1
proto_set_keep 1
[ -n "$PPP_IPPARAM" ] && {
	[ -n "$LLLOCAL" ] && proto_add_ipv6_address "$LLLOCAL" 128
}
proto_send_update "$PPP_IPPARAM"

[ -d /etc/ppp/ip-up.d ] && {
	for SCRIPT in /etc/ppp/ip-up.d/*
	do
		[ -x "$SCRIPT" ] && "$SCRIPT" "$@"
	done
}

if [ -n "$AUTOIPV6" ]; then
	ZONE=$(fw3 -q network $PPP_IPPARAM 2>/dev/null)

	json_init
	json_add_string name "${PPP_IPPARAM}_6"
	json_add_string ifname "@$PPP_IPPARAM"
	json_add_string proto "dhcpv6"
	[ -n "$ZONE" ] && json_add_string zone "$ZONE"
	[ -n "$EXTENDPREFIX" ] && json_add_string extendprefix 1
	[ -n "$IP6TABLE" ] && json_add_string ip6table $IP6TABLE
	[ -n "$PEERDNS" ] && json_add_boolean peerdns $PEERDNS
	json_close_object
	ubus call network add_dynamic "$(json_dump)"
fi
```
</details>

e9hack, 2024-08-16:
Usually you don't need any entry for wan6/wan_6 in /etc/config/network or /etc/config/firewall

This part generates the entries. ```${PPP_IPPARAM}``` is set to wan, which results in a wan_6 device. 
```
if [ -n "$AUTOIPV6" ]; then
	ZONE=$(fw3 -q network $PPP_IPPARAM 2>/dev/null)

	json_init
	json_add_string name "${PPP_IPPARAM}_6"
	json_add_string ifname "@$PPP_IPPARAM"
	json_add_string proto "dhcpv6"
	[ -n "$ZONE" ] && json_add_string zone "$ZONE"
	[ -n "$EXTENDPREFIX" ] && json_add_string extendprefix 1
	[ -n "$IP6TABLE" ] && json_add_string ip6table $IP6TABLE
	[ -n "$PEERDNS" ] && json_add_boolean peerdns $PEERDNS
	json_close_object
	ubus call network add_dynamic "$(json_dump)"
fi
```

I don't know if there is a conflict if wan6 exists too. It looks like the script must be update to fw4.

vincejv, 2024-08-16:
i see, i'm not sure if the solution you are proposing is the same as this: https://forum.openwrt.org/t/ways-to-check-if-ipv6-issues-are-upstream-isp-or-not/207428/8

[image]

but this solved my issue, use an alias interface instead of the actual pppoe interface for the dhcpv6 client, even if i restart the pppoe session, it can retrieve ip addresses now

closing, thanks for your help!

closed 2024-08-16 by vincejv

d-navikas, 2024-12-25:
@vincejv can you tell please, what the difference? I'm not OpenWrt power user.