openwrt/openwrt #16183: qualcommax: ipq60xx: ipq807x: fap650: capture radio data in monitor mode

legale, 2024-08-16:
### Describe the bug

When capturing raw frames in monitor mode on a Qualcomm IPQ6018-based device (Yuncore FAP650 [https://github.com/openwrt/openwrt/pull/14745 2](https://github.com/openwrt/openwrt/pull/14745)) or another ipq6018, ipq8072 device using the ath11k driver, the captured radiotap frames contain an extra 8 bytes of junk data.
I have tried the same on another qualcommax device (yuncore ax880 [https://github.com/openwrt/openwrt/pull/13853 1](https://github.com/openwrt/openwrt/pull/13853)). The same problem.

### OpenWrt version

r27140-ccc06f6716

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq60xx

### Device

Yuncore FAP650

### Image kind

Self-built image

### Steps to reproduce

```sh
iw phy0 interface add wlan0 type monitor
ip link set wlan0 up
iw wlan0 set channel 36
tcpdump -i wlan0 -w /tmp/radio.pcap
```

### Actual behaviour

A monitor interface is created and brought up, the channel switches to 36, and radio traffic capture begins. After capturing a certain amount of packets, I load the saved /tmp/radio.pcap file into Wireshark for analysis. In the screenshot below (Screenshot 1), it is evident that there are unrecognized data at the very end of the frame.

Screenshot 1 - original pcap (malformed data)
[image]

If I manually remove 8 bytes from the end of each frame and adjust the frame size in the header accordingly, the captured frames become correct. Screenshot 2 is attached.
Screenshot 2 - fixed pcap
[image]

### Expected behaviour

Undamaged captured frames.

### Additional info

I have tried the same  on another qualcommax ipq807x device (yuncore ax880 [https://github.com/openwrt/openwrt/pull/13853 1](https://github.com/openwrt/openwrt/pull/13853)). The same  behaviour (malformed pcap).

### Diffconfig

```text
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq60xx=y
CONFIG_TARGET_qualcommax_ipq60xx_DEVICE_yuncore_fap650=y
CONFIG_IMAGEOPT=y
CONFIG_LLDPD_WITH_CDP=y
CONFIG_LLDPD_WITH_CUSTOM=y
CONFIG_LLDPD_WITH_DOT1=y
CONFIG_LLDPD_WITH_DOT3=y
CONFIG_LLDPD_WITH_EDP=y
CONFIG_LLDPD_WITH_FDP=y
CONFIG_LLDPD_WITH_LLDPMED=y
CONFIG_LLDPD_WITH_PRIVSEP=y
CONFIG_LLDPD_WITH_SONMP=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_atftp=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_ebtables-legacy=y
CONFIG_PACKAGE_ebtables-legacy-utils=y
# CONFIG_PACKAGE_firewall4 is not set
CONFIG_PACKAGE_gdb=m
CONFIG_PACKAGE_gre=y
CONFIG_PACKAGE_hostapd-utils=m
CONFIG_PACKAGE_iftop=m
CONFIG_PACKAGE_ip-bridge=m
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_ip6tables-extra=y
CONFIG_PACKAGE_ip6tables-mod-nat=y
CONFIG_PACKAGE_iperf=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_iptables-mod-extra=y
CONFIG_PACKAGE_iptables-mod-filter=y
CONFIG_PACKAGE_iptables-mod-nat-extra=y
CONFIG_PACKAGE_iptables-zz-legacy=y
CONFIG_PACKAGE_iputils-ping=m
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-crypto-md5=m
CONFIG_PACKAGE_kmod-ebtables=y
CONFIG_PACKAGE_kmod-ebtables-ipv4=y
CONFIG_PACKAGE_kmod-ebtables-ipv6=y
CONFIG_PACKAGE_kmod-ebtables-watchers=y
CONFIG_PACKAGE_kmod-gre=y
CONFIG_PACKAGE_kmod-gre6=y
CONFIG_PACKAGE_kmod-ip6-tunnel=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ip6tables-extra=y
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-extra=y
CONFIG_PACKAGE_kmod-ipt-filter=y
CONFIG_PACKAGE_kmod-ipt-nat=y
CONFIG_PACKAGE_kmod-ipt-nat-extra=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_kmod-ipt-raw=y
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-iptunnel6=y
CONFIG_PACKAGE_kmod-lib-textsearch=y
CONFIG_PACKAGE_kmod-mtd-rw=m
CONFIG_PACKAGE_kmod-netlink-diag=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-vxlan=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_less=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcares=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libev=y
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libffi=y
CONFIG_PACKAGE_libgdbm=y
CONFIG_PACKAGE_libgmp=m
CONFIG_PACKAGE_libip4tc=y
CONFIG_PACKAGE_libip6tc=y
CONFIG_PACKAGE_libiperf3=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_liblzma=y
CONFIG_PACKAGE_libmosquitto-ssl=y
CONFIG_PACKAGE_libmpfr=m
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libopenssl-conf=y
CONFIG_PACKAGE_libopenssl-legacy=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpopt=y
CONFIG_PACKAGE_libpython3=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libsqlite3=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libtirpc=m
CONFIG_PACKAGE_libwolfssl=m
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lldpd=y
CONFIG_PACKAGE_lsof=m
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-uqr=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-gre=y
CONFIG_PACKAGE_luci-proto-vxlan=y
CONFIG_PACKAGE_luci-proto-wireguard=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_nano=y
# CONFIG_PACKAGE_nftables-json is not set
CONFIG_PACKAGE_openssl-util=m
CONFIG_PACKAGE_python3=y
CONFIG_PACKAGE_python3-asyncio=y
CONFIG_PACKAGE_python3-base=y
CONFIG_PACKAGE_python3-cgi=y
CONFIG_PACKAGE_python3-cgitb=y
CONFIG_PACKAGE_python3-codecs=y
CONFIG_PACKAGE_python3-ctypes=y
CONFIG_PACKAGE_python3-dbm=y
CONFIG_PACKAGE_python3-decimal=y
CONFIG_PACKAGE_python3-distutils=y
CONFIG_PACKAGE_python3-email=y
CONFIG_PACKAGE_python3-light=y
CONFIG_PACKAGE_python3-logging=y
CONFIG_PACKAGE_python3-lzma=y
CONFIG_PACKAGE_python3-multiprocessing=y
CONFIG_PACKAGE_python3-ncurses=y
CONFIG_PACKAGE_python3-openssl=y
CONFIG_PACKAGE_python3-pip=y
CONFIG_PACKAGE_python3-pydoc=y
CONFIG_PACKAGE_python3-readline=y
CONFIG_PACKAGE_python3-sqlite3=y
CONFIG_PACKAGE_python3-unittest=y
CONFIG_PACKAGE_python3-urllib=y
CONFIG_PACKAGE_python3-uuid=y
CONFIG_PACKAGE_python3-xml=y
CONFIG_PACKAGE_resolveip=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_rsync=y
CONFIG_PACKAGE_rtty-openssl=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_ss=y
CONFIG_PACKAGE_strace=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_usteer=y
CONFIG_PACKAGE_valgrind=m
CONFIG_PACKAGE_vnstat=m
CONFIG_PACKAGE_vxlan=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_wpa-cli=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-openssl=y
CONFIG_PACKAGE_xtables-legacy=y
CONFIG_PACKAGE_zlib=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE=""
CONFIG_VERSION_DIST="OpenWrt"
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER=""
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_WOLFSSL_HAS_NO_HW=y
CONFIG_PACKAGE_jansson=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
CONFIG_PACKAGE_kmod-nft-fib=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libnftnl=y
# CONFIG_VERSION_CODE_FILENAMES is not set
# CONFIG_VERSION_FILENAMES is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-08-27:
@legale: To follow this ticket

zhouruixi, 2024-09-13:
same issue with Xiaomi AX3600 https://github.com/svpcom/wfb-ng/issues/365

legale, 2024-09-13:
> same issue with Xiaomi AX3600 [svpcom/wfb-ng#365](https://github.com/svpcom/wfb-ng/issues/365)

Any ideas? My only idea at the moment is to patch kernel part driver to fix incorrect packet here. But it's ugly hack.

zhouruixi, 2024-09-16:
> > same issue with Xiaomi AX3600 [svpcom/wfb-ng#365](https://github.com/svpcom/wfb-ng/issues/365)
> 
> Any ideas? My only idea at the moment is to patch kernel part driver to fix incorrect packet here. But it's ugly hack. 

sorry no, I am not a programer.

Neustradamus, 2024-12-28:
@legale: Can you edit your description, the 2 URLs are not good and have a 2 or a 1 in the link names.

equid0x, 2025-11-29:
I wonder if its a QinQ VLAN header from the succeeding packet being incorrectly captured with the current packet? Might explain some problems with ath11k and rx-gro-list which combines and splits packets and also the DSA multicast issues where tagged traffic is not receiving multicast frames in all instances....