openwrt/openwrt #16184: WRT1900 / WRT1200AC v2 softoffload does not work

ja16779, 2024-08-16:
### Describe the bug

I have a par of linksys router wrt1900 / wrt 1200 and the softoffloading does not work, when enable the pages does not open, ever the ping answer

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900AC v2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

anomeome, 2024-08-17:
Forum [thread](https://forum.openwrt.org/t/wrt1200ac-does-not-work-softoffloading/207507) with WIP link which is the apparent resolution.

brada4, 2024-08-17:
In the forum thread @jow- was mentioned as knowledgeable on subject. Essence - when dsa user ports are excluded from soft offload abusing latest fw4 code - offloaded forwarding starts to happen. Which avoids and softly points to a bug or shortage in particular DSA driver set.

brada4, 2024-08-17:
Steps to reproduce:
enable firewall soft offload
Actual behaviour:
Forwarding stops
Expected behaviour:
Accelerated forwarding happens

brada4, 2024-08-19:
You can avoid suspected buggy DSA driver using latest firewall4 code
https://forum.openwrt.org/t/wrt1200ac-does-not-work-softoffloading/207507/12?u=brada4
I.e. offload discharges to bridge interface additionally keeping LAN ARP and wifi roaming undisturbed by itself, still doing heaviest packet re-addressing part fast.

closed 2024-12-25 by ja16779