openwrt/openwrt #16191: PWM fan - rpi5

AQPSal, 2024-08-17:
### Describe the bug

Since this PR [https://github.com/openwrt/openwrt/pull/16104] my rpi5 fan has been going full blast. It was resolved by manually installing kmod-hwmon-pwmfan, which was not included in the image as it depends on kmod-thermal. The solution may be to remove kmod-thermal as a dependency to pwmfan kmod.

### OpenWrt version

r27143-0b7d99147b

### OpenWrt release

snapshot

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-18:
It is up to you to install drivers for addon hardware.
e.g usb port driver is included but you install driver depending on whether printer modem or  webcam is connected.

AQPSal, 2024-08-18:
> It is up to you to install drivers for addon hardware. e.g usb port driver is included but you install driver depending on whether printer modem or webcam is connected.

I would suggest you review  the PR - the pwmfan kmod is part  of the image configuration for the rpi5. The PR accidentally removes it by unconfiguring the thermal kmod.

Noltari, 2024-08-20:
I've reverted https://github.com/openwrt/openwrt/pull/16104

closed 2024-08-20 by Noltari