openwrt/openwrt #16192: RPI-5 memory leak

AQPSal, 2024-08-17:
### Describe the bug

It takes about 48 hours to use all of my rpi-5 4GB and router reboots. Log is lost upon reboot.

I tried to identify the application causing the leak but I am not knowledgeable enough to figure it out. While I only noticed this week, it has been happening since June (must have updated the router with a new snapshot around then). Please let me know what information you would need to diagnose.

[image]

### OpenWrt version

r27143-0b7d99147b

### OpenWrt release

snapshot

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-18:
Basic router functions fit into 64MB RAM
Forum is a better place to dismantle heap if packages. Graph looks like garbage-collecting java (not avail for openwrt but may be present in container) or a memory leak yielding widely logged crash

AQPSal, 2024-08-18:
> Basic router functions fit into 64MB RAM Forum is a better place to dismantle heap if packages. Graph looks like garbage-collecting java (not avail for openwrt but may be present in container) or a memory leak yielding widely logged crash

Its an image of the collectd memory tab showing the progressive use of memory. When rebooted, the system uses less than 100MBs. PS AUX, htop does not reveal any culprits

brada4, 2024-08-18:
>  identify the application causing the leak but I am not knowledgeable enough to figure it

Bug tracker is not a support forum. Here you have to clearly identify bug source.

AQPSal, 2024-08-18:
> > identify the application causing the leak but I am not knowledgeable enough to figure it
> 
> Bug tracker is not a support forum. Here you have to clearly identify bug source.

I see your point. I'll bring it up in the forum. Thx.

Noltari, 2024-08-20:
@AQPSal my RPi 5 (8G) has been running 24/7 for the past months without this issue, so it must be specific to any of the packages that you have installed on your device.
Therefore, I don't think that this is issue is related to the RPi 5.

brada4, 2024-08-20:
Xref: https://forum.openwrt.org/t/continuous-ram-memory-consumption-increase-over-time/207942
brief: agh glitch

closed 2024-08-20 by Noltari

AQPSal, 2024-08-26:
In my case I found the issue seems to be caused by collectd (by way of installing programs one by one) - I am not sure which plugin or if it is the app itself. This repo may not be the ideal place report this; nevertheless, leaving the information here just in case.