openwrt/openwrt #16195: No mesh connection SNAPSHOT r27160

vwghub, 2024-08-18:
### Describe the bug

After updating my two GL-iNet MT-6000 devices, a mesh connection could not be established.  

### OpenWrt version

SNAPSHOT (r27160-b72c4b5386)

### OpenWrt release

SNAPSHOT (r27160-b72c4b5386)

### OpenWrt target/subtarget

SNAPSHOT (r27160-b72c4b5386)

### Device

GL-iNet MT-6000

### Image kind

Official downloaded image

### Steps to reproduce

Main router used to connect to Internet with PPPoE and a secondary mesh router on another floor to provide connectivity there.  Using built in mesh with wpad-mbedtls WED disabled.

### Actual behaviour

I can connect by wi-fi to each router but the two routers cannot see each other.  The router overview page shows a mesh connection at 20MHz, 6.0Mbp.  Connecting to the mesh router results in no TCP/IP configuration as that's provided from the main router. 

### Expected behaviour

Mesh connectivity between the two routers.  This works in SNAPSHOT r27154.  

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nbd168, 2024-08-18:
Please try the latest version, should be fixed in 8d257238654ff9f236ef7978dea0f5a010df4f3e

nbd168, 2024-08-20:
Please try this patch on top of latest mt76 master: https://nbd.name/p/058a61cf

nbd168, 2024-08-21:
Scratch that, a better fix is in latest OpenWrt master

closed 2024-12-12 by blogic