openwrt/openwrt #16200: 23.05.4 failed to build x86/64 using imagebuilder.   opkg_install_pkg: Checksum or size mismatch for package busybox.

RandomTask-1, 2024-08-19:
### Describe the bug

Attempting to build using the 23.05.4 imagebuilder package for x86/64 fails.

### OpenWrt version

openwrt-imagebuilder-23.05.3-x86-64.Linux-x86_64.tar.xz

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

x86/64

### Device

mini pc

### Image kind

Self-built image

### Steps to reproduce

Setup Build machine
Download and extract https://downloads.openwrt.org/releases/23.05.3/targets/x86/64/openwrt-imagebuilder-23.05.3-x86-64.Linux-x86_64.tar.xz
Execute make image PROFILE="generic"

### Actual behaviour

Build fails with the below errors:
Configuring ppp-mod-pppoe.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package busybox. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package busybox.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:144: _call_image] Error 2
make: *** [Makefile:262: image] Error 2

### Expected behaviour

Completed build.  

### Additional info

23.05.3 does work as expected, this appears to be a 23.05.4 issue.

The below is the output from two consecutive build attempts:
~/openwrt-imagebuilder-23.05.4-x86-64.Linux-x86_64$ make image PROFILE="generic"
Generate local signing keys...
Generate local certificate...
Checking 'true'... ok.
Checking 'false'... ok.
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
[... 816 lines cut ...]
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:144: _call_image] Error 2
make: *** [Makefile:262: image] Error 2

### Diffconfig

```text
can't find file Config.in
can't open file 'tmp/.diffconfig.stage1' at ./scripts/kconfig.pl line 32.
can't find file Config.in
can't open file 'tmp/.diffconfig.stage2' at ./scripts/kconfig.pl line 32.
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_x86_64_DEVICE_generic=y
CONFIG_TARGET_DEVICE_PACKAGES_x86_64_DEVICE_generic=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

amanjuman, 2024-08-28:
Having the same issue here.

DDPF02, 2024-09-01:
Same issue!

jkaetz, 2024-10-01:
This appears to be resolved in 23.05.5

ynezz, 2024-10-02:
It might be occasional Fastly CDN issue, would be nice to get more details next time, see https://github.com/openwrt/openwrt/issues/16511#issuecomment-2387643680

closed 2024-10-02 by ynezz