openwrt/openwrt #16210: wireguard config issue when multiple peers have '0.0.0.0/0' in allowed_ips

ttoocs, 2024-08-21:
### Describe the bug

I've been working on getting wireguard up and running, and messing around, at least one of the issues is that the my  Openwrt that I'm using as a wireguard server seems to still handshake, but ignore packets,  But it only happened after adding a peer to the wire guard interface... Like it's getting blocked.. 

Sure enough: `allowed ips: (none)`
[image]

In further testing, it seems that it is related to the '0.0.0.0/0' pass-through ip:
* If a peer has '0.0.0.0/24' in allowed_ips
  * 
  * I've also tried multiple examples with non-'0.0.0.0/0' ip's that behave just fine (they do, even in the middle of the rest)
* Normal 'allowed_ips' subnets masks work just fine, multiple are permitted and repeat values don't cause issues.

I'm also trying to trace this down myself in the code... I'm picking this package as the closest thing I can find to it is the wireguard.sh script. I took a quick look at that wireguard.sh, but I'm not really sure where it gets called from / it's arguments. But it does look like it's tied to generating configuration for wireguard interfaces.

I am able to manually correct it (till next reboot / stop->restart of the interface) with issuing the correct `wg set wg0 peer ${PeerPubkey} allowed-ips '0.0.0.0/0'` commands, and the wire guard interface then works for forwarding data to any ip.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

32.05.3

### OpenWrt target/subtarget

ath79/genric

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Steps to reproduce

Installed dependencies by the luci-software manager for luci-wireguard
Then reboot ( otherwise the interface wireguard type won't show)
In luci in network -> add Interface, add a new wireguard interface, - generate a key
Then in the peers tab, add two peers with:
* A valid pubkey value (can be generated nearby... else `wg genkey | wg pubkey` should generate a valid pubkey.. the issue is visible without ever having the other peer exist).
* Allowed-ip of  '0.0.0.0/0'
Then save / apply,
On on that page you stop->restart (restart by itself doesn't seem to reload these) or reboot
Once it's done,
You can check with `wg` command in shell/ssh:
* Check to see any peers seem to have their allowed_ips go missing or unlisted or '(none)'
  * if the peer isn't the bottom-most '0.0.0.0/0' entry in /etc/config/network [This Bug] would be that they have their 'allowed_ips' go missing.

### Actual behaviour

wireguard config issue when multiple peers have '0.0.0.0/0' in allowed_ips

### Expected behaviour

Correctly apply the allowed_ips '0.0.0.0/0' to each of the wireguard peers

### Additional info

Another similar edge-case could be '::/0' , and it's also included in various wire guard guides for routing all traffic.  - I haven't tested

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ttoocs, 2024-08-21:
```
root@TPLinkC7:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''
```
Sorry, the ath79/genric was missing an e: 'ath79/generic'

Neustradamus, 2024-08-21:
@ttoocs: I invite you to edit the main description ;)

gadall, 2024-08-21:
@ttoocs WireGuard uses the Allowed IPs lists for incoming packets, given the source address, to determine whether the incoming packet should be allowed. It also uses them as a sort of a routing table for outgoing packets. Once a packet enters the interface, this is how WireGuard decides which peer the packet should be sent to.

Like any routing table, an overlapping but more specific prefix prevails. For example one peer can have 10.0.0.0/8 and the other peer can have 10.0.1.0/24 and it should work as expected. But if you try to assign identical prefixes to two peers, when assigning the prefix to the second peer it would be removed from the first peer. You can't have the same prefix on two peers at the same time.

This is what your screenshot appears to show: all peers got 0.0.0.0/0 successfully, but only the last one stuck.

This is how wireguard itself works, the actual kernel module. And the wg command doesn't report an error when configuring a duplicate prefix.

Perhaps what would be desirable here is to have the graphical interface check for such a condition.

The only way you can have multiple peers with 0.0.0.0/0 is to have a dedicated wireguard interface for each. You'd then of course have to use routing tables / rules to select the interface to route to.

jow-, 2024-08-21:
Not a bug, configuration issue.

closed 2024-08-21 by jow-

gadall, 2024-08-21:
@jow- At least as a wishlist / feature request, it is a reasonable expectation that the interface should check for and report invalid configurations, or cases where the configuration failed to apply, in this case both.

jow-, 2024-08-21:
Sure it is reasonable, and patches are welcome. But I don't foresee any LuCI maintainer working to implement that.

ttoocs, 2024-08-21:
Ah, I see how `wg` is itself also switching allowed-ips to match what @gadall said. A specific wireguard interface does seems to want unique allowed-ips. (I guess I didn't notice it 'clear' other peers as it is being set)

Will try setting up multiple interfaces as an alternative. Thank you