openwrt/openwrt #16214: a266230 is breaking mediatek builds

LuisMitaHL, 2024-08-21:
### Describe the bug

Hi,
a2662309aae1655966d7d8f31b71ddc6edbede87 is breaking Mediatek builds.
Reverting the commit fix the problem

### OpenWrt version

a2662309aae1655966d7d8f31b71ddc6edbede87

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Cudy TR3000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-08-21:
See https://github.com/openwrt/openwrt/pull/16211

robimarko, 2024-08-22:
Fixed via:
https://github.com/openwrt/openwrt/commit/963be1423f64156fa66a8a94cf0207ce2945bf34

closed 2024-08-22 by robimarko