openwrt/openwrt #16228: Filesystem changes do not persist across reboot on 23.05.4

Juul, 2024-08-22:
### Describe the bug

I flashed the [23.05.4 NanoBeam M5 XW factory image](https://downloads.openwrt.org/releases/23.05.4/targets/ath79/generic/openwrt-23.05.4-ath79-generic-ubnt_nanobeam-m5-xw-squashfs-factory.bin) using TFTP. It booted correctly but all configuration changes disappeared on reboot. I attempted to reflash with the same image and the behavior persisted. 

I then flashed the 22.03.7 NanoBeam M5 XW factory image and this worked perfectly, having no problem persisting filesystem changes across reboots.

### OpenWrt version

Not sure, didn't record this but it was 23.05.4 downloaded from the link in the description on August 19th 2024

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti NanoBeam M5 (XW)

### Image kind

Official downloaded image

### Steps to reproduce

Flash image. Boot. Change files in e.g. /etc/config. Reboot.

### Actual behaviour

Filesystem changes are reset to defaults on reboot.

### Expected behaviour

Filesystem changes should persist across reboots

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-22:
Ols show content of /proc/mtd

Juul, 2024-08-27:
> Ols show content of /proc/mtd

Here is the contents on 22.03.7.

Let me know if you need me to repeat the same from 23.05.04

```
# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00760000 00010000 "firmware"
mtd3: 00230000 00010000 "kernel"
mtd4: 00530000 00010000 "rootfs"
mtd5: 00190000 00010000 "rootfs_data"
mtd6: 00040000 00010000 "cfg"
mtd7: 00010000 00010000 "art"
```

closed 2024-09-12 by Juul