openwrt/openwrt #16233: bind-server is missing dependency on bind-rndc

krei-se, 2024-08-23:
### Describe the bug

When installing bind-server via LuCi you get the error, that rndc-confgen failed as the dependency on bind-rndc is missing.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Steps to reproduce

install bind-server via opkg or LuCI

### Actual behaviour

rndc-confgen fails

### Expected behaviour

rndc-confgen should work

### Additional info

Dependency on bind-libs is in bind-server and bind-rndc but i had to install it manually. The package-install via LuCI seems broken sometimes?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-08-23:
Yes, there is no nred to keep blob on the disk while server is running. @pprindeville

krei-se, 2024-08-25:
Thanks for getting back at me @brada4  I see now. Well i will close this issue then, i guess who installs this will know / figure it out :)

closed 2024-08-25 by krei-se