openwrt/openwrt #16234: Snapshot r27167-5d2a008670 - x86/64 i915 not loading properly

acordtag, 2024-08-23:
### Describe the bug

Using 23.05.5 there is a video output - Intel i915 drm
`
kern.warn kernel: [    3.177362] i915 0000:00:02.0: Direct firmware load for i915/tgl_dmc_ver2_12.bin failed with error -2
kern.warn kernel: [    3.186598] i915 0000:00:02.0: Falling back to sysfs fallback for: i915/tgl_dmc_ver2_12.bin
kern.info kernel: [    3.846508] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
kern.info kernel: [    3.872876] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
...
kern.info kernel: [    5.365329] fbcon: i915drmfb (fb0) is primary device
kern.info kernel: [    5.427469] Console: switching to colour frame buffer device 160x45
kern.info kernel: [    5.455417] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
...
kern.notice kernel: [   65.747438] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/tgl_dmc_ver2_12.bin. Disabling runtime power management.
`

Using SNAPSHOT there is NO video output
`
kern.notice kernel: [    3.656814] i915 0000:00:02.0: [drm] Transparent Hugepage support is recommended for optimal performance on this platform!
kern.warn kernel: [    3.670967] i915 0000:00:02.0: Direct firmware load for i915/tgl_dmc_ver2_12.bin failed with error -2
kern.warn kernel: [    3.680219] i915 0000:00:02.0: Falling back to sysfs fallback for: i915/tgl_dmc_ver2_12.bin
kern.info kernel: [    4.474047] i915 0000:00:02.0: [drm] [ENCODER:307:DDI A/PHY A] failed to retrieve link info, disabling eDP
kern.info kernel: [    4.512803] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
...
kern.info kernel: [    5.369546] i915 0000:00:02.0: [drm] fb1: i915drmfb frame buffer device
kern.notice kernel: [   66.079162] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/tgl_dmc_ver2_12.bin. Disabling runtime power management.
`

this seems to be very related to this forum post about rebuilding `dmc.cpio` using working compiled `glk_dmc_ver1_xx.bin`
https://forum.openwrt.org/t/silly-little-i915-issue-on-x86/150405/22?page=2

### OpenWrt version

SNAPSHOT

### OpenWrt release

r27167-5d2a008670

### OpenWrt target/subtarget

x86

### Device

x86_64

### Image kind

Self-built image

### Steps to reproduce

intsall SNAPSHOT Kernel 6.6 on x64 box with intel i915 video drivers

### Expected behaviour

should work as stable 23.05.4

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

joez, 2024-09-03:
@acordtag try #16276 out, the PR is definitly for your issue
Use i915 driver as a kernel module, so that initramfs is not required, and it can also save some space from the kernel image comparing being a built-in driver

acordtag, 2024-09-03:
Thank you. Could you explain a bit if this PR goes ahead what would be the kmod name we need to use to build the image?
i915-firmware or drm-i915 

this one below is if we want to use igpu video encoding, etc?
intel-igpu-firmware-dmc

Thank you.

joez, 2024-09-06:
just enable drm-i915, then all the dependencies will be selected automatically, expecially the i915-firmware.
intel-igpu-firmware-dmc is for finer adjustments, it only installs dmc part of i915-firmware, but in most cases installing all the i915 firmware should make your life easier.