openwrt/openwrt #16237: Builds on Xiaomi AX6S/AX3200 seem to fail (maybe even MT7622)

HW-OLE, 2024-08-23:
### Describe the bug

When wanting to build for MT7622 I always get the same error regardless of the settings. Maybe this has something to do with the fact that I was trying to ingest proprietary MTK drivers but I've also tried 4 versions of those now and it always lands on a Error 

### OpenWrt version

multiple, dont know where to see when building

### OpenWrt release

23.05.3 and SNAPSHOT is what I've tried

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Self-built image

### Steps to reproduce

`git clone https://git01.mediatek.com/openwrt/feeds/mtk-openwrt-feeds` 
`git clone https://git.openwrt.org/openwrt/openwrt.git` 

`cd openwrt`
`echo "src-git mtk_openwrt_feed https://git01.mediatek.com/openwrt/feeds/mtk-openwrt-feeds" >> feeds.conf.default`
`./scripts/feeds update -a`
`./scripts/feeds install -a`

`cp -af ./feeds/mtk_openwrt_feed/master/files/* .`
`for file in $(find ./feeds/mtk_openwrt_feed/master/patches-base -name "*.patch" | sort); do patch -f -p1 -i ${file}; done`

`make menuconfig` (Selecting Target and subtarget (Mediatek RALINK ARM and mt7622 and Xiaomi AX6S as the device

`make V=s -j4` 

Also tried with -j1) All according to the official mediatek documentation [here](https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/)

At first I tried the instructions from bricco1981/@anon4457646 's Repo [here](https://github.com/bricco1981/MT7622-mtkwifi) but that also fails. I think there might be an issue with buidling atm. Some user in the forum also said there was a issue that was fixed yesterday evening/today's morning. But after trying to build after that it still failed.

### Actual behaviour

The build fails every time. The last time I tried it ended with the Errors below. One Error that carries from build to build is the toplevel.mk Error. That one happened on every single build I did now. The other ones happened only today, so no clue what those are about. But from what I see the kernel fails to compile, so it seems like a very serious error

`make[7]: *** [scripts/Makefile.vmlinux:37: vmlinux] Error 1`
`make[6]: *** [/home/ole/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-6.6.47/Makefile:1164: vmlinux] Error 2`
`make[5]: *** [Makefile:234: __sub-make] Error 2`
`make[5]: Leaving directory '/home/ole/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-6.6.47'`
`make[4]: *** [Makefile:18: /home/ole/build/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-6.6.47/.modules] Error 2`
`make[4]: Leaving directory '/home/ole/build/openwrt/target/linux/mediatek'`
`make[3]: *** [Makefile:11: compile] Error 2`
`make[3]: Leaving directory '/home/ole/build/openwrt/target/linux'
time: target/linux/compile#6.90#5.77#12.55
    ERROR: target/linux failed to build.`
`make[2]: *** [target/Makefile:32: target/linux/compile] Error 1`
`make[2]: Leaving directory '/home/ole/build/openwrt'`
`make[1]: *** [target/Makefile:25: /home/ole/build/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2`
`make[1]: Leaving directory '/home/ole/build/openwrt'`
`make: *** [/home/ole/build/openwrt/include/toplevel.mk:248: world] Error 2`

### Expected behaviour

It should build properly and come out with an image rather than failing/Erroring.

### Diffconfig

```text
`tmp/.config-package.in:87840:error: recursive dependency detected!`
`tmp/.config-package.in:87840:	symbol PACKAGE_kmod-crypto-eip depends on PACKAGE_strongswan-mod-kdf`
`tmp/.config-package.in:109113:	symbol PACKAGE_strongswan-mod-kdf depends on PACKAGE_strongswan`
`tmp/.config-package.in:108465:	symbol PACKAGE_strongswan is selected by PACKAGE_kmod-crypto-eip
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"`

`tmp/.config-package.in:87840:error: recursive dependency detected!`
`tmp/.config-package.in:87840:	symbol PACKAGE_kmod-crypto-eip depends on PACKAGE_strongswan-mod-kdf`
`tmp/.config-package.in:109113:	symbol PACKAGE_strongswan-mod-kdf depends on PACKAGE_strongswan`
`tmp/.config-package.in:108465:	symbol PACKAGE_strongswan is selected by PACKAGE_kmod-crypto-eip
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"`

`CONFIG_TARGET_mediatek=y`
`CONFIG_TARGET_mediatek_mt7622=y`
`CONFIG_TARGET_mediatek_mt7622_DEVICE_xiaomi_redmi-router-ax6s=y`
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-08-30:
To follow this ticket

danpawlik, 2024-09-02:
Issue is in the mediateka feeds. Why did you create here a bug? What you expect from OpenWRT project maintainers?

HW-OLE, 2024-09-02:
Sorry, I also noticed this a few days ago. I apologize for making unnecessary stress.

closed 2024-09-02 by HW-OLE