openwrt/openwrt #16240: [<=23.05] Make use of the Octeon HWRNG

klondi, 2024-08-23:
### Describe the bug

Octeon processors may include a HWRNG. The HWRNG driver reports the quality of the generated data as 0.

In modern kernels this is interpreted as 1024 bits of entropy for every 1024 bits produced, on the kernel 5.15 used by OpenWRT 23.05 and earlier kernels this is instead interpreted as no quality and entails the hwrng data is not fed into the random number generator.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

octeon/generic

### Device

UBNT_E100 (CN5020p1.1-500-SCP) (Ubiquiti ER-Lite3)

### Image kind

Official downloaded image

### Steps to reproduce

To check if the kthread feeding the hwrng is running do:
# ps w | fgrep 'hwrngd' | fgrep -v fgrep

To check the quality of the hwrng do:
cat /sys/module/rng_core/parameters/current_quality

### Actual behaviour

The first command returns nothing, the second one returns 0

### Expected behaviour

The first command returns a line with the kthread, the second command returns 0 (on modern kernels) or 1024 (on 5.15 and earlier).

### Additional info

This issue can be solved on kernels prior to 6.2 by setting up the following option in the kernel cmdline: rng_core.default_quality=1024

I have tested the quality of the generated data using  rngtest and the generated data has a similar quality to that of urandom (which shouldn't be a surprise since it is whitened by the chip).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

bluecmd, 2024-08-25:
FWIW, on my Cisco vEdge 1000 running on 6.6 via https://github.com/openwrt/openwrt/pull/15838 this seems to work fine:

```
root@oob-lab-sw1:~# cat /sys/devices/virtual/misc/hw_random/rng_available
octeon 
root@oob-lab-sw1:~# cat /proc/sys/kernel/random/entropy_avail
256
```

klondi, 2024-08-25:
> FWIW, on my Cisco vEdge 1000 running on master with 6.6 via #15838 this seems to work fine:
> 
> ```
> root@oob-lab-sw1:~# cat /sys/devices/virtual/misc/hw_random/rng_available
> octeon 
> root@oob-lab-sw1:~# cat /proc/sys/kernel/random/entropy_avail
> 256
> root@oob-lab-sw1:~# cat /sys/module/rng_core/parameters/current_quality
> 1024
> ```

Sorry I wasn't clear enough. This issue was fixed on kernel version 6.2, but that's not what is currently in use nor what the documentation say is planned (6.1).

Just to make it easier to test, can you check if ps w lists the hwrngd kthread too?

bluecmd, 2024-08-26:
> Just to make it easier to test, can you check if ps w lists the hwrngd kthread too?

Sure, there is no `hwrngd` but a `hwrng` kthread.

```
root@oob-lab-sw1:~# ps aux | grep hwrng
root       109  0.0  0.0      0     0 ?        S    Aug25   0:00 [hwrng]
root      8209  0.0  0.0   1952   896 pts/1    S+   08:34   0:00 grep hwrng
```

klondi, 2024-08-26:
Then that means that your system's HWRNG is being used to provide entropy with linux 6.6 :)

I would love to see a fix on 23.05 if possible but waiting for the next release is an option, too.