openwrt/openwrt #16243: internet does not work on snapshot build

olumolu, 2024-08-23:
### Describe the bug

I use my router as a repeater with wifi
But it connect with my main router but does not provide internet 

### OpenWrt version

snapshot 

### OpenWrt release

r27205-5d107bbbbb

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Steps to reproduce

Install the image over my already working 23.5.4
When i install my snapshot build i find it connect to my other router over wifi but does not provide internet with wifi or lan there was no Internet.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-08-29:
Please attach syslog.

olumolu, 2024-08-29:
Not possible to add syslog as i have switch backed to 23.5

danpawlik, 2024-08-29:
@olumolu without logs, nobody will help. Nobody have glass ball for fortune telling

olumolu, 2024-08-29:
I understand in that case the way left is to close this as unresolved.

namiltd, 2024-08-29:
So please close