openwrt/openwrt #16254: I found that openwrt tried to download a non-existent file during the compilation process

liu2-3zhi, 2024-08-24:
### Describe the bug

I found that openwrt tried to download a non-existent file during the compilation process
http://roy.marples.name/downloads/dhcpcd/dhcpcd-9.4.1.tar.xz

but it has been moved to the github page https://github.com/NetworkConfiguration/dhcpcd/releases/latest

make[3]: Leaving directory '/home/zelly/openwrt/feeds/packages/sound/fdk-aac'
time: package/feeds/packages/fdk-aac/compile#10.02#1.50#12.31
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 502
Warning: Problem : HTTP error. Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 502
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location http://roy.marples.name/downloads/dhcpcd/dhcpcd-9.4.1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    14    0    14    0     0      0      0 --:--:--  0:00:27 --:--:-

### OpenWrt version

lates

### OpenWrt release

main

### OpenWrt target/subtarget

I have no idea.

### Device

X86

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

liu2-3zhi, 2024-08-24:
> 报告的版本无效。这是来自干净的存储库吗？`lates`

I don't know how to specify the version, since I'm using unreleased source code.

rany2, 2024-08-24:
This PR should address this issue and update to a new version: https://github.com/openwrt/packages/pull/24879

liu2-3zhi, 2024-08-25:
Still can't compile

rm -f /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/lib/dhcpcd/dhcpcd-hooks/10-mtu
install -d /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/usr/share/dhcpcd/hooks
install -m 0444 10-wpa_supplicant 15-timezone 29-lookup-hostname 50-yp.conf /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/usr/share/dhcpcd/hooks
install -d /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/usr/man/man8 install -m 0444 dhcpcd-run-hooks.8 /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/usr/man/man 8 make[5]: Leaving directory '/home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/hooks' make[4]: Leaving directory '/home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10' touch /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.built rm -rf /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd.installed /home/zelly/openwrt/build_dir/target-x86_64 _musl/dhcpcd-10.0.10/.pkgdir/dhcpcd mkdir -p /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd install -d -m0755 /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/sbin /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/etc /home/zelly/openwrt/build_dir/target- x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/lib/dhcpcd/dhcpcd-hooks install -m0755 /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/sbin/dhcpcd /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/sbin/ install -m0755 /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/lib/dhcpcd/dhcpcd-run-ks / home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/lib/dhcpcd/ install -m0644 /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/ipkg-install/lib/dhcpcd/dhcpcd-hooks/ * /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/lib/dhcpcd/dhcpcd-hooks/
install -d -m0755 /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/etc/init.d
install -m0755 ./files/dhcpcd.init /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd/etc/init.d/dhcpcd
install: cannot stat './files/dhcpcd.init': No such file or directory
make[3]: *** [Makefile:74: /home/zelly/openwrt/build_dir/target-x86_64_musl/dhcpcd-10.0.10/.pkgdir/dhcpcd.installed] Error 1 make[3]: Leaving directory '/home/zelly/openwrt/feeds/packages/net/dhtd' time: package/feeds/packages/dhtd/compile#8.58#0.80 #1.79 ERROR: package/feeds/packages/dhtd failed to build. make[2]: *** [package/Makefile:177: package/feeds/packages/dhtd/compile] Error 1 make[2]: Leaving directory '/home/zelly/openwrt' make[1]: *** [package/Makefile:171: /home/zelly/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: *** Waiting for unfinished jobs.... make[2]: Leaving directory '/home/zelly/openwrt' make[1]: Leaving directory '/home/zelly/openwrt' make: *** [/home/zelly/openwrt/include/toplevel.mk:248 ] Error 2

closed 2024-09-22 by thess