openwrt/openwrt #16259: Please consider backporting the support for Winbond W25N04KV spi nand, it's already in main kernel 6.9.

Rimits, 2024-08-25:
### Describe the bug

The offical build didn't included this patch, self build image can't use offical repo. Hence post it here.
The patch is alredy in the mainline kernel. 
[https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/drivers/mtd/nand/spi/winbond.c?h=v6.11-rc4&qt=grep&q=winbond](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/drivers/mtd/nand/spi/winbond.c?h=v6.11-rc4&qt=grep&q=winbond)
Please consider backporting it. Thanks.

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

mediatek/filogic

### Device

modified mt7986 router XD

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-08-27:
Here you go: https://github.com/openwrt/openwrt/pull/16272

hauke, 2024-08-29:
Do you have a device with this chip?

Rimits, 2024-08-30:
My router was modified with this chip,so it didn't really count.
But the u-boot support is added by ```./openwrt/package/boot/uboot-mediatek/patches/100-22-mtd-spi-nand-backport-from-upstream-kernel.patch```
Maybe the author of this patch knows some models that use this chip.
Since u-boot support is added, why not adding the kernel support to make devices with this chip really work.

namiltd, 2024-09-22:
Chip added, topic to close,

closed 2024-09-23 by Rimits