openwrt/openwrt #16261: IPv4 DNS broken on snapshot builds

rucketeg, 2024-08-25:
### Describe the bug

DNS using IPv4 is broken for me in the recent snapshot builds.

In dual stack environments this issue is visible as additional 5sec delay during each name resolution. First the DNS query is sent using IPv4, it seems like it never reaches the DNS server running on OpenWrt as no response is generated. 5 seconds later the same query is sent using IPv6 and reply comes in as expected.

This issue was confirmed with `dnsmasq`, `unbound` and `bind` used as a DNS server.

Interestingly, using `dig` or `host` utility to generate the DNS query produces slightly different UDP package (missing the Don't fragment flag) which is reaching the resolver correctly.

### OpenWrt version

r27208-cf765b1be6

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.4

### Image kind

Self-built image

### Steps to reproduce

Build a recent snapshot, try to ping google.com, watch the DNS traffic generated.

### Actual behaviour

DNS query is sent over IPv4 but it's silently dropped. 5 seconds later the same query is sent over IPv6, reaching the resolver.

### Expected behaviour

DNS works over IPv4.

### Additional info

Images built before b5c53848c3792355a2dd4ff0e06bf4d230e1f5fb are not showing the same behavior. Disabling GRO using `ethtool -K eth0 rx-gro-list off` is mitigating this issue.

### Diffconfig

```text
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_ethtool=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_px5g-mbedtls=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

schuettecarsten, 2024-08-27:
Maybe related to https://github.com/openwrt/openwrt/issues/16066 and https://github.com/openwrt/openwrt/commit/b5c53848c3792355a2dd4ff0e06bf4d230e1f5fb#commitcomment-143548442 ?

SuperMitic, 2024-12-03:
I think I'm experiencing a very similar issue. I have openWRT x86/64 (r28242-8349196571) installed in a VM inside a Proxmox node. I found that all VMs (and Proxmox host itself) are unable to get responses from DNS requests made to any server (except DNS servers internal to the local network or cloudflare 1.1.1.1 DNS). All other DNS servers (such as 8.8.8.8) do not respond to requests. I have done some pcap traces on the router and I notice that requests are sent on the WAN interface, but no response is ever received from the DNS server (DNS response missing). Only requests made with dig or host work. However, I discovered that when using E1000 as a virtual network interface instead of virtio in the openWRT VM the requests work correctly. 
The problem is that the E1000 interface is much slower than the virtualized virtio.

Beneter, 2025-03-03:
I am facing the same issue on Release 24.10.0. You can find some more details in the [forum](https://forum.openwrt.org/t/bizarre-dns-problems-with-debian-linux-after-upgrade-to-24-10/225860).

vgdh, 2025-06-30:
the issue still persist

Octanum, 2026-11-03:
Still happening on **25.12.6** (r33311-981c978eff, kernel 6.12.112, dnsmasq 2.93), Raspberry Pi 4B, LAN on the onboard bcmgenet `eth0`. Same workaround (`ethtool -K eth0 rx-gro-list off`). I narrowed down the trigger and have a small repro, in case it helps.

**Trigger:** two same-size UDP datagrams sent back to back from one socket (same 5-tuple) to the router. That's exactly what glibc does for `getaddrinfo(AF_UNSPEC)` without EDNS (A + AAAA), so `wget`, `ntpd`, `getent ahosts`, `ssh`/`ping` by name all hang ~20s. Single queries (e.g. `dig A`, `dig AAAA` run separately) work fine.

**Repro** (any LAN host, Python 3):
```python
import socket, struct, random, sys
SERVER = sys.argv[1]  # router LAN IP
def q(name, qtype):   # plain query, no EDNS
    hdr = struct.pack(">HHHHHH", random.randrange(65536), 0x0100, 1, 0, 0, 0)
    return hdr + b"".join(bytes([len(p)]) + p.encode() for p in name.split(".")) + b"\0" + struct.pack(">HH", qtype, 1)
ok = 0
for _ in range(10):
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM); s.connect((SERVER, 53)); s.settimeout(3)
    s.send(q("github.com", 1)); s.send(q("github.com", 28))   # A + AAAA, back to back
    try: s.recv(512); s.recv(512); ok += 1
    except socket.timeout: pass
    s.close()
print(f"both answered: {ok}/10")
```
`rx-gro-list on`: **0/10**. `rx-gro-list off` on eth0: **10/10**.

**Results** (10 tries each, no EDNS):

| Case | gro-list on | gro-list off |
|---|---|---|
| A + AAAA, same socket, back to back | 0/10 | 10/10 |
| A + A, or A + MX, same socket | 0/10 | 10/10 |
| Local name (`router.lan`), same socket | 0/10 | 10/10 |
| Same pair 1 ms apart, or from two sockets | 10/10 | 10/10 |
| Same pair sent directly to 1.1.1.1 | 10/10 | n/a |

Reproduced from a Windows 11 host and a Linux host (Intel e1000e).

**Evidence:**
- `tcpdump -i eth0` shows each pair as ONE packet with a 56-byte UDP payload (2 x 28-byte queries) and a bad checksum; the IP IDs show the sender sent two:
  `IP (... length 84) 192.168.1.198.57970 > 192.168.1.1.53: [bad udp cksum 0x71c7 -> 0x4a54!] 7000+ A? github.com. (56)`
- `Udp InCsumErrors` / `InErrors` in `/proc/net/snmp` rise in step with the failed pairs (+12 for 10 pairs). Conntrack `insert_failed` / `drop` stay 0, no NIC rx errors. So the merged skb reaches the local socket path and is dropped as a bad checksum instead of being segmented back.

**Open question:** the same pair *with* an EDNS OPT record (also equal size) is answered 20/20 even with gro-list on. I don't know why.

**Persisting the workaround:** a `/etc/hotplug.d/net/` script alone isn't enough here. `eth0` exists before hotplug runs, so it never gets an `add` event. Running it from `/etc/hotplug.d/iface/` on `ifup` (or from `/etc/rc.local`, as in #23542) works across reboots.