openwrt/openwrt #16268: MT76 fails to build

andi-at-1, 2024-08-27:
### Describe the bug

Snapshot fails for me to build MT76 with following error,
however 23.05 works fine.

In file included from /home/user/snapshot/openwrt/build_dir/target-mips_4kec_musl/linux-realtek_rtl838x/mt76-2024.08.25~904ef52a/mt7921/init.c:7:
/home/user/snapshot/openwrt/staging_dir/target-mips_4kec_musl/usr/include/mac80211-backport/linux/thermal.h:43:58: error: 'struct thermal_trip' declared inside parameter list will not be visible outside of this definition or declaration [-Werror]
   43 |                                         int (*cb)(struct thermal_trip *, void *),
      |                                                          ^~~~~~~~~~~~
/home/user/snapshot/openwrt/staging_dir/target-mips_4kec_musl/usr/include/mac80211-backport/linux/thermal.h: In function 'backport_for_each_thermal_trip':
/home/user/snapshot/openwrt/staging_dir/target-mips_4kec_musl/usr/include/mac80211-backport/linux/thermal.h:52:29: error: 'struct thermal_zone_device' has no member named 'trips'
   52 |                 ret = cb(&tz->trips[i], data);
      |                             ^~
cc1: all warnings being treated as errors

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

realtek/rtl838x

### Device

ZyXEL GS1900-24E

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-11-03:
@RussellSenior, @0xharshal, @bmork, @jwh, @plappermaul: What do you think?

plappermaul, 2024-11-04:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/include/linux/thermal.h?h=v6.6.59&id=e5bfcd30f88fdb0ce830229e7ccdeddcb7a59b04

plappermaul, 2024-11-04:
By the way: Why would one build MT76 for Realtek devices?

andi-at-1, 2024-11-05:
I dunno why that was included, but i thought it should not fail anyway so i reported it :-)

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442