openwrt/openwrt #16271: Problem with physical ports being swapped after reboot on nanopi r6s

odkrys, 2024-08-27:
### Describe the bug

before reboot
```
/sys/devices/platform/a41000000.pcie/pci0004:40/0004:40:00.0/0004:41:00.0/net/eth1
/sys/devices/platform/a40c00000.pcie/pci0003:30/0003:30:00.0/0003:31:00.0/net/eth2
```
after reboot
```
/sys/devices/platform/a40c00000.pcie/pci0003:30/0003:30:00.0/0003:31:00.0/net/eth1
/sys/devices/platform/a41000000.pcie/pci0004:40/0004:40:00.0/0004:41:00.0/net/eth2
```

### OpenWrt version

r27231-b870c16534

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R6S

### Image kind

Official downloaded image

### Steps to reproduce

Unplug the power cord and plug it back in.

### Actual behaviour

2.5Gbps LAN and WAN ports are swapped.

### Additional info

Is it possible to pin eth1 and eth2 to specific pcie in dts? Like this
```
&eth1 {
	status = "okay";
	phy-handle = <&pcie2x1l1>;

&eth2 {
	status = "okay";
	phy-handle = <&pcie2x1l2>;
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

odkrys, 2024-08-27:
It looks like the one in the friendlywrt repository is for this issue.
https://github.com/friendlyarm/kernel-rockchip/blob/925eb8ba4c765d4282a7281ddc6559a6e45c2687/arch/arm64/boot/dts/rockchip/rk3588-nanopi6-rev02.dts#L86C1-L110C3

ii343hbka, 2024-09-12:
There's a [PR](https://github.com/openwrt/openwrt/pull/16275) possibly fixing this

closed 2024-09-22 by odkrys