openwrt/openwrt #16273: mt7925: Kernel panic in mt7925_mac_sta_add then reboot into initramfs

danpawlik, 2024-08-27:
### Describe the bug

Hi,
Since commit [1]  in OpenWRT project, I got a kernel panic for mt7925e:

```
Sat Aug 24 13:56:54 2024 daemon.info hostapd: phy0-ap0: STA 8c:7a:3d:aa:aa:aa IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sat Aug 24 13:57:22 2024 daemon.info hostapd: phy1-ap0: STA 8c:7a:3d:aa:aa:aa IEEE 802.11: authenticated
Sat Aug 24 13:57:23 2024 daemon.info hostapd: phy1-ap0: STA 8c:7a:3d:aa:aa:aa IEEE 802.11: authenticated
[  131.602797] Unable to handle kernel paging request at virtual address ffffffffffffff98
[  131.610718] Mem abort info:
[  131.613515]   ESR = 0x0000000096000005
Sat Aug 24 13:57[  131.617253]   EC = 0x25: DABT (current EL), IL = 32 bits
:23 2024 kern.al[  131.623957]   SET = 0, FnV = 0
ert kernel: [  1[  131.628382]   EA = 0, S1PTW = 0
31.602797] Unabl[  131.632900]   FSC = 0x05: level 1 translation fault
e to handle kern[  131.639148] Data abort info:
el paging reques[  131.643404]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
t at virtual add[  131.650260]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
ress fffffffffff[  131.656685]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
fff98
Sat Aug 2[  131.663371] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000046b4c000
4 13:57:23 2024 [  131.671441] [ffffffffffffff98] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
kern.alert kerne[  131.681511] Internal error: Oops: 0000000096000005 [#1] SMP
[  131.688447] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet mt7925e(O) mt7925_common(O) mt7921e(O) mt7921_common(O) wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7996e(O) mt792x_lib(O) mt76_connac_lib(O) mt76(O) mac80211(O) libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables chacha_neon cfg80211(O) xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sfp sch_cake rtc_pcf8563 poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcurve25519_generic libcrc32c libchacha compat(O) at24 crypto_safexcel sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route
[  131.688621]  cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact pwm_fan i2c_mux_pca954x i2c_mux ifb ip6_udp_tunnel udp_tunnel sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug(O) usbcore usb_common aquantia
[  131.807714] CPU: 0 PID: 2065 Comm: hostapd Tainted: G           O       6.6.47 #0
[  131.815186] Hardware name: Bananapi BPI-R4 (DT)
[  131.819704] pstate: 20400005 (nzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  131.826653] pc : mt7925_mac_sta_add+0x394/0x748 [mt7925_common]
[  131.832576] lr : mt7925_mac_sta_add+0x330/0x748 [mt7925_common]
[  131.838489] sp : ffffffc08298b5c0
[  131.841793] x29: ffffffc08298b5c0 x28: ffffff80c595fb58 x27: ffffff80c59526d0
[  131.848919] x26: 0000000000000004 x25: 0000000000000001 x24: 0000000000000001
[  131.856044] x23: ffffff80c01f0a98 x22: ffffff80c91b99d0 x21: 0000000000000000
[  131.863168] x20: ffffff80c5951fe0 x19: ffffff80c01f0cc8 x18: 0000000000000000
[  131.870293] x17: 0000000000000000 x16: 0000000000000000 x15: 0000007fb6d65f70
[  131.877419] x14: 000000000000000e x13: 0000000000000000 x12: 0000000000000000
[  131.884544] x11: 0a10000001008e88 x10: 00000003aaaaeee7 x9 : 0a10000001008e88
[  131.891669] x8 : ffffff80ff769cc0 x7 : 1000000000000000 x6 : 0000000000000000
[  131.898794] x5 : 0000000000000000 x4 : ffffff80ff769f08 x3 : 0000000000000000
[  131.905918] x2 : 0000000000000000 x1 : ffffff80c01f0cc8 x0 : 0000000000000000
[  131.913043] Call trace:
[  131.915479]  mt7925_mac_sta_add+0x394/0x748 [mt7925_common]
[  131.921045]  ieee80211_check_fast_rx+0x178/0x3f8 [mac80211]
[  131.926636]  ieee80211_find_sta_by_link_addrs+0xac4/0x177c [mac80211]
[  131.933083]  sta_info_move_state+0x10/0x18 [mac80211]
[  131.938144]  ieee80211_nan_func_terminated+0x850/0x45e4 [mac80211]
[  131.944331]  ieee80211_nan_func_terminated+0xe5c/0x45e4 [mac80211]
[  131.950518]  ieee80211_nan_func_terminated+0x3898/0x45e4 [mac80211]
[  131.956792]  cfg80211_vendor_cmd_reply+0x18a8/0x52a4 [cfg80211]
[  131.962719]  genl_family_rcv_msg_doit+0xa8/0x108
[  131.967331]  genl_rcv_msg+0x1b0/0x244
[  131.970983]  netlink_rcv_skb+0x5c/0x128
[  131.974809]  genl_rcv+0x34/0x48
[  131.977941]  netlink_unicast+0x1e0/0x2c8
[  131.981854]  netlink_sendmsg+0x198/0x3c8
[  131.985766]  ____sys_sendmsg+0x1c4/0x274
[  131.989682]  ___sys_sendmsg+0x7c/0xc0
[  131.993333]  __sys_sendmsg+0x44/0x98
[  131.996898]  __arm64_sys_sendmsg+0x20/0x28
[  132.000982]  invoke_syscall.constprop.0+0x4c/0xe0
[  132.005679]  do_el0_svc+0x3c/0xb8
[  132.008983]  el0_svc+0x18/0x4c
[  132.012030]  el0t_64_sync_handler+0x118/0x124
[  132.016376]  el0t_64_sync+0x150/0x154
[  132.020030] Code: d37d1860 12000318 8b000261 12001c75 (78598042)
[  132.026111] ---[ end trace 0000000000000000 ]---
[  132.036313] pstore: backend (ramoops) writing error (-28)
[  132.041701] Kernel panic - not syncing: Oops: Fatal exception
[  132.047434] SMP: stopping secondary CPUs
[  132.051346] Kernel Offset: disabled
[  132.054823] CPU features: 0x0,00000000,20000000,1000400b
[  132.060123] Memory Limit: none
[  132.068688] Rebooting in 3 seconds..
PANIC at PC : 0x00000000430047d0
```

Wireless config:
```
config wifi-device 'radio2'
        option type 'mac80211'
        option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option channel '48'
        option txpower '20'
        option country 'PL'
        option cell_density '0'
        option htmode 'HE80'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'SomeSSID'
        option encryption 'sae'
        option key 'testtesttest'
        option wnm_sleep_mode '1'
        option bss_transition '1'
        option ocv '0'
        option ieee80211k '1'
```

Please let me know, if the issue should be created also on kernel bugzilla.

[1] https://github.com/openwrt/openwrt/commit/20bf5f35e70bf2385a6aa9a49053c72e7fdb1896

### OpenWrt version

r27229-ebe7c5f1a3

### OpenWrt release

r27229-ebe7c5f1a3

### OpenWrt target/subtarget

mt7925

### Device

Bananapi BPI-R4

### Image kind

Official downloaded image

### Steps to reproduce

Just walk though the apartment and the kernel panic will come.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2024-09-06:
Kindly ping @nbd168

Jason-Duffy, 2024-09-16:
Hi @nbd168 - I'm having the same issue, but different snapshot build and target. I'm more than happy to help with debugging on this device if needed, just let me know. 

## OpenWrt version
r27290-64dae1052b

## OpenWrt release
SNAPSHOT

## OpenWrt target/subtarget
BCM2712

## Device
Raspberry Pi 5

danpawlik, 2024-09-16:
@Jason-Duffy just curious: could you try to revert this commit https://github.com/openwrt/openwrt/commit/fd6ce0dea95272eec20eb222b08c7e685827caee (taken from comment: https://github.com/openwrt/openwrt/issues/16396#issuecomment-2353790321)

Jason-Duffy, 2024-09-24:
@danpawlik I'm on the Raspberry Pi 5, which currently only has snapshot openWRT support, so I don't think I can revert sadly. (Unless there's another way to build a snapshot I'm not aware of?).

nbd168, 2024-10-16:
Please test if adding this patch to `package/kernel/mac80211/patches/subsys` helps: https://nbd.name/p/44208690

danpawlik, 2024-10-17:
thanks Felix. I will try to test your patch ASAP, but it can take over 2 weeks.
Maybe @Jason-Duffy you can test, I can create images for you - without patch and with patch, to compare if that works.

Jason-Duffy, 2024-10-17:
Ah brilliant! Yes that would work well, just let me know when you're ready.

danpawlik, 2024-10-22:
@Jason-Duffy thanks.
I added github workflows config + triggered build without nbd168 patch: https://github.com/danpawlik/openwrt-builder/actions/workflows/rpi-5.yaml .
If all is fine, I will do next build with his patch and let's see.

Jason-Duffy, 2024-10-24:
Thanks for that @danpawlik - I tested this, and the WiFi card wasn't coming up at all. After some head scratching I realised I'd misread the title of this issue when I commented initially. Sorry about that!

The chipset on the WiFi card I'm using is in fact the MT7915, not the MT7925. The problem is presenting in exactly the same way though, so I still think there is value in me testing on my hardware. 

The only issue is the drivers are different, would you be able to build with these instead please?

kmod-mt7915-firmware
kmod-mt7915e

I tried to opkg install them myself but was unable to due to kernel mismatch. I diffed all the packages and found other differences in our builds, but this was the most obvious issue. 

kmod-mt76-connac
kmod-mt76-core

Are both still required.

danpawlik, 2024-10-24:
@Jason-Duffy triggered building. Soon should be available with mt7915, but dunno if @nbd168 patch will also cover that.

Jason-Duffy, 2024-10-29:
Weirdly I'm now having trouble recreating the issue at all! I've asked my colleagues to test on their hardware, and will let you know what we find.

Jason-Duffy, 2024-11-11:
Update on this - we haven't been able to recreate the bug. Sorry I couldn't be more helpful on this, and thanks for your assistance.

fs30000, 2025-01-31:
I'm having this crash, but on Debian 12 with kernel 6.12.7. Banana R4. Panics each 2 or 3 days. Something going on... I have a QCNCM865 card, but that also has problems. Doesn't even start.

/var/lib/systemd/pstore/dmesg-ramoops-0

> <1>[171951.076734] Unable to handle kernel paging request at virtual address ffffffffffffffa0
<1>[171951.084769] Mem abort info:
<1>[171951.087653]   ESR = 0x0000000096000006
<1>[171951.091479]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[171951.096867]   SET = 0, FnV = 0
<1>[171951.100008]   EA = 0, S1PTW = 0
<1>[171951.103227]   FSC = 0x06: level 2 translation fault
<1>[171951.108187] Data abort info:
<1>[171951.111144]   ISV = 0, ISS = 0x00000006, ISS2 = 0x00000000
<1>[171951.116705]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
<1>[171951.121841]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
<1>[171951.127231] swapper pgtable: 4k pages, 39-bit VAs, pgdp=000000004517d000
<1>[171951.134015] [ffffffffffffffa0] pgd=0000000045894003, p4d=0000000045894003, pud=0000000045894003, pmd=0000000000000000
<0>[171951.144716] Internal error: Oops: 0000000096000006 [#1] SMP
<7>[171951.150367] Modules linked in: wireguard libchacha20poly1305 chacha_neon libchacha poly1305_neon ip6_udp_tunnel udp_tunnel libcurve25519_generic nft_masq nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_set nf_tables libcrc32c mt7925e mt7925_common mt792x_lib mt76_connac_lib mt76 mac80211 libarc4 cfg80211 fuse ip_tables x_tables
<7>[171951.190794] CPU: 1 UID: 0 PID: 3202 Comm: hostapd Not tainted 6.12.7-bpi-r4-main #3
<7>[171951.198526] Hardware name: Banana Pi BPI-R4 (DT)
<7>[171951.203219] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[171951.210256] pc : mt7925_sta_set_decap_offload+0xb4/0x178 [mt7925_common]
<7>[171951.217042] lr : mt7925_sta_set_decap_offload+0x50/0x178 [mt7925_common]
<7>[171951.223820] sp : ffffffc082183550
<7>[171951.227210] x29: ffffffc082183550 x28: ffffff80c12fecb0 x27: 0000000000000000
<7>[171951.234424] x26: 0000000000000001 x25: 0000000000000001 x24: 0000000000000001
<7>[171951.241636] x23: ffffff80c95ce8c8 x22: ffffff80c95c26f0 x21: ffffff80c12fea80
<7>[171951.248849] x20: ffffff80c6545de0 x19: ffffff80c95c2000 x18: 0000000000000000
<7>[171951.256062] x17: 0000000000000000 x16: 0000000000000000 x15: 00000055942d87a0
<7>[171951.263274] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
<7>[171951.270486] x11: 0000000000000040 x10: ffffffc081639e28 x9 : ffffffc081639e20
<7>[171951.277698] x8 : ffffff80c0400028 x7 : 0000000000000000 x6 : 0000000000000000
<7>[171951.284909] x5 : ffffff80c0400000 x4 : ffffff80c04000a8 x3 : 0000000000000000
<7>[171951.292121] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff80c12fecb0
<7>[171951.299333] Call trace:
<7>[171951.301857]  mt7925_sta_set_decap_offload+0xb4/0x178 [mt7925_common]
<7>[171951.308288]  ieee80211_check_fast_rx+0x1c8/0x504 [mac80211]
<7>[171951.313979]  _sta_info_move_state+0xd8/0x4ac [mac80211]
<7>[171951.319313]  sta_info_move_state+0x14/0x20 [mac80211]
<7>[171951.324471]  sta_apply_auth_flags.constprop.0+0x88/0x19c [mac80211]
<7>[171951.330845]  sta_apply_parameters+0x214/0x3d4 [mac80211]
<7>[171951.336264]  ieee80211_add_station+0xd4/0x180 [mac80211]
<7>[171951.341681]  nl80211_new_station+0x45c/0x624 [cfg80211]
<7>[171951.347019]  genl_family_rcv_msg_doit+0xc0/0x128
<7>[171951.351719]  genl_rcv_msg+0x1e4/0x26c
<7>[171951.355459]  netlink_rcv_skb+0x5c/0x128
<7>[171951.359372]  genl_rcv+0x38/0x50
<7>[171951.362591]  netlink_unicast+0x2e4/0x33c
<7>[171951.366590]  netlink_sendmsg+0x17c/0x3b4
<7>[171951.370589]  ____sys_sendmsg+0x18c/0x2f8
<7>[171951.374591]  ___sys_sendmsg+0x80/0xdc
<7>[171951.378329]  __sys_sendmsg+0x68/0xc8
<7>[171951.381982]  __arm64_sys_sendmsg+0x24/0x30
<7>[171951.386155]  invoke_syscall+0x48/0x110
<7>[171951.389985]  el0_svc_common.constprop.0+0x40/0xe0
<7>[171951.394766]  do_el0_svc+0x1c/0x28
<7>[171951.398160]  el0_svc+0x30/0xd0
<7>[171951.401293]  el0t_64_sync_handler+0x100/0x12c
<7>[171951.405727]  el0t_64_sync+0x194/0x198
<0>[171951.409469] Code: d280003a f9420381 12001c7b aa1c03e0 (785a0021)
<4>[171951.415637] ---[ end trace 00000

/var/lib/systemd/pstore/dmesg-ramoops-1

> <1>[171951.076734] Unable to handle kernel paging request at virtual address ffffffffffffffa0
<1>[171951.084769] Mem abort info:
<1>[171951.087653]   ESR = 0x0000000096000006
<1>[171951.091479]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[171951.096867]   SET = 0, FnV = 0
[... 34 lines cut ...]
<7>[171951.341681]  nl80211_new_station+0x45c/0x624 [cfg80211]
<7>[171951.347019]  genl_family_rcv_msg_doit+0xc0/0x128
<7>[171951.351719]  genl_rcv_msg+0x1e4/0x26c
<7>[171951.355459]  netlink_rcv_skb+0x5c/0x128
<7>[171951.359372]  genl_rcv+0x38/0x50
<7>[171951.362591]  netlink_unicast+0x2e4/0x33c
<7>[171951.366590]  netlink_sendmsg+0x17c/0x3b4
<7>[171951.370589]  ____sys_sendmsg+0x18c/0x2f8
<7>[171951.374591]  ___sys_sendmsg+0x80/0xdc
<7>[171951.378329]  __sys_sendmsg+0x68/0xc8
<7>[171951.381982]  __arm64_sys_sendmsg+0x24/0x30
<7>[171951.386155]  invoke_syscall+0x48/0x110
<7>[171951.389985]  el0_svc_common.constprop.0+0x40/0xe0
<7>[171951.394766]  do_el0_svc+0x1c/0x28
<7>[171951.398160]  el0_svc+0x30/0xd0
<7>[171951.401293]  el0t_64_sync_handler+0x100/0x12c
<7>[171951.405727]  el0t_64_sync+0x194/0x198
<0>[171951.409469] Code: d280003a f9420381 12001c7b aa1c03e0 (785a0021)
<4>[171951.415637] ---[ end trace 0000000000000000 ]---
<3>[171951.421880] pstore: backend (ramoops) writing error (-28)
<0>[171951.427357] Kernel panic - not syncing: Oops: Fatal exception
<2>[171951.433178] SMP: stopping secondary CPUs
<0>[171951.437181] Kernel Offset: disabled
<0>[171951.440746] CPU features: 0x00,00000102,00200000,4200420b
<0>[171951.446221] Memory Limit: none

morrownr, 2025-05-08:
@fs30000 

I have been trying to track down information on this issue. Are you aware of anything more since you posted the above message in Jan 25?

I cannot see where anyone has bisected this problem. What I do see is the following patch that preceded the various reports of this problem. 

https://patchwork.kernel.org/project/linux-wireless/patch/df2524e3a63a93699bde441e4ab70a70bbac53ef.1720248331.git.sean.wang@kernel.org/

A good bisect and this problem could probably be fixed very soon.

fs30000, 2025-05-08:
> [@fs30000](https://github.com/fs30000)
>
> I have been trying to track down information on this issue. Are you aware of anything more since you posted the above message in Jan 25?
>
> I cannot see where [...]

I swapped the card to a mt7922. So, i haven't followed up on this issue ever since. Problem is, the present card is also having some issues, like sites slow to load, apps slow to start, etc. Logs show nothing. As i upated the kernel to 6.12.23 (using Frank-w repo for banana pi r4), maybe  time to test the mt7925 again. Or the qualcomm card, but last time, it was total crap. Didn't even started properly. Because of country stuff, etc.

On the other hand,  i have a second banana pi r4 with openwrt with the mt7996e card and that one is working fine now, after many months of instability.