openwrt/openwrt #16274: WiFi broken on Xiaomi Redmi AX6000 on SNAPSHOTS newer than r27168

OpenWRT-fanboy, 2024-08-27:
### Describe the bug

Recent updates to snapshot (likely mt76) have completely broken WiFi on the Redmi AX6000

### OpenWrt version

r27229-ebe7c5f1a3

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Steps to reproduce

reboot then go to Network/Wireless

### Actual behaviour

from r27229

[image]

### Expected behaviour

from r27168

[image]

### Additional info

Kernel log shows the ports going up and down every few seconds. This did not happen on r27168.

[  806.326669] mt7530-mdio mdio-bus:1f lan2: Link is Down
[  806.735981] mt7530-mdio mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[  808.351624] br-lan: port 1(lan2) entered disabled state
[  808.357211] br-lan: port 4(wan) entered blocking state
[  808.362606] br-lan: port 4(wan) entered forwarding state
[  811.407881] mt7530-mdio mdio-bus:1f lan3: Link is Down
[  811.481173] mt7530-mdio mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control off
[  812.417958] br-lan: port 2(lan3) entered disabled state
[  812.423845] br-lan: port 1(lan2) entered blocking state
[  812.429335] br-lan: port 1(lan2) entered forwarding state
[  816.487965] mt7530-mdio mdio-bus:1f lan2: Link is Down
[  816.698659] mt7530-mdio mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  817.499232] br-lan: port 1(lan2) entered disabled state
[  817.505177] br-lan: port 2(lan3) entered blocking state
[  817.510656] br-lan: port 2(lan3) entered forwarding state

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

OpenWRT-fanboy, 2024-08-28:
Syslog attached. Note that aside from the port instability, there were no appreciable differences from the r27168 syslog.

[sys 27229.txt](https://github.com/user-attachments/files/16788972/sys.27229.txt)

LearZhou, 2024-08-28:
A hard reboot may help. My Xiaomi AX3000T suffered similar issue with recent snapshots, only unplug/plug the power can bring it back.

OpenWRT-fanboy, 2024-08-29:
> A hard reboot may help. My Xiaomi AX3000T suffered similar issue with recent snapshots, only unplug/plug the power can bring it back.

Unfortunately, this does not remedy the situation

OpenWRT-fanboy, 2024-08-29:
This might be potentially related:

https://github.com/openwrt/openwrt/issues/16281

Neustradamus, 2024-08-30:
To follow this ticket

ThatMishakov, 2024-09-01:
I've also had an issue with this router, ever since getting it, where WiFi (both 2.4 and 5Ghz) would break after few hours to few days, making connected devices unable to access the internet or local network (which is so bad I've had its WiFi disabled and have been using it with RT3200 as dumb AP). This even happens on the default settings after resetting the router, and only setting the WPA2 password and my country. I have not seen anything suspicious in the logs, and I don't know how can I diagnose it further than that. Perhaps this issue could be related, as it seems to be happening more quickly on recent snapshots.

OpenWRT-fanboy, 2024-09-01:
> I've also had an issue with this router, ever since getting it, where WiFi (both 2.4 and 5Ghz) would break after few hours to few days, making connected devices unable to access the internet or [...]

Try running a stable release, like 23.05.4. Prior to this recent issue, it has been rock solid for me.

OpenWRT-fanboy, 2024-09-01:
Important update.

I have a couple Redmi AX6000, both used as APs on my network. The first has two printers connected to the wired ports, while the second is only connected to the network using the WAN port. When I update the second Redmi AX6000 to the latest snapshot, WiFi comes on, although it does take marginally longer to fully initialize (kernel log ends at 32 seconds instead of 18 seconds).

As previously described, on the first Redmi AX6000, WiFi does not work, and the ports go up and down. Here's the interesting part. When I disconnect the Ethernet cables for the printers and reboot, the WiFi works and behaves just like the second one. (This ones takes 51 seconds, though) Surprisingly, when I reconnect the cables, the link goes up and stays up.

Therefore, the broken WiFi is just a side effect from the port instability, which only manifests at boot. Of course, no one wants to be pulling and reconnecting cables every time they reboot a router, but this is a workaround until the bug gets squashed.

everything411, 2024-09-02:
apply this patch https://github.com/openwrt/openwrt/issues/16281#issuecomment-2322095969 solve this issue for me

ThatMishakov, 2024-09-02:
> Try running a stable release, like 23.05.4. Prior to this recent issue, it has been rock solid for me.

I've had this issue on all firmwares, from pre-23 snapshots, to stable 23 versions, and recent snapshots. On stable it doesn't happen as quickly, but wireless still consistently breaks after a few days
> The ports go up and down

I have not had any of these issues. This router has been rock solid as NAT router, and when WiFi breaks, wired ports and routing does not break for me. Also, when WiFi breaks, it feels like it's RX and unicast what's breaking, since broadcast still seemingly works (devices still get DHCP leases) and I think I've seen all devices getting stuck at 6Mb/s rate or whatever.

Out of my mt76 routers, my Redmi AC2100 (with MT7615N) has also seemingly been misbehaving lately (but not to the point where WiFi would completely break on all devices), though RT3200's WiFi has been rock solid, so I think it's mt76 WiFi drivers weirdness

cliobrando, 2024-09-09:
> Important update.
> 
> I have a couple Redmi AX6000, both used as APs on my network. The first has two printers connected to the wired ports, while the second is only connected to the network using the WAN port. When I update the second Redmi AX6000 to the latest snapshot, WiFi comes on, although it does take marginally longer to fully initialize (kernel log ends at 32 seconds instead of 18 seconds).
> 
> As previously described, on the first Redmi AX6000, WiFi does not work, and the ports go up and down. Here's the interesting part. When I disconnect the Ethernet cables for the printers and reboot, the WiFi works and behaves just like the second one. (This one takes 51 seconds, though) Surprisingly, when I reconnect the cables, the link goes up and stays up.
> 
> Therefore, the broken WiFi is just a side effect from the port instability, which only manifests at boot. Of course, no one wants to be pulling and reconnecting cables every time they reboot a router, but this is a workaround until the bug gets squashed.

Same behavior here, AX6000 as dumb AP, two cables connected: one from the router (openwrt x86 - r8169) and the other from a desktop PC (intel gb lan).
The AP cannot start if the PC is connected, If I remove the cable, or disable the nic from the control panel, the AP starts.
The cable from the router can be connected and doesn't interfere with the normal boot process.

Dmesg shows the same behavior reported: a loop of reconnects from the PC port.

ptpt52, 2024-09-09:
same behavior on my mt7981 device.

st7105, 2024-09-13:
r27350-c4a9265160 - problem resolved.

OpenWRT-fanboy, 2024-09-13:
> r27350-c4a9265160 - problem resolved.

Same. Not only does it fix the bug, the 18 second boot is back!

closed 2024-09-13 by OpenWRT-fanboy