openwrt/openwrt #16278: Static routes in /etc/config/network don't work

Nuxssss, 2024-08-28:
### Describe the bug

Static routes in /etc/config/network don't work

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Steps to reproduce

add to network config
```
config route
        option interface 'lan'
        option target '0.0.0.0/0'
        option gateway '192.168.1.1'
        option source '192.168.1.0/24'
        option table 'main'
```

```
service network restart
ip route show table main
```

### Actual behaviour

```
[~] ip r
default via 172.20.0.254 dev wan  src 172.20.0.199  metric 10 
10.10.20.0/24 dev lan1 scope link  src 10.10.20.1 
172.20.0.0/24 dev wan scope link  metric 10 
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1 
213.110.226.32/27 dev wanmac-public scope link  metric 20 
```

### Expected behaviour

The routing table contains the specified route

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2024-08-28:
The source must be an IP, not a prefix. Doing the ip command equivalent reveals the issue:

```
root@OpenWrt:~# ip route add 0.0.0.0/0 via 192.168.1.1 src 192.168.1.0/24 dev br-lan
ip: an inet address is expected rather than "192.168.1.0/24"
root@OpenWrt:~#
```

Apart from that, what is this route supposed to achieve? Judging from your output, the router seems to have assigned 192.168.1.1 to it's own br-lan, so you want to route 0/0 via br-lan from br-lan?

Include a more complete network configuration.

Nuxssss, 2024-08-29:
my fault, but I would like to see at least error messages when trying to do something like this

brada4, 2024-08-29:
Fill the issue in ../luci that permits entering/saving invalid values.