openwrt/openwrt #16281: filogic: GL-MT6000: after backport pending Ethernet PHY driver patches WAN issue reported

pesa1234, 2024-08-29:
### Describe the bug

No IP on the WAN interface. Unplug/plug cabling to re-trigger DHCP.

And a kernel panic occours.

### OpenWrt version

r27243-dd1bfbe971

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Connect the WAN port to a cable modem - ONT.

Wait and you don't get the IP or you got a kernel panic.

A lot of users get the issue, please have a look here: https://forum.openwrt.org/t/gl-inet-flint-2-gl-mt6000-discussions/173524/1724

Kernel panic detail:

```

[60067.863413] ------------[ cut here ]------------
[60067.868034] phy_check_link_status+0x0/0xc0: returned: -110
[60067.873553] WARNING: CPU: 1 PID: 1046 at phy_state_machine+0x90/0x28c
[60067.879981] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet wireguard rndis_host pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mt7915e(O) mt76_connac_lib(O) mt76(O) mac80211(O) libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables chacha_neon cfg80211(O) cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables usbnet slhc sch_cake poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha ipheth compat(O) crypto_safexcel sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
[60067.880154]  act_mirred act_gact ifb ip6_udp_tunnel udp_tunnel tun sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug(O) usbcore usb_common aquantia mii
[60067.997212] CPU: 1 PID: 1046 Comm: kworker/u8:2 Tainted: G           O       6.6.47 #0
[60068.005108] Hardware name: GL.iNet GL-MT6000 (DT)
[60068.009795] Workqueue: events_power_efficient phy_state_machine
[60068.015705] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[60068.022646] pc : phy_state_machine+0x90/0x28c
[60068.026988] lr : phy_state_machine+0x90/0x28c
[60068.031329] sp : ffffffc085753d50
[60068.034627] x29: ffffffc085753d50 x28: 0000000000000000 x27: 61c8864680b583eb
[60068.041743] x26: ffffff8000011028 x25: ffffff8006d2db00 x24: ffffff8000017405
[60068.048859] x23: 00000000ffffff92 x22: 0000000000000006 x21: ffffff8000271490
[60068.055975] x20: ffffff80002714e8 x19: ffffff8000271000 x18: 00000000000001db
[60068.063090] x17: 0000000000000000 x16: 0000000000000000 x15: ffffffc080bda1e8
[60068.070206] x14: 0000000000000591 x13: 00000000000001db x12: 00000000ffffffea
[60068.077321] x11: 00000000ffffefff x10: ffffffc080c321e8 x9 : ffffffc080bda190
[60068.084436] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
[60068.091551] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[60068.098667] x2 : ffffff803fd9c490 x1 : ffffffbfbf1e9000 x0 : 000000000000002e
[60068.105783] Call trace:
[60068.108215]  phy_state_machine+0x90/0x28c
[60068.112212]  process_one_work+0x174/0x2cc
[60068.116210]  worker_thread+0x2e8/0x4d0
[60068.119944]  kthread+0xd8/0xdc
[60068.122986]  ret_from_fork+0x10/0x20
[60068.126548] ---[ end trace 0000000000000000 ]---
```

### Additional info

Reverting this upstream patch https://github.com/openwrt/openwrt/commit/1069514978c419cc5bdfc18c5670b46d0443a686 solve the issue.

Thanks

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hnyman, 2024-08-29:
I see the problem intermittently with MT6000 with r27243-dd1bfbe971.
FTTH ethernet, no modem.
I have not had any kernel crashes so far, but sometimes there is no DHCP connection on wan after booting.

With the same build, I have had mixed results:

Yesterday, there was no address after flashing.

Today I got an address at the first boot after flashing the same image, but after a new reboot there was no address. However, when I disconnected the wan cable and reconnected it, I did get an address (like mentioned in forum https://forum.openwrt.org/t/gl-inet-flint-2-gl-mt6000-discussions/173524/1726 )
Same with the next boot: only disconnecting the cable triggered DHCP to work.

Could there be something wrong with link status detection?

I did my own test build with 10695149 and fca306c5 reverted, and the router always gets an address at boot. So, the problem likely lies with the new ethernet PHY driver patches for Mediatek.

LuisMitaHL, 2024-08-29:
Same bug with Confiabits MT7981: `phy_check_link_status+0x0/0xc0: returned: -110`

Neustradamus, 2024-08-30:
To follow this ticket

mrkiko, 2024-08-30:
To follow this ticket as well.

dangowrt, 2024-08-30:
Please try to remove only

target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch

@onkyfly34 any ideas?

pesa1234, 2024-08-30:
> Please try to remove only target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch @onkyfly34 any ideas?

As I come back home I'll do it.

Thanks for your answer

jebrossard, 2024-08-30:
I can restore connectivity after a restart with "ifconfig eth1 down" followed by "ifconfig eth1 up".  This is helpful as physical access to the WAN cable is a bit of a PITA with my setup.

pesa1234, 2024-08-30:
@dangowrt by simply removing it I get errors on other patches that needs to be fixed.
```
.../openwrt/target/linux/mediatek/patches-6.6/733-12-net-phy-mediatek-Fix-alignment-in-callback-functions.patch
```

Thanks

jglooije, 2024-08-30:
Same problem on GL.iNet GL-MT6000.
I put this in startup(rc.local) to work around and not get stuck after powerfailure.
sleep 5
ifdown wan
ifup wan

hnyman, 2024-08-30:
> > try to remove only 
> > target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch

> simply removing it I get errors on other patches that needs to be fixed.

We need to remove (or modify) also the whitespace-alignment patch 12 of the series (that is just no-op cosmetics). That enables preparing the kernel sources.
However, the actual compilation will fail, as the removed "10" patch also modifies mtk.h by moving some symbols from mtk-ge.c that also the main patch 11 expects to find in mtk.h:
```
declared (first use in this function); did you mean 'MTK_PHY_PAGE_EXTENDED_52B5'?
  162 |         phy_modify_paged(phydev, MTK_PHY_PAGE_EXTENDED_1,
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
      |                                  MTK_PHY_PAGE_EXTENDED_52B5
drivers/net/phy/mediatek/mtk-2p5ge.c:162:34: note: each undeclared identifier is reported only once for each function it appears in
drivers/net/phy/mediatek/mtk-2p5ge.c:163:26: error: 'MTK_PHY_AUX_CTRL_AND_STATUS' undeclared (first use in this function); did you mean 'PHY_AUX_CTRL_STATUS'?
  163 |                          MTK_PHY_AUX_CTRL_AND_STATUS,
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                          PHY_AUX_CTRL_STATUS
drivers/net/phy/mediatek/mtk-2p5ge.c:164:29: error: 'MTK_PHY_ENABLE_DOWNSHIFT' undeclared (first use in this function)
  164 |                          0, MTK_PHY_ENABLE_DOWNSHIFT);
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~
make[11]: *** [scripts/Makefile.build:243: drivers/net/phy/mediatek/mtk-2p5ge.o] Error 1
```

A solution might be editing patch 10 by leaving the symbols related mtk-ge.c and mtk.h stuff, but removing the other modifications. I will try that.

hnyman, 2024-08-30:
At the first glance, this seems to work for me with r27269-c6f04f718e

* delete the no-op cosmetics patch 12:    target/linux/mediatek/patches-6.6/733-12-net-phy-mediatek-Fix-alignment-in-callback-functions.patch
* modify the patch 10: Keep mtk.h stuff but remove new functionality:   target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch

Router got a proper wan address ok after flashing and also after the next reboot.

Modifications needed:
```
diff --git a/target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch b/target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch
index dcbe957836..1afad579db 100644
--- a/target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch
+++ b/target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch
@@ -28,24 +28,6 @@ Signed-off-by: SkyLake.Huang <skylake.huang@mediatek.com>
  drivers/net/phy/mediatek/mtk.h         | 21 ++++++
  4 files changed, 116 insertions(+), 4 deletions(-)
 
---- a/drivers/net/phy/mediatek/mtk-ge-soc.c
-+++ b/drivers/net/phy/mediatek/mtk-ge-soc.c
-@@ -1396,6 +1396,7 @@ static struct phy_driver mtk_socphy_driv
- 		PHY_ID_MATCH_EXACT(MTK_GPHY_ID_MT7981),
- 		.name		= "MediaTek MT7981 PHY",
- 		.config_init	= mt798x_phy_config_init,
-+		.read_status	= mtk_gphy_cl22_read_status,
- 		.config_intr	= genphy_no_config_intr,
- 		.handle_interrupt = genphy_handle_interrupt_no_ack,
- 		.probe		= mt7981_phy_probe,
-@@ -1413,6 +1414,7 @@ static struct phy_driver mtk_socphy_driv
- 		PHY_ID_MATCH_EXACT(MTK_GPHY_ID_MT7988),
- 		.name		= "MediaTek MT7988 PHY",
- 		.config_init	= mt798x_phy_config_init,
-+		.read_status	= mtk_gphy_cl22_read_status,
- 		.config_intr	= genphy_no_config_intr,
- 		.handle_interrupt = genphy_handle_interrupt_no_ack,
- 		.probe		= mt7988_phy_probe,
 --- a/drivers/net/phy/mediatek/mtk-ge.c
 +++ b/drivers/net/phy/mediatek/mtk-ge.c
 @@ -9,10 +9,6 @@
@@ -59,115 +41,6 @@ Signed-off-by: SkyLake.Huang <skylake.huang@mediatek.com>
  #define MTK_PHY_PAGE_EXTENDED_2			0x0002
  #define MTK_PHY_PAGE_EXTENDED_3			0x0003
  #define MTK_PHY_RG_LPI_PCS_DSP_CTRL_REG11	0x11
-@@ -251,6 +247,7 @@ static struct phy_driver mtk_gephy_drive
- 		.name		= "MediaTek MT7531 PHY",
- 		.probe		= mt7531_phy_probe,
- 		.config_init	= mt7531_phy_config_init,
-+		.read_status	= mtk_gphy_cl22_read_status,
- 		/* Interrupts are handled by the switch, not the PHY
- 		 * itself.
- 		 */
---- a/drivers/net/phy/mediatek/mtk-phy-lib.c
-+++ b/drivers/net/phy/mediatek/mtk-phy-lib.c
-@@ -109,6 +109,98 @@ int mtk_phy_write_page(struct phy_device
- }
- EXPORT_SYMBOL_GPL(mtk_phy_write_page);
- 
-+/* This function deals with the case that 1G AN starts but isn't completed. We
-+ * set AN_NEW_LP_CNT_LIMIT with different values time after time to let our
-+ * 1G->100Mbps hardware automatic downshift to fit more partner devices.
-+ */
-+static int extend_an_new_lp_cnt_limit(struct phy_device *phydev)
-+{
-+	int mmd_read_ret;
-+	u32 reg_val;
-+	int timeout;
-+
-+	/* According to table 28-9 & Figure 28-18 in IEEE 802.3,
-+	 * link_fail_inhibit_timer of 10/100/1000 Mbps devices ranges from 750
-+	 * to "1000ms". Once MTK_PHY_FINAL_SPEED_1000 is set, it means that we
-+	 * enter "FLP LINK GOOD CHECK" state, link_fail_inhibit_timer starts and
[... 287 lines cut ...]
+#define MTK_PHY_AUX_CTRL_AND_STATUS		0x14
+/* suprv_media_select_RefClk */
+#define   MTK_PHY_LP_DETECTED_MASK		GENMASK(7, 6)
+#define   MTK_PHY_ENABLE_DOWNSHIFT		BIT(4)
+
 #define MTK_PHY_PAGE_EXTENDED_52B5		0x52b5
 
+/* Registers on Token Ring debug nodes */
+/* ch_addr = 0x0, node_addr = 0xf, data_addr = 0x2 */
+#define   AN_STATE_MASK			GENMASK(22, 19)
+#define   AN_STATE_SHIFT		19
+#define   AN_STATE_TX_DISABLE		1
+
+/* ch_addr = 0x0, node_addr = 0xf, data_addr = 0x3c */
+#define AN_NEW_LP_CNT_LIMIT_MASK		GENMASK(23, 20)
+#define AUTO_NP_10XEN				BIT(6)
+
+/* Registers on MDIO_MMD_VEND1 */
+#define MTK_PHY_LINK_STATUS_MISC	(0xa2)
+#define   MTK_PHY_FINAL_SPEED_1000	BIT(3)
+
 /* Registers on MDIO_MMD_VEND2 */
 #define MTK_PHY_LED0_ON_CTRL			0x24
 #define MTK_PHY_LED1_ON_CTRL			0x26
```

cmonroe, 2024-08-30:
I ran into a similar issue on many of our Mediatek targets and am still investigating. The most common symptom is that udhcpc does not start on WAN link up. It is certainly a race condition as it works OK sometimes and fails others.

If I revert netifd [68c8a4f9](https://git.openwrt.org/?p=project/netifd.git;a=commit;h=68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd) the issue seems to go away or at least happen infrequently enough that it is much harder to reproduce.

I will try restoring the commit above but remove patch 733-10 as @dangowrt suggested to see if it also resolves the issue.

OpenWRT-fanboy, 2024-09-02:
Related bug on Redmi AX6000: https://github.com/openwrt/openwrt/issues/16274#issuecomment-2323542947

pesa1234, 2024-09-03:
@dangowrt @nbd168 should we check with latest changes on netfid? example: https://github.com/openwrt/netifd/commit/cd2a7964f2c09994c09388a76e1a9c6aef4297a2

jglooije, 2024-09-03:
> @dangowrt @nbd168 should we check with latest changes on netfid? example: [openwrt/netifd@cd2a796](https://github.com/openwrt/netifd/commit/cd2a7964f2c09994c09388a76e1a9c6aef4297a2)
> 
> Thanks!

Tested, did not solve the problem om my router GL.iNet GL-MT6000

cmonroe, 2024-09-03:
> > @dangowrt @nbd168 should we check with latest changes on netfid? example: [openwrt/netifd@cd2a796](https://github.com/openwrt/netifd/commit/cd2a7964f2c09994c09388a76e1a9c6aef4297a2)
> > Thanks!
> 
> Tested, did not solve the problem om my router GL.iNet GL-MT6000

I was just about to post the same but you beat me to it.

I'm going back to test with the latest netifd/OpenWrt master + commenting out the lines which register mtk_gphy_cl22_read_status() as the read_status callback. This should be effectively the same as (mostly) removing the patch like you did above.

dangowrt, 2024-09-03:
@hnyman Can you confirm that omitting `733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch` in the way you described in  https://github.com/openwrt/openwrt/issues/16281#issuecomment-2322095969 does the trick and fixes the issue?

cmonroe, 2024-09-03:
I'm not quite ready to declare victory but it seems quite promising that the change below on top of OpenWrt master from today works around the issue for me as well.

In one of our test networks it only took 3-4 reboots to trigger a failure when using OpenWrt master from today with the latest netifd changes. I then applied the patch below and we're up to 12 reboots without issue.

```
--- a/drivers/net/phy/mediatek/mtk-ge-soc.c
+++ b/drivers/net/phy/mediatek/mtk-ge-soc.c
@@ -1396,7 +1396,7 @@ static struct phy_driver mtk_socphy_driv
 		PHY_ID_MATCH_EXACT(MTK_GPHY_ID_MT7981),
 		.name = "MediaTek MT7981 PHY",
 		.config_init = mt798x_phy_config_init,
-		.read_status = mtk_gphy_cl22_read_status,
+		//.read_status = mtk_gphy_cl22_read_status,
 		.config_intr = genphy_no_config_intr,
 		.handle_interrupt = genphy_handle_interrupt_no_ack,
 		.probe = mt7981_phy_probe,
@@ -1414,7 +1414,7 @@ static struct phy_driver mtk_socphy_driv
 		PHY_ID_MATCH_EXACT(MTK_GPHY_ID_MT7988),
 		.name = "MediaTek MT7988 PHY",
 		.config_init = mt798x_phy_config_init,
-		.read_status = mtk_gphy_cl22_read_status,
+		//.read_status = mtk_gphy_cl22_read_status,
 		.config_intr = genphy_no_config_intr,
 		.handle_interrupt = genphy_handle_interrupt_no_ack,
 		.probe = mt7988_phy_probe,
--- a/drivers/net/phy/mediatek/mtk-ge.c
+++ b/drivers/net/phy/mediatek/mtk-ge.c
@@ -247,7 +247,7 @@ static struct phy_driver mtk_gephy_drive
 		.name = "MediaTek MT7531 PHY",
 		.probe = mt7531_phy_probe,
 		.config_init = mt7531_phy_config_init,
-		.read_status = mtk_gphy_cl22_read_status,
+		//.read_status = mtk_gphy_cl22_read_status,
 		/* Interrupts are handled by the switch, not the PHY
 		 * itself.
 		 */
```

cmonroe, 2024-09-03:
Made it to 20 boot loops without issue in our manual test so most likely it's resolved with the change above.

Our automated test setup also saw this issue quite often.. I'll report back when it finishes running as well but it takes 4-6hrs.

hnyman, 2024-09-03:
@dangowrt
I can confirm that so far my approach has worked well for me.
I have flashed the MT6000 router 4 times with different firmware images (of r27269-c6f04f718e, r27279-8b6c63cfaf twice and r27297-2bec6f48e6) and rebooted it otherwise some 5-10 times.
There has always been a proper WAN address from upstream DHCP, and normal traffic has been possible.

Likely the approach by cmonroe also works, as he just hides the same new functions in a less invasive way.

jebrossard, 2024-09-04:
So where do we go from here?  I'm hoping to test a build/snapshot with a fix as I had a 100% failure rate on restarts.

hnyman, 2024-09-04:
> hoping to test a build/snapshot with a fix

You can build by yourself, or you can download my test build from dropbox.
https://www.dropbox.com/scl/fo/51htl4fzsx8jhg5u70mln/h?rlkey=nn1lb2yxsnw1eqkaamtop3qjj&dl=0

There won't be any official buildbot build until @dangowrt merges something into the OpenWrt main repo.
(It might be that he wants to first mention the issue to the Linux kernel net-next mailing list, from where this whole new driver patchset is emerging. )

onlyfly34, 2024-09-04:
> Please try to remove only target/linux/mediatek/patches-6.6/733-10-net-phy-mediatek-Extend-1G-TX-RX-link-pulse-time.patch @onkyfly34 any ideas?

It's kind of weird since I've tested this patch with all my platforms with MT7531. I'll see what I can do in error log.

dangowrt, 2024-09-04:
@onlyfly34 It could be that the problem were are seeing is the result of the new `.read_status` function not playing well with https://github.com/openwrt/netifd/commit/68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd...

onlyfly34, 2024-09-04:
> You can build by yourself, or you can download my test build from dropbox. https://www.dropbox.com/scl/fo/51htl4fzsx8jhg5u70mln/h?rlkey=nn1lb2yxsnw1eqkaamtop3qjj&dl=0
>
> There won't be any official [...]

Hi @Neustradamus,
    Would you please help printing out some return values in mtk_gphy_cl22_read_status() and extend_an_new_lp_cnt_limit()? See the three lines added below:

```
static int extend_an_new_lp_cnt_limit(struct phy_device *phydev)
{
	int mmd_read_ret;
	u32 reg_val;
	int timeout;

	/* According to table 28-9 & Figure 28-18 in IEEE 802.3,
	 * link_fail_inhibit_timer of 10/100/1000 Mbps devices ranges from 750
	 * to "1000ms". Once MTK_PHY_FINAL_SPEED_1000 is set, it means that we
	 * enter "FLP LINK GOOD CHECK" state, link_fail_inhibit_timer starts and
	 * this PHY's 1G training starts. If 1G training never starts, we do
	 * nothing but leave.
	 */
	timeout = read_poll_timeout(mmd_read_ret = phy_read_mmd, reg_val,
				    (mmd_read_ret < 0) ||
				    reg_val & MTK_PHY_FINAL_SPEED_1000,
				    10000, 1000000, false, phydev,
				    MDIO_MMD_VEND1, MTK_PHY_LINK_STATUS_MISC);
+	phydev_err(phydev, "%s: Training Indicator: 0x%x\n", __func__, reg_val);
	if (mmd_read_ret < 0)
		return mmd_read_ret;

	if (!timeout) {
		/* Once we found MTK_PHY_FINAL_SPEED_1000 is set, no matter 1G
		 * AN is completed or not, we'll set AN_NEW_LP_CNT_LIMIT again
		 * and again.
		 */
		mtk_tr_modify(phydev, 0x0, 0xf, 0x3c, AN_NEW_LP_CNT_LIMIT_MASK,
			      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK, 0xf));
		mdelay(1500);

		timeout = read_poll_timeout(mtk_tr_read, reg_val,
					    (reg_val & AN_STATE_MASK) !=
					    (AN_STATE_TX_DISABLE <<
					     AN_STATE_SHIFT),
					    10000, 1000000, false, phydev,
					    0x0, 0xf, 0x2);
+		phydev_err(phydev, "%s: AN State: 0x%x\n", __func__, reg_val);
		if (!timeout) {
			mdelay(625);
			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
				      AN_NEW_LP_CNT_LIMIT_MASK,
				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
						 0x8));
			mdelay(500);
			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
				      AN_NEW_LP_CNT_LIMIT_MASK,
				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
						 0xf));
		} else {
			return -ETIMEDOUT;
		}
	}

	return 0;
}
```

```
int mtk_gphy_cl22_read_status(struct phy_device *phydev)
{
	int ret;

	ret = genphy_read_status(phydev);
	if (ret)
		return ret;

	if (phydev->autoneg == AUTONEG_ENABLE && !phydev->autoneg_complete) {
		ret = phy_read_paged(phydev, MTK_PHY_PAGE_EXTENDED_1,
				     MTK_PHY_AUX_CTRL_AND_STATUS);
		if (ret < 0)
			return ret;

		/* Once LP_DETECTED is set, it means that"ability_match" in
		 * IEEE 802.3 Figure 28-18 is set. This happens after we plug in
		 * cable. Also, LP_DETECTED will be cleared after AN complete.
		 */
		if (!FIELD_GET(MTK_PHY_LP_DETECTED_MASK, ret))
			return 0;

		ret = phy_read(phydev, MII_CTRL1000);
		if (ret & (ADVERTISE_1000FULL | ADVERTISE_1000HALF)) {
			ret = extend_an_new_lp_cnt_limit(phydev);
+			phydev_err(phydev, "%s: counter limit ret: %d\n", __func__, ret);
			if (ret < 0)
				return ret;
		}
	}

	return 0;
}
```
If possible, please provide full boot log to me.

onlyfly34, 2024-09-05:
> @onlyfly34 It could be that the problem were are seeing is the result of the new `.read_status` function not playing well with [openwrt/netifd@68c8a4f](https://github.com/openwrt/netifd/commit/68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd)...

Not sure why LAN's ethernet PHY driver's `read_status()` breaks WAN's DHCP...........

onlyfly34, 2024-09-05:
> @onlyfly34 Please find bootlogs with otherwise normal OpenWrt r27299-66559946ac, but with your debug code included.
> 
> https://gist.github.com/hnyman/7689040926c88db3b8cac833c86777f0
> 
> I have uploaded full system logs and kernel log (dmesg) from
> 
> * 1: first boot after flashing the firmware - detection ok.
> * 2: warm reboot - not ok until triggered by wan cable reconnect (2 b)
> * 3: cold reboot - not ok until triggered by wan cable reconnect

Thanks for your logs so much. It seems that all your LAN network phy drivers are brought up well...
However, I observe that mtk_gphy_cl22_read_status() returns -ETIMEDOUT in some users' cases.
So @hnyman , I think you can try if these 2 fix works:

Fix 1: Comment out the eight lines below in mtk_gphy_cl22_read_status() and see if fail case still happens
```
int mtk_gphy_cl22_read_status(struct phy_device *phydev)
{
	int ret;

	ret = genphy_read_status(phydev);
	if (ret)
		return ret;

	if (phydev->autoneg == AUTONEG_ENABLE && !phydev->autoneg_complete) {
		ret = phy_read_paged(phydev, MTK_PHY_PAGE_EXTENDED_1,
				     MTK_PHY_AUX_CTRL_AND_STATUS);
		if (ret < 0)
			return ret;

		/* Once LP_DETECTED is set, it means that"ability_match" in
		 * IEEE 802.3 Figure 28-18 is set. This happens after we plug in
		 * cable. Also, LP_DETECTED will be cleared after AN complete.
		 */
		//if (!FIELD_GET(MTK_PHY_LP_DETECTED_MASK, ret))
		//	return 0;

		//ret = phy_read(phydev, MII_CTRL1000);
		//if (ret & (ADVERTISE_1000FULL | ADVERTISE_1000HALF)) {
		//	ret = extend_an_new_lp_cnt_limit(phydev);
		//	if (ret < 0)
		//		return ret;
		//}
	}

	return 0;
}
```

Fix 2: Use original mtk_gphy_cl22_read_status() and just remove `return -ETIMEDOUT;` in extend_an_new_lp_cnt_limit() and see if  fail case still happens.
```
static int extend_an_new_lp_cnt_limit(struct phy_device *phydev)
{
	int mmd_read_ret;
	u32 reg_val;
	int timeout;

	/* According to table 28-9 & Figure 28-18 in IEEE 802.3,
	 * link_fail_inhibit_timer of 10/100/1000 Mbps devices ranges from 750
	 * to "1000ms". Once MTK_PHY_FINAL_SPEED_1000 is set, it means that we
	 * enter "FLP LINK GOOD CHECK" state, link_fail_inhibit_timer starts and
	 * this PHY's 1G training starts. If 1G training never starts, we do
	 * nothing but leave.
	 */
	timeout = read_poll_timeout(mmd_read_ret = phy_read_mmd, reg_val,
				    (mmd_read_ret < 0) ||
				    reg_val & MTK_PHY_FINAL_SPEED_1000,
				    10000, 1000000, false, phydev,
				    MDIO_MMD_VEND1, MTK_PHY_LINK_STATUS_MISC);
	if (mmd_read_ret < 0)
		return mmd_read_ret;

	if (!timeout) {
		/* Once we found MTK_PHY_FINAL_SPEED_1000 is set, no matter 1G
		 * AN is completed or not, we'll set AN_NEW_LP_CNT_LIMIT again
		 * and again.
		 */
		mtk_tr_modify(phydev, 0x0, 0xf, 0x3c, AN_NEW_LP_CNT_LIMIT_MASK,
			      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK, 0xf));
		mdelay(1500);

		timeout = read_poll_timeout(mtk_tr_read, reg_val,
					    (reg_val & AN_STATE_MASK) !=
					    (AN_STATE_TX_DISABLE <<
					     AN_STATE_SHIFT),
					    10000, 1000000, false, phydev,
					    0x0, 0xf, 0x2);
		/* If phy is linked up during this period, it won't enter
		 * "AN_STATE_TX_DISABLE" state anymore. In this case, we
		 * should't return -ETIMEDOUT
		 */
		if (!timeout) {
			mdelay(625);
			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
				      AN_NEW_LP_CNT_LIMIT_MASK,
				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
						 0x8));
			mdelay(500);
			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
				      AN_NEW_LP_CNT_LIMIT_MASK,
				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
						 0xf));
		}/* else {
			return -ETIMEDOUT;
		}*/
	}

	return 0;
}
```

Sorry to trouble you. But I can't reproduce the fail cases on my mt7986 platforms so far. Let me know the results so I can keep working on this.

hnyman, 2024-09-05:
@onlyfly34 

fix1:
seems to work ok:  first boot right after the flash, 2x warm reboot, 1x cold boot. All ok.
No real debug logs available, as all debug printouts were in the disabled function sections...

fix 2:
fails: 
* already at the first boot after flashing, there was no wan IP, and only cable disconnect/reconnect helped.  (log in gist)
* Curiously, after that, the router did not boot up from a "reboot" command nicely but seemed to get somehow stuck in the next boot process. (That happened once also yesterday during my testing).  After a really long boot process it finally booted up ok with IP address.  Long log in gist)
* Cold reboot from power-off then failed to get IP - disconnect helped

Logs in
https://gist.github.com/hnyman/01f1010f023a295551fd843399683d6f

OpenWRT-fanboy, 2024-09-05:
> Not sure why LAN's ethernet PHY driver's `read_status()` breaks WAN's DHCP...........

It's definitely an odd bug. Breaks my WiFi, too:

https://github.com/openwrt/openwrt/issues/16274#issuecomment-2323542947

onlyfly34, 2024-09-06:
@dangowrt @hnyman 
Looks like extend_an_new_lp_cnt_limit() breaks normal device link flow.
Let me add some fix here:
1. Remove unnecessary ETIMEDOUT return.
2. Change `mdelay()` to `msleep()` to allow CPU to be preempted. At first, I thought we need `mdelay()` so I can get precise delay time in `extend_an_new_lp_cnt_limit()`. However, `msleep()` can still work in this case after my test.
3. Call `genphy_read_status()` again so those delay functions won't affect normal devices in `extend_an_new_lp_cnt_limit().`

Final snippet
```
static int extend_an_new_lp_cnt_limit(struct phy_device *phydev)
{
	int ret;
	u32 reg_val;
	int timeout;
	int err;

	/* According to table 28-9 & Figure 28-18 in IEEE 802.3,
	 * link_fail_inhibit_timer of 10/100/1000 Mbps devices ranges from 750
	 * to "1000ms". Once MTK_PHY_FINAL_SPEED_1000 is set, it means that we
	 * enter "FLP LINK GOOD CHECK" state, link_fail_inhibit_timer starts and
	 * this PHY's 1G training starts. If 1G training never starts, we do
	 * nothing but leave.
	 */
	timeout = read_poll_timeout(ret = phy_read_mmd, reg_val,
				    (ret < 0) ||
				    reg_val & MTK_PHY_FINAL_SPEED_1000,
				    10000, 500000, false, phydev,
				    MDIO_MMD_VEND1, MTK_PHY_LINK_STATUS_MISC);
	if (ret < 0)
		return ret;

	if (!timeout) {
		/* Once we found MTK_PHY_FINAL_SPEED_1000 is set, no matter 1G
		 * AN is completed or not, we'll set AN_NEW_LP_CNT_LIMIT again
		 * and again.
		 */
		mtk_tr_modify(phydev, 0x0, 0xf, 0x3c, AN_NEW_LP_CNT_LIMIT_MASK,
			      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK, 0xf));
		msleep(1500);

		/* Read phy status again to make sure the following step won't
		 * affect normal devices.
		 */
		ret = genphy_read_status(phydev);
		if(ret)
			return ret;
		if(phydev->link)
			return 0;

		timeout = read_poll_timeout(mtk_tr_read, reg_val,
					    (reg_val & AN_STATE_MASK) !=
					    (AN_STATE_TX_DISABLE <<
					     AN_STATE_SHIFT),
					    10000, 1000000, false, phydev,
					    0x0, 0xf, 0x2);
		if (!timeout) {
			msleep(625);
			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
				      AN_NEW_LP_CNT_LIMIT_MASK,
				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
						 0x8));
			msleep(500);
			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
				      AN_NEW_LP_CNT_LIMIT_MASK,
				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
						 0xf));
		}
	}

	return 0;
}
```

@hnyman you may just replace this function and see if this fix works.

hnyman, 2024-09-06:
Thanks, I can test that earlier on Sunday, so I hope that somebody else can do that already earlier to provide you feedback.

pesa1234, 2024-09-06:
> @dangowrt @hnyman @OpenWRT-fanboy Looks like extend_an_new_lp_cnt_limit() breaks normal device link flow. Let me add some fix here:
> 
>     1. Remove unnecessary ETIMEDOUT return.
> 
>     2. Change `mdelay()` to `msleep()` to allow CPU to be preempted. At first, I thought we need `mdelay()` so I can get precise delay time in `extend_an_new_lp_cnt_limit()`. However, `msleep()` can still work in this case after my test.
> 
>     3. Call `genphy_read_status()` again so those delay functions won't affect normal devices in `extend_an_new_lp_cnt_limit().`
> 
>     4. Decrease `MTK_PHY_FINAL_SPEED_1000` polling timeout to 500ms.
> 
>     5. Replace `mmd_read_ret` with `ret`
> 
> 
> Final snippet
> 
> ```
> static int extend_an_new_lp_cnt_limit(struct phy_device *phydev)
> {
> +	int ret;
> 	u32 reg_val;
> 	int timeout;
> 
> 	/* According to table 28-9 & Figure 28-18 in IEEE 802.3,
> 	 * link_fail_inhibit_timer of 10/100/1000 Mbps devices ranges from 750
> 	 * to "1000ms". Once MTK_PHY_FINAL_SPEED_1000 is set, it means that we
> 	 * enter "FLP LINK GOOD CHECK" state, link_fail_inhibit_timer starts and
> 	 * this PHY's 1G training starts. If 1G training never starts, we do
> 	 * nothing but leave.
> 	 */
> 	timeout = read_poll_timeout(ret = phy_read_mmd, reg_val,
> 				    (ret < 0) ||
> 				    reg_val & MTK_PHY_FINAL_SPEED_1000,
> +				    10000, 500000, false, phydev,
> 				    MDIO_MMD_VEND1, MTK_PHY_LINK_STATUS_MISC);
> 	if (ret < 0)
> 		return ret;
> 
> 	if (!timeout) {
> 		/* Once we found MTK_PHY_FINAL_SPEED_1000 is set, no matter 1G
> 		 * AN is completed or not, we'll set AN_NEW_LP_CNT_LIMIT again
> 		 * and again.
> 		 */
> 		mtk_tr_modify(phydev, 0x0, 0xf, 0x3c, AN_NEW_LP_CNT_LIMIT_MASK,
> 			      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK, 0xf));
> +		msleep(1500);
> 
> +		/* Read phy status again to make sure the following step won't
> +		 * affect normal devices.
> +		 */
> +		ret = genphy_read_status(phydev);
> +		if (ret)
> +			return ret;
> +		if (phydev->link)
> +			return 0;
> 
> 		timeout = read_poll_timeout(mtk_tr_read, reg_val,
> 					    (reg_val & AN_STATE_MASK) !=
> 					    (AN_STATE_TX_DISABLE <<
> 					     AN_STATE_SHIFT),
> 					    10000, 1000000, false, phydev,
> 					    0x0, 0xf, 0x2);
> 		if (!timeout) {
> +			msleep(625);
> 			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
> 				      AN_NEW_LP_CNT_LIMIT_MASK,
> 				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
> 						 0x8));
> +			msleep(500);
> 			mtk_tr_modify(phydev, 0x0, 0xf, 0x3c,
> 				      AN_NEW_LP_CNT_LIMIT_MASK,
> 				      FIELD_PREP(AN_NEW_LP_CNT_LIMIT_MASK,
> 						 0xf));
> +		} /* else {
> +			return -ETIMEDOUT;
> +		} */
> 	}
> 
> 	return 0;
> }
> ```
> 
> @hnyman @OpenWRT-fanboy you may just replace this function and see if this fix works.

I applied this and it seems ok from my side. Here for the patch https://github.com/pesa1234/openwrt/commit/9686f293dd80630d5f0225a1c9bdf31f9d0ec841

onlyfly34, 2024-09-06:
> 
> I applied this and it seems ok from my side. Here for the patch [pesa1234@9686f29](https://github.com/pesa1234/openwrt/commit/9686f293dd80630d5f0225a1c9bdf31f9d0ec841)

Thanks. But you should directly remove these three lines in your commit.
```
/* else {
	return -ETIMEDOUT;
} */
```

If this fix also works on other boards, I'll also propose it to Linux net-next mainline.

pesa1234, 2024-09-06:
Done 

> Thanks. But you should directly remove these three lines in your commit.
>
> ```
> /* else {
> 	return -ETIMEDOUT;
> } */
> ```
>
> If this fix also works on other boards, I'll also propose it to Linux net-next [...]

Done, thanks, let's wait for other tester.

jglooije, 2024-09-07:
> Done
> 
> > > I applied this and it seems ok from my side. Here for the patch [pesa1234@9686f29](https://github.com/pesa1234/openwrt/commit/9686f293dd80630d5f0225a1c9bdf31f9d0ec841)
> > 
> > 
> > Thanks. But you should directly remove these three lines in your commit.
> > ```
> > /* else {
> > 	return -ETIMEDOUT;
> > } */
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > If this fix also works on other boards, I'll also propose it to Linux net-next mainline.
> 
> [4449586](https://github.com/openwrt/openwrt/commit/4449586deff63af21e650dd4432b23dff483b743) Done, thanks, let's wait for other testers.

Doesn't work for me. Still have to manual reset interface after boot up. GL.iNet GL-MT6000

pesa1234, 2024-09-07:
> Doesn't work for me. Still have to manual reset interface after boot up. GL.iNet GL-MT6000

Do you apply the patch? Can you import the other one with log to understand what happen?

Thanks

jglooije, 2024-09-07:
> Do you apply the patch? Can you import the other one with log to understand what happen?
>
> Thanks

I applied this [pesa1234@9686f29](https://github.com/pesa1234/openwrt/commit/9686f293dd80630d5f0225a1c9bdf31f9d0ec841)
If you have a patch with the debug code in it i can test it for you no problem.

pesa1234, 2024-09-07:
@jglooije please try to compile these sources tree https://github.com/pesa1234/openwrt/tree/next-r4.1.8.mtk.test

Or download from there the patch 733-10...

jglooije, 2024-09-07:
> @jglooije please try to compile these sources tree https://github.com/pesa1234/openwrt/tree/next-r4.1.8.mtk.test
> 
> Or download from there the patch 733-10...

Copied the 733-10 patch to my local tree and the wan port is working after boot and has internet connectivity.
Sat Sep  7 20:43:34 2024 kern.info kernel: [   50.276003] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 2.5Gbps/Full - flow control off
Thanks

hnyman, 2024-09-08:
@onlyfly34 
I am not sure if the "final snippet" works ok.
It failed right on my first boot after flashing MT6000. cable reconnect was needed.
The next warm reboot succeeded ok.
But the 2nd warm reboot again failed.

Both the exact patch that I used and the system log in first boot can be found in:

* patch: https://gist.github.com/hnyman/52709d4875717c0a5d373eef09dcc739#file-patch-for-733-10-patch
* log boot after flash: https://gist.github.com/hnyman/52709d4875717c0a5d373eef09dcc739#file-newfix-after-flash-fail-txt
(I left there the debug print statements into the patch.)
* log 2nd warm reboot: https://gist.github.com/hnyman/52709d4875717c0a5d373eef09dcc739#file-2nd-warm-reboot-failure-log

Ps. your final snippet apparently omitted some mmd_read_ret --> ret changes, so I got them from pesa123's patch.

pesa1234, 2024-09-08:
> @onlyfly34 I am not sure if the "final snippet" works ok. It failed right on my first boot after flashing MT6000. cable reconnect was needed. The next warm reboot succeeded ok. But the 2nd warm reboot again failed.
> 
> 
> 
> Ps. your final snippet apparently omitted some mmd_read_ret --> ret changes, so I got them from pesa123's patch.

@onlyfly34 

Thanks. I also add the debug messages as @hnyman in any case you need it --> https://raw.githubusercontent.com/pesa1234/MT6000_cust_build/main/testing/log/bootlog.log

Thanks a lot to all.

Andrea

onlyfly34, 2024-09-09:
> @onlyfly34 I am not sure if the "final snippet" works ok. It failed right on my first boot after flashing MT6000. cable reconnect was needed. The next warm reboot succeeded ok. But the 2nd warm [...]

This is a little bit strange. I don't see "AN state xxxxx" message in your log. That means your lan2~lan5 should already link up at last and `extend_an_new_lp_cnt_limit()` won't affect your MT7531 ethernet phy anymore.

Have you ever tried to boot up device with no cables connected to your LAN?

ptpt52, 2024-09-09:
revert commit 1069514978c419cc5bdfc18c5670b46d0443a686
and the issue fixed for me

pesa1234, 2024-09-09:
> revert commit [1069514](https://github.com/openwrt/openwrt/commit/1069514978c419cc5bdfc18c5670b46d0443a686) and the issue fixed for me

If you apply what @onlyfly34 suggest do you solve?

hnyman, 2024-09-10:
@onlyfly34 

I tried your suggestion of booting with no LAN cables, with interesting results:
* 2 boots with 0 LAN cables (and then via WiFi) = IP ok.
* 1 boot with 2 LAN cables attached = IP ok
* 2 boots with 4 LAN cables attached = NO IP until WAN cable disconnect/reconnect

https://gist.github.com/hnyman/34bd5f19a0a776ba70b69839ed0cfe4c

(same build as in Sunday)

onlyfly34, 2024-09-10:
> @hnyman Very interesting results, as I have all FIVE LAN cables attached, including LAN1 at 2.5 Gb which is connected to another MT6000 functioning as a dumb AP. I never get a WAN IP and have to down and up eth1 to get a DHCP lease. ("ifconfig eth1 down" followed by "ifconfig eth1 up").

Which build do you use?

onlyfly34, 2024-09-10:
> @jebrossard Please test with 0-5 cables connected to see if that helps for getting wan IP. @onlyfly34 @dangowrt and @nbd168 might be interested of the results.
> 
> Could this be some kind of race condition between multiple LAN interfaces, causing somehow harm for the wan interface ???
> 
> Ps.
> 
> > including LAN1 at 2.5 Gb which is connected
> 
> I have avoided using that LAN1.

@hnyman 
IMO, in Linux, phy state machine is protected by mutex lock so read_status() function of each phy shouldn't have race condition.

In your enviroment:
eth1 (RTL8221B-VB-CG 2.5Gbps PHY) -> 2.5Gbps (WAN)
lan1 (RTL8221B-VB-CG 2.5Gbps PHY) -> 2.5Gbps
lan2 (mt7531 port 0) -> 1Gbps
lan3 (mt7531 port 1) -> 10Mbps
lan4 (mt7531 port 2) -> 1Gbps
lan5 (mt7531 port 3) -> 100Mbps

Also I observe quite a lot `extend_an_new_lp_cnt_limit()` lines in your logs, which means that this phy is trying to link at 1G but DSP fails to start training. What is link partner connected to your lan4? How long is that cable?
```
Tue Sep 10 17:29:02 2024 kern.err kernel: [   30.252512] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 17:29:02 2024 kern.err kernel: [   30.262454] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:02 2024 kern.info kernel: [   30.262666] mt7530-mdio mdio-bus:1f lan4: Link is Down
Tue Sep 10 17:29:03 2024 kern.err kernel: [   30.782564] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 17:29:03 2024 kern.err kernel: [   30.791622] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:03 2024 kern.err kernel: [   31.317149] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 17:29:03 2024 kern.err kernel: [   31.326191] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:04 2024 kern.err kernel: [   31.846427] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 17:29:04 2024 kern.err kernel: [   31.855460] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:04 2024 kern.info kernel: [   32.170345] mt7530-mdio mdio-bus:1f lan3: Link is Up - 10Mbps/Full - flow control rx/tx
Tue Sep 10 17:29:04 2024 kern.err kernel: [   32.375620] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 17:29:04 2024 kern.err kernel: [   32.384648] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:05 2024 kern.err kernel: [   32.902525] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 17:29:05 2024 kern.err kernel: [   32.911554] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:05 2024 kern.info kernel: [   32.920028] br-lan: port 4(lan4) entered disabled state
Tue Sep 10 17:29:05 2024 kern.info kernel: [   32.925321] br-lan: port 3(lan3) entered blocking state
Tue Sep 10 17:29:05 2024 kern.info kernel: [   32.930537] br-lan: port 3(lan3) entered forwarding state
Tue Sep 10 17:29:06 2024 kern.err kernel: [   33.451772] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 17:29:06 2024 kern.err kernel: [   33.460814] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:06 2024 kern.err kernel: [   33.975092] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x10
Tue Sep 10 17:29:06 2024 kern.err kernel: [   33.983961] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 17:29:07 2024 kern.err kernel: [   34.511747] MediaTek MT7531 PHY mt7530-0:02: extend_an_new_lp_cnt_limit: Training Indicator: 0x10
Tue Sep 10 17:29:07 2024 kern.err kernel: [   34.525026] MediaTek MT7531 PHY mt7530-0:02: mtk_gphy_cl22_read_status: counter limit ret: 0
```

Would you please test if 4 scenario works? (If scenario 1 & 2 are ok, we can almost conclude `extend_an_new_lp_cnt_limit()` works in your 10Mbps/100Mbps connect)
Scenario 1 - Connect only lan3 and see if WAN gets IP
Scenario 2 - Connect only lan5 and see if WAN gets IP
Scenario 3 - Connect all LAN&WAN cables **except lan4**  and see if WAN gets IP

jebrossard, 2024-09-10:
@nbd168 @dangowrt
Okay, I did a quick test, as I am working from home...
OpenWrt SNAPSHOT r27317-8e6543fae5 
All LAN cables disconnected, I get a WAN IP via DHCP with EVERY reboot.
All LAN cables connected, I NEVER get a WAN IP.

@onlyfly34 Why do you avoid using LAN1?  It has been working great for me.

This testing with multiple LAN connections may explain why some never see this issue, while for others it is seen with every boot.

pesa1234, 2024-09-10:
As soon as I connect all LAN cables I also get the issue

hnyman, 2024-09-10:
Well, regarding my connections:

- lan2 (mt7531 port 0) -> 1Gbps  = switch
- lan3 (mt7531 port 1) -> 10Mbps  = NAS turned off (normally gigabit connection when on)
- lan4 (mt7531 port 2) -> 1Gbps  = PC, connected with 2 x 1.5 m long cables, with an extension piece in the middle (so that I can easier attach PC to other routers, if needed. (Has worked ok with WNDR3700, R7800, DL-WRX36 and MT6000 until this new driver)
- lan5 (mt7531 port 3) -> 100Mbps  = printer

Scenarios:
* only lan3 (NAS turned off): ok  (13 seconds from first eth0 action to WAN IP)
* only lan5 (printer): ok  (13 seconds)
* Lan2 (switch), lan3 and lan5: ok  (39 seconds)
* only lan4 (PC): ok (39 seconds)
* only lan2, lan4, lan5:  ok (first ok in 25 seconds, lost but again ok in 37 seconds)
* only lan3 lan4 lan5: ok (first time up after 40 seconds, but lost, up again in 50 seconds)
* lan2 lan3 lan4 lan5: this time ok, but after 4 minutes  (208 seconds). Actually first time shortly ok after 30 seconds, but eth1 went again down and recovered several times during the next few minutes. During the long time there is also no LAN connectivity as also the LAN ports cycle down/up.

I tried booting a second time with all 4 cables attached, and this time the router ended up in a cycle, where lan3 lan4 and lan5 each came down and recovered repeatedly in a looping fashion. I managed to stop that by disconnecting lan5.
Log also about that is also attached.

So far it seems that something goes awry when 4 cables are connected. (Note: I haven't tried all 5 LAN ports...)

Log extracts of most of the attempts in github:
https://gist.github.com/hnyman/066d336785dd047d3380d78fc98c69e0

Just thinking aloud: In addition to this new phy driver, another recent addition is also @dangowrt commit for netifd https://github.com/openwrt/netifd/commit/68c8a4f94cd3c 
That was already speculated early in this issue: https://github.com/openwrt/openwrt/issues/16281#issuecomment-2329405813 , https://github.com/openwrt/openwrt/issues/16281#issuecomment-2322598996

Log extract from the lan2 lan3 lan4 lan5 try until the cycling starts:

```Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.635224] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.654672] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.663134] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.665721] mt7530-mdio mdio-bus:1f lan1: configuring for phy/2500base-x link mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.721476] br-lan: port 1(lan1) entered blocking state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.726706] br-lan: port 1(lan1) entered disabled state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.731999] mt7530-mdio mdio-bus:1f lan1: entered allmulticast mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.738256] mtk_soc_eth 15100000.ethernet eth0: entered allmulticast mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.747807] mt7530-mdio mdio-bus:1f lan1: entered promiscuous mode
Tue Sep 10 20:52:21 2024 daemon.notice netifd: Interface 'lan' is enabled
Tue Sep 10 20:52:21 2024 daemon.notice netifd: Interface 'lan' is setting up now
Tue Sep 10 20:52:21 2024 daemon.notice netifd: Interface 'lan' is now up
Tue Sep 10 20:52:21 2024 kern.info kernel: [   19.767458] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
Tue Sep 10 20:52:21 2024 kern.err kernel: [   20.281281] MediaTek MT7531 PHY mt7530-0:00: extend_an_new_lp_cnt_limit: Training Indicator: 0x8050
Tue Sep 10 20:52:21 2024 kern.err kernel: [   20.290320] MediaTek MT7531 PHY mt7530-0:00: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.298768] br-lan: port 2(lan2) entered blocking state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.303983] br-lan: port 2(lan2) entered disabled state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.309221] mt7530-mdio mdio-bus:1f lan2: entered allmulticast mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.317099] mt7530-mdio mdio-bus:1f lan2: entered promiscuous mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.331637] mt7530-mdio mdio-bus:1f lan3: configuring for phy/gmii link mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.339402] br-lan: port 3(lan3) entered blocking state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.344662] br-lan: port 3(lan3) entered disabled state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.349911] mt7530-mdio mdio-bus:1f lan3: entered allmulticast mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.357986] mt7530-mdio mdio-bus:1f lan3: entered promiscuous mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.373238] mt7530-mdio mdio-bus:1f lan4: configuring for phy/gmii link mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.380876] br-lan: port 4(lan4) entered blocking state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.386098] br-lan: port 4(lan4) entered disabled state
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.391356] mt7530-mdio mdio-bus:1f lan4: entered allmulticast mode
Tue Sep 10 20:52:21 2024 kern.info kernel: [   20.399615] mt7530-mdio mdio-bus:1f lan4: entered promiscuous mode
Tue Sep 10 20:52:22 2024 kern.info kernel: [   20.415876] mt7530-mdio mdio-bus:1f lan5: configuring for phy/gmii link mode
[... cut ...]
Tue Sep 10 20:53:13 2024 kern.err kernel: [   71.424868] MediaTek MT7531 PHY mt7530-0:01: extend_an_new_lp_cnt_limit: Training Indicator: 0x50
Tue Sep 10 20:53:13 2024 kern.err kernel: [   71.433739] MediaTek MT7531 PHY mt7530-0:01: mtk_gphy_cl22_read_status: counter limit ret: 0
Tue Sep 10 20:53:13 2024 daemon.notice netifd: Network device 'eth1' link is down
Tue Sep 10 20:53:13 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Sep 10 20:53:13 2024 daemon.notice netifd: Interface 'wan6' has link connectivity loss
```

pesa1234, 2024-09-10:
So, I get the issue if all lan are connected and one of them is at 10Mbps.

Here are present log with and without the issue.
When I get the issue, my NUC is off and LAN is 10Mbps, when I didn't get it is ON at 1Gbps.
Log https://github.com/pesa1234/MT6000_cust_build/tree/main/testing/log

hnyman, 2024-09-10:
I tested with my lan3 NAS on, so that lan2 lan3 lan4 are 1Gbit and lan5 printer is 100 Mbit:
Still no wan IP until a wan cable disconnect/reconnect.

pesa1234, 2024-09-10:
My LAN connection speed are in order: 2.5 - 1 - 1 - 100 - 10, when I get issue. 2.5 - 1 - 1 - 100 - 1, when Ok but I test only two times.

Thanks guys for take care about this issue

hnyman, 2024-09-11:
I made a test with all of lan2, lan3, lan4 and lan5 being connected with a 1 Gbit link.
Still the interface cycling activity until I disconnected one (lan5). Then it stabilised

https://gist.github.com/hnyman/b37d0e0bff0ccbd4d7a619d345878660

pesa1234, 2024-09-11:
@hnyman I'll try this evening and I let you know!

Thanks

hnyman, 2024-09-11:
> maybe it is not related only to our device what do you think?

No, likely it is about at least the whole mediatek target, but might affect others. 
There are possibly related issue reports which mention e.g. E8450 and AX6000:
#16315  and  #16274

pesa1234, 2024-09-11:
@hnyman THANKS!

10 boot with all cables connected and the problem seems disappeared.

dangowrt, 2024-09-11:
I also suspected the presence of that netifd commit was the main difference the setup of users reporting this problem and the setup @onlyfly34 has tested with the series. The question remains why reverting https://github.com/openwrt/openwrt/commit/1069514978c419cc5bdfc18c5670b46d0443a686 solves the issue and if anything can be done to keep the new behavior in netifd which is required to make sure ethtool settings flow control or speed settings are applied also for non-DSA PHYs as well as PHYs on SFP modules. Matching on flags *`IFF_UP` AND NOT `IFF_LOWER_UP`* seemed a bit fishy to begin with, but I didn't find a better way to apply ethtool settings on PHY attachment (there just isn't any dedicated event for that), nor a way to setup ethtool settings in a meaningful way *before* the PHY is attached *and keep the settings once the PHY does get attached*.
Any other solutions for that problem, in kernel or userlands (ie. netifd) would be welcome, of course.

cmonroe, 2024-09-12:
> I also suspected the presence of that netifd commit was the main difference the setup of users reporting this problem and the setup @onlyfly34 has tested with the series. The question remains why reverting [1069514](https://github.com/openwrt/openwrt/commit/1069514978c419cc5bdfc18c5670b46d0443a686) solves the issue and if anything can be done to keep the new behavior in netifd which is required to make sure ethtool settings flow control or speed settings are applied also for non-DSA PHYs as well as PHYs on SFP modules. Matching on flags _`IFF_UP` AND NOT `IFF_LOWER_UP`_ seemed a bit fishy to begin with, but I didn't find a better way to apply ethtool settings on PHY attachment (there just isn't any dedicated event for that), nor a way to setup ethtool settings in a meaningful way _before_ the PHY is attached _and keep the settings once the PHY does get attached_. Any other solutions for that problem, in kernel or userlands (ie. netifd) would be welcome, of course.

My first comment in this thread I had reverted the netifd change and it seemed to mostly fix (workaround?) the issue for me with the original patch set in OpenWrt master. The test system where I could see the problem most often was a SDG-8734v (MT7988A w/10G SFP WAN + 10G copper LAN + 3x 1G LAN ports on the MT7988 switch). The SFP+ WAN port has a generic 10G copper SFP, lan1 (AQR113C 10G) has a 10G link partner and lan2 (MT7988 1G) has a 1G link partner. In this case, only 1 port (lan2) should be using the new read_status() callback.

Unlike some have posted above, I did not see constant link bouncing on this system. udhcpc would just never start on the wan even though it was in up/running state with tx/rx packets. If I manually started udhcpc on the wan it would immediately get an IP. Looking at the logs I saw something like his:
```
Thu Aug 29 12:47:16 2024 kern.info kernel: [    7.375626] mtk_soc_eth 15100000.ethernet wan: mediatek frame engine at 0xffffffc083300000, irq 103
Thu Aug 29 12:47:16 2024 kern.info kernel: [   21.592688] mtk_soc_eth 15100000.ethernet wan: switched to inband/10gbase-r link mode
Thu Aug 29 12:47:16 2024 kern.info kernel: [   23.824964] mtk_soc_eth 15100000.ethernet wan: configuring for inband/10gbase-r link mode
Thu Aug 29 12:47:18 2024 kern.info kernel: [   25.769047] mtk_soc_eth 15100000.ethernet wan: Link is Up - 10Gbps/Full - flow control off
Thu Aug 29 12:47:18 2024 kern.info kernel: [   26.292085] mtk_soc_eth 15100000.ethernet wan: Link is Down
Thu Aug 29 12:47:22 2024 kern.info kernel: [   30.235191] mtk_soc_eth 15100000.ethernet wan: configuring for inband/10gbase-r link mode
Thu Aug 29 12:47:22 2024 daemon.notice netifd: Interface 'wan' is enabled
Thu Aug 29 12:47:23 2024 kern.info kernel: [   31.317550] mtk_soc_eth 15100000.ethernet wan: Link is Up - 10Gbps/Full - flow control off
```

Which means the netifd messages for link up, connectivity and setting up are all missing.  e.g. the same flow produces the following logs in a good case:
```
Wed Sep 11 15:10:16 2024 kern.info kernel: [    7.375954] mtk_soc_eth 15100000.ethernet wan: mediatek frame engine at 0xffffffc083280000, irq 103
Wed Sep 11 15:10:16 2024 kern.info kernel: [   21.744643] mtk_soc_eth 15100000.ethernet wan: switched to inband/10gbase-r link mode
Wed Sep 11 15:10:16 2024 kern.info kernel: [   24.097333] mtk_soc_eth 15100000.ethernet wan: configuring for inband/10gbase-r link mode
Wed Sep 11 15:10:21 2024 kern.info kernel: [   28.876682] mtk_soc_eth 15100000.ethernet wan: configuring for inband/10gbase-r link mode
Wed Sep 11 15:10:21 2024 daemon.notice netifd: Interface 'wan' is enabled
Wed Sep 11 15:10:25 2024 daemon.notice netifd: Network device 'wan' link is up
Wed Sep 11 15:10:25 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Wed Sep 11 15:10:25 2024 daemon.notice netifd: Interface 'wan' is setting up now
Wed Sep 11 15:10:25 2024 kern.info kernel: [   33.408478] mtk_soc_eth 15100000.ethernet wan: Link is Up - 10Gbps/Full - flow control off
Wed Sep 11 15:10:25 2024 daemon.notice netifd: wan (6754): udhcpc: started, v1.36.1
```

I'm not sure if the events which drive those missing messages were not posted by the kernel or if netifd somehow dropped them but I thought it was an interesting clue. Note the time delta between the final " wan: configuring for inband/10gbase-r link mode" message and "wan: Link is Up - 10Gbps/Full - flow control off". In the failure case it's just about 1 second. In the good case it's almost 5 seconds.

For the last week or so we've been using OpenWrt master (with the netifd change mentioned here) but I just commented out the calls to register read_status callbacks in the Mediatek PHY drivers. So far I've not heard of any other instances of the issue occurring in automated test networks or otherwise.

This is a total guess but I assume there's some kind of timing conflict between the work happening in the read_status callback and netifd trying to re-configure the interface on IFF_UP. Since mtk_gphy_cl22_read_status() / extend_an_new_lp_cnt_limit() appears to write some phy registers, does it need some kind of lock to prevent a race between what's happening in extend_an_new_lp_cnt_limit() and some other areas of phylink?

hnyman, 2024-09-12:
@dangowrt 
Could you please revert the problematic change from netifd & OpenWrt, while the thinking and development of an actually working change there goes on...

[8 more replies]

closed 2024-09-13 by pesa1234