openwrt/openwrt #16292: KuWfi  (YunCore) AX835 LEDs not working

matlab22, 2024-08-30:
### Describe the bug

I can not configure the LED behavior of the AX835 router. I can set them up using LUCI but no LED is turning on when expected.

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mediatek/filogic

### Device

YunCore AX835

### Image kind

Official downloaded image

### Steps to reproduce

```
root@OpenWrt:/#ls/sys/class/leds/ 
blue:wifi5   green:wifi2  hwwatchdog   mt76-phy0    mt76-phy1    red:system
root@OpenWrt:/# echo 0 > /sys/class/leds/blue:wifi5/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/blue:wifi5/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/green:wifi2/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/blue:wifi5/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/green:wifi2/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/hwwatchdog/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/mt76-phy0/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/mt76-phy1/brightness
root@OpenWrt:/# echo 1 > /sys/class/leds/red:system/brightness
```
nothing is turned on.
I also tried:
```
echo none > /sys/class/leds/red:system/trigger
echo 1 > /sys/class/leds/red:system/brightness
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2024-09-07:
@yogo1212: Can you check this ticket?

yogo1212, 2024-09-09:
hwwatchdog is not a led.
mt76-phy0 and mt76-phy1 are not wired to anything, afaik.
green behaves normally after setting its trigger to `none`.

blue and red light up briefly after setting brightness to 1 but go back to being dark immediately afterwards.
i'm sure this was working at some point in my staging branch.
@matlab22 can you test whether there is blinking for you as well?

@Neustradamus you appear to be very interested - are you using ax835 as well?

yogo1212, 2024-11-01:
another issue (regular resets) is forcing me to look back into device support for the yuncore ax835.

the problem lies with the [voltage regulator](https://github.com/openwrt/openwrt/blob/main/target/linux/mediatek/dts/mt7981b-yuncore-ax835.dts#L33).

i suspect gpio really controls a "switch" and there's a transistor of some kind sitting behind that gpio. the regulator doesn't trigger it. maybe it's not a FET and there's not enough current. pulling the pin up using gpio does enable the functioning of all LEDs (regardless of how they're driven).

yogo1212, 2024-11-02:
there is no voltage at all

yogo1212, 2024-11-02:
found it!

Documentation/devicetree/bindings/regulator/fixed-regulator.yaml
```
  enable-active-high:
    description:
      Polarity of GPIO is Active high. If this property is missing,
      the default assumed is Active low.
    type: boolean

yogo1212, 2024-11-02:
positive test - https://github.com/openwrt/openwrt/pull/16839

Neustradamus, 2024-11-03:
@matlab22: Can you check the @yogo1212 PR?
- https://github.com/openwrt/openwrt/pull/16839

closed 2024-11-03 by openwrt-bot

matlab22, 2024-11-05:
LEDs are working on r28009-11ba2b141f thank you :)

commit d64e1ed4 (2024-11-03): mediatek: YunCore AX835: fix voltage regulator

Specifying GPIO_ACTIVE_HIGH on the GPIO for the voltage regulator doesn't
suffice. The regulator itself requires enable-active-high to be set.

Fixes: #16292

Link: https://github.com/openwrt/openwrt/pull/16839
commit 6bc930c4 (2024-11-04): mediatek: YunCore AX835: fix voltage regulator

Specifying GPIO_ACTIVE_HIGH on the GPIO for the voltage regulator doesn't
suffice. The regulator itself requires enable-active-high to be set.

Fixes: #16292

Link: https://github.com/openwrt/openwrt/pull/16839