openwrt/openwrt #16294: cdc-ether modem USB 3.0 5G

antoniovalenzuela, 2024-08-31:
Hi,

Openwrt-23.05 branch (git-24.212.79335-cdbe903)
Linksys WRT3200ACM

I have a ZTE modem/router USB 3.0 network 4G/5G connected by USB to a computer it negotiates at high Gbps speed. 

I have a question with CDC Ether, because my router shows 480 Mbps. Is the speed limit related to the driver? 

Can I increase the speed? 

It is connected to the Linksys USB 2.0/3.0 port 

kmod-usb-net-cdc-ether 5.15.162-1

S:  Manufacturer=ZTE,Incorporated
S:  Product=ZTE Mobile Broadband
S:  SerialNumber=MU5120ZTED0000000
C:* #Ifs= 3 Cfg#= 1 Atr=80 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=06 Prot=00 Driver=cdc_ether
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=32ms
I:  If#= 1 Alt= 0 #EPs= 0 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_ether
I:* If#= 1 Alt= 1 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=cdc_ether
E:  Ad=8e(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=0f(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=125us
T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh=
1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.15
S:  Manufacturer=Linux 5.15.162 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=f10f8000.usb3
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hubE:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480
M
    |__ Port 1: Dev 5, If 0, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 5, If 1, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 5, If 2, Class=, Driver=usb-storage, 480M

namiltd, 2024-08-31:
480M is the bandwidth of usb3.0, you can't get more than that

vincejv, 2024-09-03:
Attach dmesg log to check if USB device/cable is really USB 3.0.
Also double check if the cable and port is USB 3.0, I'm having the same issue with ZTE F50, turns out this was falsely advertised as a USB 3.0 device

antoniovalenzuela, 2024-09-03:
yes, it is 3.0, I used different cables.

Computer 4.2 Gbps
[image]

The Linksys port is 2.0/3.0
[165212.680144] usb 2-1: new high-speed USB device number 8 using xhci-hcd
[165212.863754] usb-storage 2-1:1.0: USB Mass Storage device detected
[165212.870084] usb-storage 2-1:1.0: Quirks match for vid 19d2 pid 1225: 1
[165212.876957] scsi host2: usb-storage 2-1:1.0
[165215.030949] usb 2-1: new high-speed USB device number 9 using xhci-hcd
[165215.232678] cdc_ether 2-1:1.0 eth1: register 'cdc_ether' at usb-f10f8000.usb3-1, ZTE CDC Ethernet Device, 54:1f:8d:db:9d:89
[165215.247046] usb-storage 2-1:1.2: USB Mass Storage device detected
[165215.256857] scsi host2: usb-storage 2-1:1.2

I'm wondering if it's the cdc_ether driver or because the router port is not a real 3.0.

antoniovalenzuela, 2024-09-03:
dmesg | grep xhci

[    2.953454] xhci-hcd f10f8000.usb3: new USB bus registered, assigned bus number 3
[    2.960979] xhci-hcd f10f8000.usb3: **Host supports USB 3.0 SuperSpeed**
[   16.114845] cdc_ether 2-1:1.0 eth1: register 'cdc_ether' at usb-f10f8000.usb3-1, ZTE CDC Ethernet Device, 54:1f:8d:db:9d:89

lsusb -t (Bus 03 5000M - Bus 02 480M)

/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 3, If 0, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 3, If 1, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 3, If 2, Class=, Driver=usb-storage, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=orion-ehci/1p, 480M

Maybe it is necessary to force for dual use 2.0/3.0 ¿...?

vincejv, 2024-09-03:
yep definitely has an issue with the usb driver/kernel or device

you should be seeing "new SuperSpeed USB device number xxx" here in the logs, not high-speed

> [165212.680144] usb 2-1: **new high-speed USB** device number 8 using xhci-hcd
> [165212.863754] usb-storage 2-1:1.0: USB Mass Storage device detected
> [165212.870084] usb-storage 2-1:1.0: Quirks match for vid 19d2 pid 1225: 1

raenye, 2024-10-27:
If you use a USB Ethernet dongle (GbE or 2.5GbE), do you get USB3 speeds?

antoniovalenzuela, 2024-11-08:
> If you use a USB Ethernet dongle (GbE or 2.5GbE), do you get USB3 speeds?

I didn't understand your question exactly.

The ZTE modem/router has a USB-3 port and WiFi networks (5/2.4 GHz)

I have it connected via USB to the WRT-3200ACM router (CDC Ether)

raenye, 2024-11-08:
> I have it connected via USB to the WRT-3200ACM router (CDC Ether)

I was asking about this part. Does the WRT-3200ACM show USB3 speeds with other USB3 devices? (e.g., mass storage, ethernet dongle, etc.)

luisimasd, 2024-12-02:
Identical issue here

```
OpenWrt 22.03.5, r20134-5f15225c1e
Linux OpenWrt 5.10.176 #0 SMP Thu Apr 27 20:28:15 2023 armv7l GNU/Linux
kmod-usb3 - 5.10.176-1 
kmod-usb-net-cdc-ether - 5.10.176-1 
kmod-usb-xhci-hcd - 5.10.176-1
```

```
lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 25, If 0, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 25, If 1, Class=, Driver=cdc_ether, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 32, If 0, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 32, If 1, Class=, Driver=cdc_ether, 480M
    |__ Port 1: Dev 32, If 2, Class=, Driver=, 480M
```

2 LTE usb devices connected to both usb3.0 ports. however they are not working at full speed

```
lsusb -v

Bus 003 Device 025: ID 12d1:1c52 HUAWEI Mobile HUAWEI Mobile
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            2
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x12d1
  idProduct          0x1c52
  bcdDevice            1.02
  iManufacturer           1 HUAWEI Mobile
  iProduct                2 HUAWEI Mobile
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0047
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2
      bInterfaceSubClass      6
      bInterfaceProtocol      0
      iInterface              5 CDC Ethernet Control Model (ECM)
      CDC Header:
        bcdCDC               1.10
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      CDC Ethernet:
[... 346 lines cut ...]
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat           2 micro seconds
Device Status:     0x0000
  (Bus Powered)
```

Looks correct to me, operating at 5000mpbs as expected.

There is something around cdc_ether devices or similar.

luisimasd, 2024-12-02:
GPT info, that could explain it.

[image]

vincejv, 2024-12-03:
> In relation to another USB3.0 device...
I unplugged a LTE mode and replaced it with a USB3.0 usb pendrive
lsusb output...

@luisimasd what about dmesg log?

vincejv, 2024-12-03:
dmesg log for the USB 3 modem and USB 3 (flash drive or other usb3 device)