openwrt/openwrt #16296: TP-Link RE210 v1 doesn't boot when sysupgraded to versions 22.03.7, 23.05.3 and 23.05.4

noobtheproswag, 2024-08-31:
### Describe the bug

[This issue has been opened after I made a mistake in the creation of the [other one](https://github.com/openwrt/openwrt/issues/16295). The contents of the issue are the exact same otherwise.]

RE210 doesn't boot on versions specified in the title. "Doesn't boot" sounds very vague so I will attach the serial console logs here, for each of the three versions. 

[Serial log for 22.03.7.txt](https://github.com/user-attachments/files/16824172/Serial.log.for.22.03.7.txt)
[Serial log for 23.05.3.txt](https://github.com/user-attachments/files/16824181/Serial.log.for.23.05.3.txt)
[Serial log for 23.05.4.txt](https://github.com/user-attachments/files/16824183/Serial.log.for.23.05.4.txt)

They all get stuck in "Starting kernel..."
I sha256sum'd these log files and they all have the exact same hash, which means logs are exactly the same.

For comparison, here is the serial log for version 22.03.2 which works fine. This log also contains the flashing operation. (I flashed it in the exact same way I flashed all of the versions I tested.)

[Serial log for 22.03.2.txt](https://github.com/user-attachments/files/16824271/Serial.log.for.22.03.2.txt)

I tested multiple versions with the method I specified in the "Steps to reproduce" section, here are the test results: (By test I mean whether it boots or not)

22.03.x Series:
22.03.2: Boots!
22.03.3: Boots!
22.03.4: Boots!
22.03.5: Boots!
22.03.6: Boots!
22.03.7: Doesn't boot!

23.05.x Series:
23.05.0: Boots!
23.05.1: Boots!
23.05.2: Boots!
23.05.3: Doesn't boot!
23.05.4: Doesn't boot!

Note that I have successfully tftp booted the corresponding kernel images for all three of the versions that don't boot when sysupgraded to.

The latest snapshot as of writing (2024-08-29 11:21:19 r27243-dd1bfbe971) boots fine when sysupgraded to in the way described below.

Edit: github actions marked my issue as invalid, so I have moved the original contents here and only left one release and one version.

Versions: r20341-591b7e93d3, r23809-234f1a2efa and r24012-d8dd03c46f
Releases: 22.03.7, 23.05.3 and 23.05.4

### OpenWrt version

r20341-591b7e93d3

### OpenWrt release

22.03.7

### OpenWrt target/subtarget

ramips/mt7620

### Device

TP-Link RE210 v1

### Image kind

Official downloaded image

### Steps to reproduce

All tests have been performed with serial console hooked up. I flashed each version from [23.05.4 kernel image](https://downloads.openwrt.org/releases/23.05.4/targets/ramips/mt7620/openwrt-23.05.4-ramips-mt7620-tplink_re210-v1-initramfs-kernel.bin) (which I booted off of tftp and with failsafe mode enabled each time.)

I have downloaded all sysupgrade images directly from the OpenWRT firmware selector.

I transferred the images to the RE210 with:

scp -O ./[image name].bin  root@192.168.1.1:/tmp

and verified their sha256 on the device by doing:

sha256sum /tmp/[image name].bin

and comparing the output to the original one from the firmware selector webpage I downloaded them from.

I flashed them with:

sysupgrade -n -p -v /tmp/[image name].bin

### Actual behaviour

Device gets stuck at "Starting kernel ..." and doesn't do anything else.

### Expected behaviour

I would expect it to boot like the other versions I tested which do boot.

### Additional info

Please keep in mind that this is my first time reporting an issue to OpenWrt. I have read [OpenWrt bugs page](https://openwrt.org/bugs) and [Simon Tatham's "How to report bugs effectively"](https://www.chiark.greenend.org.uk/~sgtatham/bugs.html) but my report might be lacking necessary details. If there is any other info I need to provide, or if I need to perform more testing, please describe me what I need to do, and I will try my best to do so. Thank you for your amazing work, and for being patient with me! :D

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-08-31:
Check again version 23.05.04 but without LuCI.
On the page https://firmware-selector.openwrt.org/?version=23.05.4&target=ramips%2Fmt7620&id=tplink_re210-v1 remove luci package and download the image.

noobtheproswag, 2024-09-01:
I removed the luci package from the list, then built it and sysupgraded to it in the same way. It still gets stuck in "Starting kernel ...".

namiltd, 2024-09-01:
I suspected that the system file was too large. Removing LuCI was supposed to slim it down. Unfortunately, that was a false lead.

andyboeh, 2024-09-01:
I'm seeing the exact same issue on the RE200v1. That's kind of bad, as the RE200v1 cannot be disassembled easily and does not have any known recovery mode in U-Boot.

Djfe, 2024-09-02:
I suspect this to be caused by a kernel bump that happened between Dec 3rd (22.03.6) and March 22nd (23.05.3).
But it's not like I'm an expert or anything.

openwrt-22.03:
612158176528584500c352105c76581744958b2e
debf4b56cdcc91f9e41bba4a13c4b8370e0a7f1d
c4df9473ef00116fa168a3320567acf01f876d8a
f2e8d5974c062fa3b125e70e0bd247f3be23ca96
a352312f0dd3dec317ad1ef294523c91db9f232a
61c6bc2eaafaf58164b6f3635eeefc7a886b7d67
ce37d2c690f762898de968fae6459a92dec01f37

openwrt-23.05:
4c1d13e995c3f19054caba8b51b9114f664e3f84
db7b247fa932e845cd7270b6e474a15d54323a41

I just opened a pull request to disable building these two devices.

namiltd, 2024-09-02:
I suggest generating intermediate version images to find out exactly which patch is at fault.

Djfe, 2024-09-02:
you mean with git bisect, right?
How do you suggest to reset the device between every test?
(by connecting a device directly to the nor flash to reset it to a working state?)

namiltd, 2024-09-02:
I would use the serial console

namiltd, 2024-09-02:
I would use the serial console.

noobtheproswag, 2024-09-02:
The RE210 does have tftp boot support in u-boot. That is what I used to recover after I sysupgraded to the broken releases. It requires serial console to manually select it out of the boot options at boot time. I have a working setup and I can do testing but I don't know how to git bisect.

namiltd, 2024-09-02:
git pull
git reset --hard commit_id

Djfe, 2024-09-02:
for singular commits the above works. once you know which commits are affected and which aren't, please tell us the commit ID. Then we could do git bisect to identify whether it actually is a kernel bump or not that introduced the issue.

Djfe, 2024-09-02:
I was under the impression of @andyboeh 's message on the mailing list:

> This is very unfortunate because there is no easy method to disassemble the RE200v1 (it is welded 
> together) and there is no know recovery method in U-Boot. When users install any of these OpenWrt versions, their device is basically trash.

Great that there is a way to recover. Not so great if disassembly isn't feasible for most users

noobtheproswag, 2024-09-02:
I will try to test the commits today and report back, but I have a dual core computer so building might take a while...

andyboeh, 2024-09-02:
@Djfe Yes, there is TFTP in U-Boot on the RE200 as well. However, it is not possible (at least, we don't know how) to trigger it using a button or a magic packet as with a lot of other devices, so I didn't consider this "recovery" :)

noobtheproswag, 2024-09-05:
Sorry for the late reply, I've had a busy couple days and the compilations taking long didnt help much.

I have compiled and boot tested all of the commits @Djfe listed above in his comment. All of them boot just fine.

Any other ideas? Should I try compiling and testing the actual releases?

noobtheproswag, 2024-09-05:
For context, I built on an up to date Debian 12 Bookworm machine with the dependencies installed according to the [OpenWrt build system setup](https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem#debianubuntumint) page. I then ran the following commands in order:

```
cd

rm -frd ./openwrt

git clone https://git.openwrt.org/openwrt/openwrt.git

cd openwrt

git pull

git checkout [branch name here]

git pull

git reset --hard [commit id here]

./scripts/feeds update -a
./scripts/feeds install -a

make menuconfig

make defconfig

make -j4 download

make -j4
```

In the menuconfig, I only changed the target to "Mediatek Ralink MIPS" and then selected RE210 in the target profile list. I didn't change anything else.

noobtheproswag, 2024-09-06:
> Note that I have successfully tftp booted the corresponding kernel images for all three of the versions that don't boot when sysupgraded to.

If the kernel images for the broken builds can be tftp booted, maybe the kernel isn't the culprit?

Djfe, 2024-09-06:
I have missed that important part. Maybe it's due to the way the sysupgrade/kernel are aligned on the flash. (though unlikely since several releases since date x are affected)
It could still be due to certain commit though.

Have you tested the commits by sysupgrading or by booting the initramfs via tftp?

noobtheproswag, 2024-09-07:
I have more confusing news... I compiled the v23.05.4 tag, sysupgraded to it, and it boots fine...

Here is the output of ` ./scripts/diffconfig.sh`:
```
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7620=y
CONFIG_TARGET_ramips_mt7620_DEVICE_tplink_re210-v1=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-wireguard=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
```

I built this with the following set of commands:

```
cd

rm -frd ./openwrt

git clone https://git.openwrt.org/openwrt/openwrt.git

cd openwrt

git pull

git checkout v23.05.4

./scripts/feeds update -a
./scripts/feeds install -a

make menuconfig

make defconfig

make -j4 download

make -j4
```

Please tell me if I can provide any other information. Could this possibly be a problem with the build system? (That's just my guess though...)

noobtheproswag, 2024-09-07:
Also if possible can someone with the permissions please add the appropriate labels to this issue? Thanks.

noobtheproswag, 2024-09-07:
To confirm, I just retested the official v23.05.4 build by sysupgrading to it and it still doesn't boot.

andyboeh, 2024-09-07:
> I have more confusing news... I compiled the v23.05.4 tag, sysupgraded to it, and it boots fine...

Yes, another user reported the same for the RE200 - please see the Forum thrad: https://forum.openwrt.org/t/tp-link-re200-v1-seemingly-bricked-after-sysupgrade/199277/22

noobtheproswag, 2024-09-07:
Is there a way to take the broken images down before more unaware people inevitably brick their devices with no other option of recovery other than tearing down the device? Who do we neeed to contact?

Djfe, 2024-09-07:
I already asked @hauke via IRC and he promised to take the images down. I sent him another reminder yesterday. Don't flood him with messages now 😅

Also your info is very helpful indeed. This proves (to me) that the issue has got to do with the tooling/build system in use.
I bet if we build older releases with the new build bots they would also fail to boot and if we were to build the new releases with the older build bots then they would boot.
affected devices: tp-link re200 v1 and re210 v1 (the ones we know of)

maybe @aparcar or @ynezz can help us.
Hopefully we can figure out why this happens and prevent it from happening again.

https://github.com/openwrt/buildbot

Djfe, 2024-09-07:
Please share your build of v23.05.4 so we can compare it in a hexeditor to the official build 👍

noobtheproswag, 2024-09-07:
Here is my working build:
[re210v1_23.05.4_working_build.zip](https://github.com/user-attachments/files/16919273/re210v1_23.05.4_working_build.zip)
(This is a booting build with LuCI)
Let me know if I can provide anything else.

Thank you for your amazing work 😃

robimarko, 2024-09-08:
Hm, the only thing that comes to mind is `mktplinkfw` builds being somehow weird when built on buildbots

DragonBluep, 2024-09-09:
I guess this is a lzma decompression issue caused by memory overlap. I noticed that @zamaudio has changed the default lzma dictionary size in https://github.com/openwrt/openwrt/pull/15610. I think this method can also fix your devices. But a more proper way should be using `OpenWrt lzma loader`. e.g.
https://github.com/openwrt/openwrt/blob/15280316faa865a933913748c2153d6fb0f1b48f/target/linux/ath79/image/common-tp-link.mk#L86-L95

namiltd, 2024-09-09:
@DragonBluep I suspected the same on https://github.com/openwrt/openwrt/issues/16296#issuecomment-2322943839. I suggested removing LuCI to save space. Unfortunately it didn't help

DragonBluep, 2024-09-09:
> @DragonBluep I suspected the same on [#16296 (comment)](https://github.com/openwrt/openwrt/issues/16296#issuecomment-2322943839). I suggested removing LuCI to save space. Unfortunately it didn't help

LuCI is not a part of the kernel, removing it won't reduce kernel size. The bootloader doesn't decompress rootfs at all.

Djfe, 2024-09-09:
@DragonBluep Interesting, but how does that explain that the same commit as the release works when built locally. Only the build bot version fails.

DragonBluep, 2024-09-09:
> @DragonBluep Interesting, but how does that explain that the same commit as the release works when built locally. Only the build bot version fails.

Usually the buildbot generate larger kernel size than our custom build because we won't select all packages and all supported devices in our local build.

henkv1, 2024-09-10:
@DragonBluep FYI, I have the same issue with my RE200v1. The v23.05.4 kernel I built is 5974768 bytes, the v23.05.4 kernel downloaded from the Firmware selector is 5673820 bytes. So I don't think the kernel size is the issue here.

DragonBluep, 2024-09-10:
> @DragonBluep FYI, I have the same issue with my RE200v1. The v23.05.4 kernel I built is '5974768 bytes, the v23.05.4 kernel downloaded from the Firmware selector is 5673820 bytes. So I don't think the kernel size is the issue here.

The kernel.size won't large than 3 MiB on ramips target...

Djfe, 2024-09-11:
what dragonbluep is saying is: kernel size isn't sysupgrade size.
You could run binwalk on the file as a way to determine kernel size

henkv1, 2024-09-11:
I compared the sizes of the kernel images. Here's the binwalk output:
```
My package:
binwalk openwrt-ramips-mt7620-tplink_re200-v1-initramfs-kernel.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
512           0x200           LZMA compressed data, properties: 0x6D, dictionary size: 8388608 bytes, uncompressed size: 11302101 bytes
1123251       0x1123B3        JBOOT STAG header, image id: 12, timestamp 0x79ACFF50, image size: 2017008662 bytes, image JBOOT checksum: 0x6B88, header JBOOT checksum: 0x591E

Buildbot package:
binwalk openwrt-23.05.4-ramips-mt7620-tplink_re200-v1-initramfs-kernel.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
512           0x200           LZMA compressed data, properties: 0x6D, dictionary size: 8388608 bytes, uncompressed size: 10975061 bytes

My sysupgrade package:
 binwalk openwrt-23.05.4-ramips-mt7620-tplink_re200-v1-squashfs-sysupgrade.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
512           0x200           LZMA compressed data, properties: 0x6D, dictionary size: 8388608 bytes, uncompressed size: 7649749 bytes
2302384       0x2321B0        Squashfs filesystem, little endian, version 4.0, compression:xz, size: 3619524 bytes, 1332 inodes, blocksize: 262144 bytes, created: 2024-07-15 22:14:18

Buildbot sysupgrade package:
binwalk openwrt-ramips-mt7620-tplink_re200-v1-squashfs-sysupgrade.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
512           0x200           LZMA compressed data, properties: 0x6D, dictionary size: 8388608 bytes, uncompressed size: 7668565 bytes
2291004       0x22F53C        Squashfs filesystem, little endian, version 4.0, compression:xz, size: 3849848 bytes, 1338 inodes, blocksize: 262144 bytes, created: 2024-07-15 22:14:18
```

zamaudio, 2024-09-12:
I tried using lzma-loader on my MR200 port but it did not seem to work.  I reduced the dictionary size to 4MB with `lzma -d22` and that fixed my port.  Thanks to dev suggestion from openwrt forum.

DragonBluep, 2024-09-14:
Kernel size:
[image]

closed 2024-10-06 by openwrt-bot

ukleinek, 2025-01-28:
Is this issue the reason that https://openwrt.org/toh/tp-link/re200 has the warning "The latest OpenWrt release (23.05.3, 23.05.4 and 22.03.7) are broken on the RE200 v1! Do not flash these images, you will brick your device!"? If so, it would be great to update it, as the warning in combination with "Supported Current Rel: 23.05.5" is irritating at best. (Cc: @andyboeh who added the note to the wiki.)

andyboeh, 2025-01-28:
If I remember correctly, I added this note due to several reports of non-booting devices in the forum. If it works now, please remove the warning.

ukleinek, 2025-01-28:
I own an RE200 v1, and if it wasn't so hard to open I'd test that. But in that situation I'd like to know beforehand if I brick my device ...

andyboeh, 2025-01-28:
OK, I thought you had already tested it. I have a disassembled unit lying around, I'll test it.

andyboeh, 2025-01-28:
> I own an RE200 v1, and if it wasn't so hard to open I'd test that. But in that situation I'd like to know beforehand if I brick my device ...

There is no need to test: The fix was backported to 23.05 on October 8th, 2024 while the lateste release in the 23.05 series was on October 1st. Hence the warning still applies. 

Snapshot or 24.10-RC images should work fine, although I did not test this.

ukleinek, 2025-01-28:
Testing of 24.10-rc7 would be very welcome as I want that one because of a dnsmasq fix included there ...

Mr-Tezar, 2026-09-04:
Hello everyone,

I have a **TP-Link RE210 V1 (EU)** that is currently completely bricked.

Unfortunately, the device's flash memory was completely erased, including the **bootloader/U-Boot**, so it no longer boots, does not start OpenWrt or the stock firmware, and I cannot access the device through the normal recovery/TFTP method.

I have been investigating the OpenWrt support for the RE210 V1, especially the work related to **OKLI and the U-Boot used by the original TP-Link firmware**.

At this point, what would help me the most is a **complete raw dump of the flash chip from a working RE210 V1 EU device**.

If anyone owns a working **TP-Link RE210 V1 EU**, I would greatly appreciate it if you could make a full backup/dump of the device's flash memory and share it with me.

Ideally, I need a **raw 1:1 dump of the entire flash chip**, including:

* U-Boot / bootloader
* firmware partitions
* factory/calibration data
* configuration/NVRAM
* any other partitions present on the original device

Please **do not modify or flash anything before making the dump**. A dump taken directly from a working stock device would be extremely useful.

I am specifically looking for the **complete raw flash dump**, not just the official TP-Link firmware update file, because I need to reconstruct the original flash layout and restore the bootloader on my completely erased device.

The device is:

**Model:** TP-Link RE210
**Hardware Version:** V1.0
**Region:** EU
**Current state:** Completely bricked / flash erased / bootloader missing

I understand that the dump may contain device-specific calibration or configuration data. I only need it for firmware recovery and reverse-engineering purposes, and I can preserve or replace those device-specific sections appropriately rather than using them as-is.

If you can provide such a dump, it would be a huge help in recovering this device and potentially documenting a proper recovery procedure for other completely bricked RE210 V1 units.

Thank you!

ukleinek, 2026-09-05:
I have a re200-v1 only, does a dump from that help, too?

Mr-Tezar, 2026-09-05:
> I have a re200-v1 only, does a dump from that help, too?

Yes, absolutely! Thank you very much for offering to help.

I would really appreciate it if you could make a complete raw dump of the flash chip from your working RE200 V1.

Ideally, I need a full 1:1 dump of the entire flash, including the bootloader and all partitions. The resulting file should be approximately 8 MB if it is the same flash size.

Please take the dump directly from the working device before modifying or flashing anything.

Even though it is an RE200 V1 rather than an RE210 V1, it could still be very useful for comparing the bootloader, flash layout, and recovery process.

Thank you again, I really appreciate your help!

ukleinek, 2026-09-05:
https://www.kleine-koenig.org/tmp/tplinkre210v1.tar.xz contains the dumps of `/dev/mtdblock{0,1,5,6}` and a description of the flashmap that also explains why the others are not interesting. I zeroed the area of `mtdblock4` which contains the overlay as I have some personal stuff in there. I hope all the other bits are generic and don't leak any private data.

Mr-Tezar, 2026-09-08:
> https://www.kleine-koenig.org/tmp/tplinkre210v1.tar.xz contains the dumps of `/dev/mtdblock{0,1,5,6}` and a description of the flashmap that also explains why the others are not interesting. I [...]

Thank you for providing the dumps and the flashmap description.

I tried combining the provided partitions and reconstructing a complete flash dump, but unfortunately I was not able to bring the device back to life.

At this point, I believe I need a **complete dump of the flash chip**, including all partitions and areas, made directly using a **CH341A programmer**. A full raw dump from the physical flash chip would allow me to make sure that no bootloader, configuration, or other required data is missing.

If you are able to make a complete CH341A dump of the flash chip, that would be extremely helpful.

ukleinek, 2026-09-10:
> At this point, I believe I need a **complete dump of the flash chip**, including all partitions and areas, made directly using a **CH341A programmer**. A full raw dump from the physical flash chip [...]

Sorry, I don't have the case open and don't have the tools to change that in a non-destructive way. I assume this is a NAND chip, I can try to do a complete dump (i.e. with OOB) when I'm able to login next time.

zamaudio, 2026-09-10:
In retrospect, everything is obvious.  But in future please always make full backups of flash when tinkering with firmware on any device.