openwrt/openwrt #16299: package: redundant downloading package tarballs

ikazuhiro, 2024-08-31:
### Describe the bug

Sine https://github.com/openwrt/openwrt/commit/bab3ae2ee7656600a185f4cef11cef94389023af, OpenWrt prefers .gz tarball than .xz for tools.  But packages in `package` directory and [package repository](https://github.com/openwrt/packages) still seem to prefer .xz.  As a result, `dl` directory contains many tarballs which have the same content.  For example, I can see following redundant tarballs in my `dl` directory.
```
autoconf-2.71.tar.gz
autoconf-2.71.tar.xz
automake-1.16.5.tar.gz
automake-1.16.5.tar.xz
e2fsprogs-1.47.0.tar.gz
e2fsprogs-1.47.0.tar.xz
expat-2.6.2.tar.gz
expat-2.6.2.tar.xz
libtool-2.4.7.tar.gz
libtool-2.4.7.tar.xz
patch-2.7.6.tar.gz
patch-2.7.6.tar.xz
zlib-1.3.1.tar.gz
zlib-1.3.1.tar.xz
```

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

all

### Device

host

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-09-01:
This is not a bug, avoiding xz is intentional after the recent xz utils drama, see bab3ae2ee7656 and 706f0e395 as examples.

closed 2024-09-01 by ynezz

ikazuhiro, 2024-09-01:
Thank you for your replay.
Yes, I know xz related issues, and my question is whether packages remain prefering .xz tarballs.  I think there is no advantage to download .xz tarballs which we have  already downloaded as other format.

ynezz, 2024-09-01:
As you can see in the linked https://github.com/openwrt/openwrt/commit/bab3ae2ee7656600a185f4cef11cef94389023af the .gz or .bz2 archives are now preferred, thus if something still downloads `autoconf-2.71.tar.xz` then its a bug and if you want that fixed, you need to provide more details.

ikazuhiro, 2024-09-01:
As I wrote, many packages in `package` directory and [packages repository](https://github.com/openwrt/packages) seems to use .xz tarballs.

```bash
$ find package/ -type f -exec grep -B 5 -E '^PKG_SOURCE:=.*\.xz$' {} + | grep -E 'PKG_(NAME|SOURCE):='

(snip)
package/utils/e2fsprogs/Makefile-PKG_NAME:=e2fsprogs
package/utils/e2fsprogs/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
package/utils/util-linux/Makefile-PKG_NAME:=util-linux
package/utils/util-linux/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
package/libs/libtool/Makefile-PKG_NAME:=libtool
package/libs/libtool/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
package/libs/mpfr/Makefile-PKG_NAME:=mpfr
package/libs/mpfr/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
package/libs/zlib/Makefile-PKG_NAME:=zlib
package/libs/zlib/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
package/libs/gmp/Makefile-PKG_NAME:=gmp
package/libs/gmp/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION)$(PKG_REVISION).tar.xz
```

In packages repository,

```bash
$ find . -type f -exec grep -B 5 -E '^PKG_SOURCE:=.*\.xz$' {} + | grep -E 'PKG_(NAME|SOURCE):='

(snip)
./libs/expat/Makefile-PKG_NAME:=expat
./libs/expat/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./devel/m4/Makefile-PKG_NAME:=m4
./devel/m4/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./devel/pkgconf/Makefile-PKG_NAME:=pkgconf
./devel/pkgconf/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./devel/automake/Makefile-PKG_NAME:=automake
./devel/automake/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./devel/patch/Makefile-PKG_NAME:=patch
./devel/patch/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./devel/bison/Makefile-PKG_NAME:=bison
./devel/bison/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./devel/libtool-bin/Makefile-PKG_NAME:=libtool
./devel/libtool-bin/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./devel/autoconf/Makefile-PKG_NAME:=autoconf
./devel/autoconf/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./utils/sed/Makefile-PKG_NAME:=sed
./utils/sed/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./utils/tar/Makefile-PKG_NAME:=tar
./utils/tar/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
./utils/coreutils/Makefile-PKG_NAME:=coreutils
./utils/coreutils/Makefile:PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
```

You can see .xz tarballs are downloaded when you build above pacakges.  If I changed `PKG_SOURCE` and `PKG_HASH` values to the same  with `tools/.../Makefile`, tarballs for tools  were shared.