openwrt/openwrt #16302: ath10k-ct fails to build due to unselected mac80211 debugfs option

Olipro, 2024-09-01:
### Describe the bug

When building with the Candelatech `ath10k-ct` packages, because warnings are treated as errors, the build will fail if `mac80211` does not have the `debugfs` option enabled since the package contains a `#warning` line that detects the option isn't selected.

This can of course be trivially worked around by enabling the debugfs option for `mac80211` in the `menuconfig`

The warning states that you are encouraged to enable this option, however, doing so is not mandatory.

Therefore, in terms of fixing the problem, either warnings should not be treated as errors, or the package should require the mac80211 debugfs option to be selected.

### OpenWrt version

c6f04f718ec89561f2261f2388857baa126eaa50

### OpenWrt release

c6f04f718ec89561f2261f2388857baa126eaa50

### OpenWrt target/subtarget

ipq806x/R7800

### Device

N/A, this is bug is at build time.

### Image kind

Self-built image

### Steps to reproduce

Choose one of the `ath10k-ct` wifi firmwares without enabling the debugfs option for `mac80211` and attempt to build

### Actual behaviour

build fails due to a `#warning` in the `ath10k-ct`'s `wmi.h` that detects the debugfs option not being turned on due to warnings treated as errors.

### Expected behaviour

build doesn't fail.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-09-09:
Can you provide the required diffconfig so I can reproduce this?

robimarko, 2024-09-09:
Ok, simply disabling mac80211 debugfs support is enough.

PR #16619 (2024-10-06): ath10k-ct: select mac80211 debugfs support
Trying to compile ath10k-ct without mac80211 debugfs support will result in:
```
openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2024.07.30~ac71b14d/ath10k-6.10/wmi.h:8083:2: error: #warning Please enable ATH10K-DEBUGFS kernel option for optimal support for CT firmware. [-Werror=cpp]
 8083 | #warning Please enable ATH10K-DEBUGFS kernel option for optimal support for CT firmware.
      |  ^~~~~~~
cc1: all warnings being treated as errors
```

Fixes: #16302 
So, since the driver itself is sa [... cut]
commit cc29e90d (2024-10-06): ath10k-ct: select mac80211 debugfs support

Trying to compile ath10k-ct without mac80211 debugfs support will result in:
openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2024.07.30~ac71b14d/ath10k-6.10/wmi.h:8083:2: error: #warning Please enable ATH10K-DEBUGFS kernel option for optimal support for CT firmware. [-Werror=cpp]
 8083 | #warning Please enable ATH10K-DEBUGFS kernel option for optimal support for CT firmware.
      |  ^~~~~~~
cc1: all warnings being treated as errors

So, since the driver itself is saying that debugfs is r [... cut]
closed 2024-10-06 by openwrt-bot