openwrt/openwrt #16303: OpenSSH starts with a sshd_config that does not allow root or empty password

grosjo, 2024-09-01:
### Describe the bug

WHen repalcing dropbear by openssh, the default config file does not allow to login, as regusing both root and empty password

### OpenWrt version

r27086-aeb9028aab

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

All devices

### Image kind

Self-built image

### Steps to reproduce

Just remove dropbear and select openssh-server

### Actual behaviour

Refuse login at boot

### Expected behaviour

Allow login at boot

### Additional info

Ideal config file would look like:

```
Port 22
ListenAddress 0.0.0.0
HostKey /etc/ssh/ssh_host_rsa_key
HostKey /etc/ssh/ssh_host_ed25519_key
SyslogFacility AUTH
LogLevel INFO
PermitRootLogin yes
PubkeyAuthentication yes
AuthorizedKeysFile	.ssh/authorized_keys
AllowAgentForwarding yes
AllowTcpForwarding yes
PrintLastLog yes
TCPKeepAlive yes
Compression delayed
UseDNS no
PermitTunnel yes
PermitEmptyPasswords yes
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

usapmz, 2024-10-13:
Experiencing the same issue.

grosjo, 2025-01-20:
Bow to fix that ? Where is stored the default config of openssh ?