openwrt/openwrt #16305: OpenWrt 23.05.4 - odhcpd DHCPv6 - sets 1 hour lease time instead of 1 day + 12 hours

a-spicer, 2024-09-01:
### Describe the bug

I just upgraded to OpenWrt 23.05.4 and noticed on the UI - https://192.168.2.1/cgi-bin/luci/admin/status/overview that Active DHCPv6 Leases shows less than 1 hour. It seems to be set for 1 hour. The leases are being renewed but not at 36 hours as IPv4 are being done. I don't see any way to configure this. /etc/config/dhcp already has "option leasetime '36h'"

### OpenWrt version

r24012-d8dd03c46f

### OpenWrt release

23.05.4

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Occured right after upgrade to 23.05.4. Also rebooted to be sure. Check Active DHCPv6 Leases in the UI/Status

### Actual behaviour

IPV6 lease time seems to be set at 1 hr. 3600 seconds. No way to change that?

### Expected behaviour

IPV6 lease time should be 36 hours - like it is for IPv4

1 day + 12 hours.
129600 seconds.

### Additional info

There was issues 9 months ago with odhcpd IPv6 as reported here:
https://github.com/openwrt/openwrt/issues/13767

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2024-09-02:
What is the actual valid lifetime of the DHCPv6 prefix delegated by your ISP? Is it only 1 hour?

a-spicer, 2024-09-02:
This is for the LAN - br-1 not the WAN.

It's set for 36 hours as it has been. Before the update the leases reflected that. Now they reflect 1 hour. IPv4 leases still reflect the 36 hours. odhcpd for IpV6 is apparently being set from somewhere other than /etc/config/dhcp.

9 months ago had another kind of odhcpd related problem (mentioned in my last comment) - which seemed to got resolved.

dave14305, 2024-09-02:
Are you using the option below in LAN interface DHCP IPv6?
[image]

dave14305, 2024-09-02:
But the prefix delegation lifetime is still important. You can’t lease an address longer than you have the “right” to own that same prefix. odhcpd looks like it will cap the lease at the prefix lifetime.

https://github.com/openwrt/odhcpd/blob/a29882318a4ccb3ae26f7cc0145e06ad4ead224b/src/dhcpv6-ia.c#L1139

a-spicer, 2024-09-02:
That's just the current ... and if I re-run it obviously goes down. It's not the prefix delegation lifetime setting.

root@OpenWrt:~# ifstatus lan
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 3410,
        "l3_device": "br-lan",
        "proto": "static",
        "device": "br-lan",
        "updated": [
                "addresses"
        ],
        "metric": 0,
        "dns_metric": 0,
        "delegation": false,
        "ipv4-address": [
                {
                        "address": "192.168.2.1",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [
                {
                        "address": "2600:1700:XXXX:a64e::",
                        "mask": 64,
                        "preferred": 2019,
                        "valid": 2019,
                        "local-address": {
                                "address": "2600:1700:XXXX:a64e::1",
                                "mask": 64
                        }
                },
                {
                        "address": "2600:1700:XXXX:a64f::",
                        "mask": 64,
                        "preferred": 2019,
                        "valid": 2019,
                        "local-address": {
                                "address": "2600:1700:XXXX:a64f::1",
                                "mask": 64
                        }
                }
        ],
        "route": [

        ],
        "dns-server": [

        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {

        }
}

dave14305, 2024-09-02:
Has the lifetime reset now on the lan? What version did you upgrade from?

a-spicer, 2024-09-02:
No. 23.05.0 to 23.05.4

Currently set at 12 hours and turned that checkbox back off - which basically does: 

**ra_useleasetime** | boolean |   | 0 | If set, the configured DHCPv4 leasetime is used both as limit for the preferred and valid lifetime of an IPv6 prefix.

Which doesn't do what it would be expected to do. I had manually set it to 36 hours in the past. Works for ipv4, WAS working for ipv6, but not since the upgrade the other day.

dave14305, 2024-09-02:
Please post the updated lifetimes of the PD and leases:
```
ifstatus lan | grep -E "uptime|preferred|valid"
ubus call dhcp ipv6leases | grep -E "preferred|valid"
```

a-spicer, 2024-09-02:
BTW I just rebooted this router so the uptime will be low.

root@OpenWrt:~# ifstatus lan | grep -E "uptime|preferred|valid"
        "uptime": 972,
                        "preferred": 2577,
                        "valid": 2577,
                        "preferred": 2547,
                        "valid": 2547,
root@OpenWrt:~# ubus call dhcp ipv6leases | grep -E "preferred|valid"
                                                        "preferred-lifetime": 2498,
                                                        "valid-lifetime": 2498
                                                        "preferred-lifetime": 2497,
                                                        "valid-lifetime": 2497
                                        "valid": 2498
                                                        "preferred-lifetime": 2498,
                                                        "valid-lifetime": 2498
                                                        "preferred-lifetime": 2497,
                                                        "valid-lifetime": 2497
                                        "valid": 2498
root@OpenWrt:~#

dave14305, 2024-09-02:
Thanks. It seems to confirm the notion that the lease duration can be no longer than the prefix lifetime.

Maybe the ISP has shortened the duration recently. Keep an eye on it. Not much changed in odhcpd between 23.05.0 and 23.05.4, except the lease reporting issue in LuCI mentioned in the previous issue.

a-spicer, 2024-09-03:
Thanks. I'm gonna keep this open for a little longer. Because the last time other things other than odhcpd were found. The odds of my ISP suddenly changing to 1 hour prefix lease time seems low. But then again this is the company that can't turn off spanning tree on LAN in the router they provide ... and now took all but 2 of the IPv6 prefixes out of /60 they are supposed to be delegating.

a-spicer, 2024-09-04:
Just completely dropped my iPhone's IPv6 lease ... phone still had IPv6 address. Had to disconnect and reconnect for it to show up again. Both on the UI and in UBUS output.

a-spicer, 2024-09-07:
Packet capture on WAN from ISP modem-router Showing Preferred Lease Time and Valid Lease Time sent by ISP router.

11:31:33.482092 IP6 (hlim 64, next-header UDP (17) payload length: 176) fe80::a2e7:aeff:fe32:48a0.547 > fe80::826
9:1aff:fe1e:84be.546: [udp sum ok] dhcp6 reply (xid=9ab9b0 (IA_NA IAID:1 T1:1800 T2:2880 (IA_ADDR OpenWrt.attloca
l.net pltime:3600 vltime:3600)) (IA_PD IAID:1 T1:1800 T2:2880 **(IA_PD-prefix 2600:1700:XXXX:a64f::/64 pltime:3600
vltime:3600))** (client-ID hwaddr type 1 80691a1e84be) (server-ID hwaddr/time type 1 time 3348282552 a0e7ae3248a0)
(preference 255) (DNS-server dsldevice6.attlocal.net) (DNS-search-list attlocal.net.))
12:01:33.501431 IP6 (hlim 64, next-header UDP (17) payload length: 176) fe80::a2e7:aeff:fe32:48a0.547 > fe80::826
9:1aff:fe1e:84be.546: [udp sum ok] dhcp6 reply (xid=e73f33 (IA_NA IAID:1 T1:1800 T2:2880 (IA_ADDR OpenWrt.attloca
l.net pltime:3600 vltime:3600)) (IA_PD IAID:1 T1:1800 T2:2880 (IA_PD-prefix 2600:1700:XXXX:a64f::/64 pltime:3600
vltime:3600)) (client-ID hwaddr type 1 80691a1e84be) (server-ID hwaddr/time type 1 time 3348282552 a0e7ae3248a0)
(preference 255) (DNS-server dsldevice6.attlocal.net) (DNS-search-list attlocal.net.))
12:31:37.961281 IP6 (hlim 64, next-header UDP (17) payload length: 176) fe80::a2e7:aeff:fe32:48a0.547 > fe80::826
9:1aff:fe1e:84be.546: [udp sum ok] dhcp6 reply (xid=2a4ad8 (IA_NA IAID:1 T1:1800 T2:2880 (IA_ADDR OpenWrt.attloca
l.net pltime:3600 vltime:3600)) (IA_PD IAID:1 T1:1800 T2:2880 (IA_PD-prefix 2600:1700:XXXX:a64f::/64 pltime:3600
vltime:3600)) (client-ID hwaddr type 1 80691a1e84be) (server-ID hwaddr/time type 1 time 3348282552 a0e7ae3248a0)
(preference 255) (DNS-server dsldevice6.attlocal.net) (DNS-search-list attlocal.net.))
13:01:40.201996 IP6 (hlim 64, next-header UDP (17) payload length: 176) fe80::a2e7:aeff:fe32:48a0.547 > fe80::826
9:1aff:fe1e:84be.546: [udp sum ok] dhcp6 reply (xid=12dc91 (IA_NA IAID:1 T1:1800 T2:2880 (IA_ADDR OpenWrt.attloca
l.net pltime:3600 vltime:3600)) (IA_PD IAID:1 T1:1800 T2:2880 (IA_PD-prefix 2600:1700:XXXX:a64f::/64 pltime:3600
vltime:3600)) (client-ID hwaddr type 1 80691a1e84be) (server-ID hwaddr/time type 1 time 3348282552 a0e7ae3248a0)
(preference 255) (DNS-server dsldevice6.attlocal.net) (DNS-search-list attlocal.net.))

dave14305, 2024-09-07:
At least you know now. Sometimes I've seen ISPs shorten lease times if they are preparing for or performing some kind of maintenance or big change. But there are also many stories of ISPs being very stingy with lease times, where the renewals happen every 30 minutes (half of 60 mins).

a-spicer, 2024-09-07:
It's actually being renewed in 30 minutes. I think the DHCPv6 process cuts the total in half and requests renewal. I don't think they are making any big change. I've had the same prefixes forever. But then again they have been in our back yard getting ready to pull FTTH fiber for this street. Maybe I didn't notice this lease time change before I upgraded. But I could have sworn it was 1 day plus all along. Getting AT&T to do anything is like pulling teeth. Probably be impossible to get them to bump this up to 12 hours or more. I would be nice if I could compile a modified odhcpd without that check and manage to install it.