openwrt/openwrt #16314: Enable VLAN filtering does not add WLAN interfaces to bridge, reboot is required

cyfdecyf, 2024-09-02:
## Steps to reproduce:

1. go to: Network → Interfaces → Devices -> `br-lan` -> Configure, enable VLAN filtering like this:

[image]

2. go to: Network → Interfaces → `lan` -> General Settings, **set device to `br-lan.1`**
3. Make sure wireless SSID network is `lan` (this is the default)
4. Save & Apply
5. Can't connect to `br-lan.1` ip address for wireless client, but wired connection is fine.

## Actual behavior:

1. Run `bridge vlan`, the output is like following (ommitting some interfaces):

   ```
   port              vlan-id
   lan2              1 PVID Egress Untagged
                     2
   lan3              1 PVID Egress Untagged
                     2
   br-lan            1
                     2
   ```
   Note **wlan interfaces are not in the output**.
2. The wireless client problem cam be manually by executing following commands:
   ```
   bridge vlan add dev phy1-ap0 vid 1 pvid untagged
   bridge vlan add dev phy0-ap0 vid 1 pvid untagged
   ```
3. Reboot can also fix the problem. After reboot, the bridge is setup correctly as `bridge vlan` shows output like:
   ```
   port              vlan-id
   lan2              1 PVID Egress Untagged
                     2
   lan3              1 PVID Egress Untagged
                     2
   br-lan            1
                     2
   phy1-ap0          1 PVID Egress Untagged
   phy0-ap0          1 PVID Egress Untagged
   ```

So **the problem only exists when applying modified config for enabling VLAN filtering**.
**The generated config is correct** because a reboot results the expected behavior.

## Expected behavior:

Applying enabling VLAN filtering config should set wlan interfaces in bridge with correct VLAN tag, without needing to reboot.

## Additional Information:

OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='ImmortalWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r27917-81a1f98d5b'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='ImmortalWrt 23.05.3 r27917-81a1f98d5b'
DISTRIB_TAINTS=''
```

The problem also exists on version `23.05.2`.

I'm new to OpenWRT. Though I'm using ImmortalWrt, I guess this problem maybe related to upstream LuCI so submiting issue here.

jow-, 2024-09-02:
Not a LuCI issue, needs to be fixed in OpenWrt base

nbd168, 2024-10-05:
Can you please send me /etc/config/network before and after the change?

cyfdecyf, 2024-10-05:
@nbd168 Thanks for looking into this issue.

I maybe not able to give you the precise "before" and "after" config. The before config is from a backup that's produced on April (I should done no config after that till enabling VLAN filtering). The after config is what I'm now using.

Before configuring VLAN filtering:

```
config interface 'loopback'
    option device 'lo'
    option proto 'static'
    option ipaddr '127.0.0.1'
    option netmask '255.0.0.0'

config globals 'globals'
    option ula_prefix 'fd01:62fd:d9b6::/48'

config device
    option name 'br-lan'
    option type 'bridge'
    list ports 'lan2'
    list ports 'lan3'
    list ports 'lan4'
    list ports 'wan'

config interface 'lan'
    option device 'br-lan'
    option proto 'static'
    option ipaddr '192.168.1.251'
    option netmask '255.255.255.0'
    option ip6assign '60'
    option gateway '192.168.1.254'
    list dns '192.168.1.254'
    list dns_search 'home'
```

I may enabled STP and IGMP snooping before enabling VLAN filtering, but turned off STP recently.

So here's current configuration with VLAN filtering:

```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd01:62fd:d9b6::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'
        option priority '16384'
        option igmp_snooping '1'

config interface 'lan'
        option device 'br-lan.1'
        option proto 'static'
        option ipaddr '192.168.1.251'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option gateway '192.168.1.254'
        list dns '192.168.1.254'
        list dns_search 'home'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:u*'
        list ports 'wan:u*'

config bridge-vlan
        option device 'br-lan'
        option vlan '2'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'lan4:t'
        list ports 'wan:t'
```

nbd168, 2024-10-06:
Thanks. I was able to reproduce the issue with your config files and managed to come up with a fix. Please try this patch: https://nbd.name/p/b77ad0c7

cyfdecyf, 2024-10-09:
@nbd168 Thanks for the fix. I'm new to OpenWRT and had never built it from source, it would take me some time to try.

nbd168, 2024-10-09:
@cyfdecyf  I've made a netifd package that you can install on 23.05.x: https://nbd.name/netifd_2024-01-04-c18cc79d-2_aarch64_cortex-a53.ipk

cyfdecyf, 2024-10-11:
@nbd168 Thanks for your package. I've tested your package and confirm the problem is fixed. Closing this issue.

closed 2024-10-11 by cyfdecyf

Neustradamus, 2024-10-11:
Attention @cyfdecyf, the patch is not applied yet, please wait @nbd168 commit before closing.
Can you reopen it?

nbd168, 2024-10-11:
It is applied already

Neustradamus, 2024-10-11:
@nbd168: Nice, good job, thanks!

Here:
- https://github.com/openwrt/openwrt/commit/7e942c53b2eb3a3f55503fd2c6aabb00a35e5ebf