openwrt/openwrt #16315: Linksys E8450 - Commit 1069514 is Causing High Load Average

CharlesMengCA, 2024-09-02:
### Describe the bug

On my Linksys E8450, recent snapshot builds would have High Load Average after rebooting the router, the LuCI got unresponsive for a while. The load average had values like 3.45,1.05,0.37.

Reverted the following commit and the load average was back to normal

https://github.com/openwrt/openwrt/commit/1069514978c419cc5bdfc18c5670b46d0443a686

I also reverted fca306c563196481c1ad488b9abf4e84026591f4 since it is a fix for the above commit.

### OpenWrt version

r27290-64dae1052

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Self-built image

### Steps to reproduce

Build current main branch with LuCI and install; After the login page is shown, login and check the Load Average

### Actual behaviour

Login would be laggy and the Load Average would be high, I saw some value like 3.45,1.05,0.37### 

### Expected behaviour

The login should be quick and the Load Average should be something like 0.67, 0.17, 0.06

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_mt7622=y
CONFIG_TARGET_mediatek_mt7622_DEVICE_linksys_e8450-ubi=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_AUTOREMOVE=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
# CONFIG_GCC_USE_VERSION_13 is not set
CONFIG_GCC_USE_VERSION_14=y
CONFIG_GCC_VERSION="14.2.0"
CONFIG_GCC_VERSION_14=y
# CONFIG_GDB is not set
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IB=y
CONFIG_IMAGEOPT=y
CONFIG_JSON_CYCLONEDX_SBOM=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
# CONFIG_KERNEL_KALLSYMS is not set
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_bridger=y
[... 905 lines cut ...]
# CONFIG_PACKAGE_rtl8821ce-firmware is not set
# CONFIG_PACKAGE_rtl8822be-firmware is not set
# CONFIG_PACKAGE_rtl8822ce-firmware is not set
# CONFIG_PACKAGE_rtl8851be-firmware is not set
# CONFIG_PACKAGE_rtl8852ae-firmware is not set
# CONFIG_PACKAGE_rtl8852be-firmware is not set
# CONFIG_PACKAGE_rtl8852ce-firmware is not set
# CONFIG_PACKAGE_snapshot-tool is not set
# CONFIG_PACKAGE_tc-tiny is not set
# CONFIG_PACKAGE_umbim is not set
# CONFIG_PACKAGE_uqmi is not set
# CONFIG_PACKAGE_wil6210-firmware is not set
# CONFIG_PACKAGE_wl12xx-firmware is not set
# CONFIG_PACKAGE_wl18xx-firmware is not set
# CONFIG_PACKAGE_wwan is not set
# CONFIG_PACKAGE_xdp-filter is not set
# CONFIG_PACKAGE_xdp-loader is not set
# CONFIG_PACKAGE_xdpdump is not set
# CONFIG_PACKAGE_xtables-legacy is not set
# CONFIG_PACKAGE_xtables-nft is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pesa1234, 2024-09-02:
Hi, seems same like https://github.com/openwrt/openwrt/issues/16281

Thanks

mrkiko, 2024-09-03:
@dangowrt maybe you are interested in knowing this happens on mt7622 apparently

dangowrt, 2024-09-03:
@mrkiko We can revert, of course, but mostly we need the attention of @onlyfly34 who is the author of that patch series.

rany2, 2024-09-03:
I personally don't have this issue on any of the devices that use that driver (i.e., mt7621/Asus RT-AX53U and mt7622/Redmi AX6s). The AX6S specifically is very similar to the Linksys E8450 spec wise, so it might be useful to try to pinpoint what might cause this issue.

OpenWRT-fanboy, 2024-09-03:
> I personally don't have this issue on any of the devices that use that driver (i.e., mt7621/Asus RT-AX53U and mt7622/Redmi AX6s). The AX6S specifically is very similar to the Linksys E8450 spec wise, so it might be useful to try to pinpoint what might cause this issue as I don't think everyone is going to experience it.
> 
> Edit: Worth noting that I only use OpenWRT as a dumb AP.

My Redmi AX6000 works fine as an AP, but breaks if you hang some devices off its ports. Can you try testing that?

See https://github.com/openwrt/openwrt/issues/16274#issuecomment-2323542947

edrikk, 2024-09-07:
As a data point, I updated 3 Redmi AX6S access points to build b870c16534.  First two came up as normal, however the 3rd experienced the heavy load, with Luci barely loading login page etc.

I didn’t know about this issue and did a power unplug/replug.   The device came back up ok.

Havent rebooted since, but having come across this ticket, wanted to note this.

CharlesMengCA, 2024-09-08:
I did a factory reset and power unplug/replug. The problem still exists if including the two commits.

I used an external logger to time the reboot. Here are the result.

Reverted two commits. `36` seconds
```
24-09-07 21:47:27 daemon.info procd: - shutdown -
...
24-09-07 21:48:03 daemon.info procd: - init complete -
```

With two commits.  `51` seconds.
```
24-09-07 17:32:26 daemon.info procd: - shutdown -
...
24-09-07 17:33:17 daemon.info procd: - init complete -
```

hnyman, 2024-09-11:
In the #16281 discussion, it has been noticed that the issue may happen when there are 3-4 LAN cables connected and some of them have different link speeds (1 Gbit / 100 Mbit /10 Mbit).  Other possible speculation is, if netifd changes from https://github.com/openwrt/netifd/commit/68c8a4f94cd3c play a role.

The "high load" might be a visible symptom from the several LAN interfaces repeatedly going down/up, as discussed in https://github.com/openwrt/openwrt/issues/16281#issuecomment-2341744884

CharlesMengCA, 2024-09-11:
When I noticed the high load, My E8450 had two Ethernet cables connected (one to main router, one to computer).

My another AP router, ASUS TUF AX4200, experienced lost responses randomly on LAN connection, which has three 1GbE and two 100M LAN cables connected.

CharlesMengCA, 2024-09-12:
For my E8450, this commit https://github.com/openwrt/openwrt/commit/c4a926516038250105aef1cc5c6ec9fd5c74e300 fixed the problem.

closed 2024-09-12 by CharlesMengCA