openwrt/openwrt #16319: I2C bus doesn't work in GL-AR750

bwarden, 2024-09-03:
Booting the GL-AR750, I can see that the gpio-based I2C controller is set up with the expected pins 16 and 17:
```
[    0.147948] gpio-17 (sda): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    0.157395] gpio-16 (scl): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    0.166869] i2c-gpio i2c: using lines 17 (SDA) and 16 (SCL)
[    1.644658] i2c_dev: i2c /dev entries driver
```
Attempting to scan the bus, with nothing connected, shows unexpected results matching all possible devices:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
```
I verified that SDA isn't shorted to ground, which would be one hardware cause for this.

Connecting a device, with verified 10k pull-ups on SDA and SCL doesn't change the scan, and communication with the device on its expected address fails.

I suspect the issue is with the device tree configuration:
https://github.com/openwrt/openwrt/blob/2bec6f48e6f8438f81bf26979d9ea296331d265d/target/linux/ath79/dts/qca9531_glinet_gl-ar750.dts#L66-L67
Instead of `GPIO_ACTIVE_LOW`, I'd expect SDA and SCL to be `(GPIO_ACTIVE_HIGH | GPIO_OPEN_DRAIN)`, as I2C lines are expected to be open drain.

I'll try to create a custom build to verify this, but I'm hoping there's an expert who knows at first glance whether I'm right.

bwarden, 2024-09-05:
This change does resolve the phantom detection:
[backport-set-ar750-i2c-gpio-open-drain.patch](https://github.com/user-attachments/files/16882484/backport-set-ar750-i2c-gpio-open-drain.patch)
```diff
diff --git a/target/linux/ath79/dts/qca9531_glinet_gl-ar750.dts b/target/linux/ath79/dts/qca9531_glinet_gl-ar750.dts
index 4d809b922a..45de917ca6 100644
--- a/target/linux/ath79/dts/qca9531_glinet_gl-ar750.dts
+++ b/target/linux/ath79/dts/qca9531_glinet_gl-ar750.dts
@@ -63,8 +63,8 @@
        i2c {
                compatible = "i2c-gpio";

-               sda-gpios = <&gpio 17 GPIO_ACTIVE_LOW>;
-               scl-gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
+               sda-gpios = <&gpio 17 (GPIO_ACTIVE_HIGH | GPIO_OPEN_DRAIN)>;
+               scl-gpios = <&gpio 16 (GPIO_ACTIVE_HIGH | GPIO_OPEN_DRAIN)>;
        };
 };
```
However, a connected device is not reported, either:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

bwarden, 2024-09-10:
With SDA and SCL configured for `GPIO_ACTIVE_HIGH | GPIO_OPEN_DRAIN` as above, I do see the lines briefly dropping voltage with a good multimeter during the i2cdetect scan. I tried connecting two DS3231 modules in turn, though, and neither was actually detected, nor did the rtc-ds1307 driver communicate with them after `echo ds3231 0x68 > /sys/bus/i2c/devices/i2c-0/new_device`.

ryanrsrs, 2024-09-15:
I'll take a look at this.

ryanrsrs, 2024-09-17:
Answer: SDA is connected to GPIO1, not GPIO17.

https://github.com/openwrt/openwrt/pull/16406

bwarden, 2024-09-17:
YES!

```
# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
# echo 'ds3231 0x68' > /sys/class/i2c-dev/i2c-0/device/new_device
# dmesg | tail
[  914.544092] rtc-ds1307 0-0068: registered as rtc0
[  914.550173] rtc-ds1307 0-0068: setting system clock to 2024-09-17T19:31:14 UTC (1726601474)
[  914.560038] i2c i2c-0: new_device: Instantiated device ds3231 at 0x68
# hwclock -u; date
Tue Sep 17 12:31:53 2024  0.000000 seconds
Tue Sep 17 12:31:53 PDT 2024
```

namiltd, 2024-09-22:
Fixed: https://github.com/ryanrsrs/openwrt/commit/8e75eed6173232b57605a18ed3dbbd3f17aed293 Topic to close.