openwrt/openwrt #16320: xdp-tools-1.4.2 bump causes build problems on qoriq

stintel, 2024-09-03:
### Describe the bug

The xdp-tools-1.4.2 bump causes build problems on qoriq:
[compile.txt](https://github.com/user-attachments/files/16854308/compile.txt)

Reverting f3080677f506d2670c1911a91c19eec4ace5d085 solves the problem.

@rockdrilla 

### OpenWrt version

r27297-2bec6f48e6

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qoriq/generic

### Device

WatchGuard Firebox M300

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_qoriq=y
CONFIG_TARGET_qoriq_generic=y
CONFIG_TARGET_qoriq_generic_DEVICE_watchguard_firebox-m300=y
CONFIG_DEVEL=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH=""
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILD_LOG=y
CONFIG_BUSYBOX_DEFAULT_PIE=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_DEBUG=y
CONFIG_DROPBEAR_MODERN_ONLY=y
CONFIG_DROPBEAR_ZLIB=y
CONFIG_DWARVES=y
# CONFIG_FEED_ecnmarker is not set
# CONFIG_FEED_floattest is not set
# CONFIG_FEED_matter is not set
# CONFIG_FEED_usteer_web_ucode is not set
# CONFIG_FEED_video is not set
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IPERF_ENABLE_MULTICAST=y
# CONFIG_JSON_OVERVIEW_IMAGE_INFO is not set
CONFIG_KEEPALIVED_BFD=y
CONFIG_KEEPALIVED_CHECKSUM_COMPAT=y
CONFIG_KEEPALIVED_FWMARK=y
CONFIG_KEEPALIVED_LINKBEAT=y
CONFIG_KEEPALIVED_LVS=y
CONFIG_KEEPALIVED_LVS_64BIT_STATS=y
CONFIG_KEEPALIVED_LVS_SYNCD=y
CONFIG_KEEPALIVED_NFTABLES=y
CONFIG_KEEPALIVED_ROUTES=y
[... 414 lines cut ...]
CONFIG_TARGET_ROOTFS_PERSIST_VAR=y
CONFIG_TARGET_ROOTFS_SQUASHFS=y
CONFIG_TARGET_SQUASHFS_BLOCK_SIZE=256
CONFIG_TESTING_KERNEL=y
CONFIG_USE_LLVM_HOST=y
CONFIG_ZLIB_OPTIMIZE_SPEED=y
# CONFIG_KEEPALIVED_IPTABLES is not set
# CONFIG_LLDPD_WITH_CDP is not set
# CONFIG_LLDPD_WITH_EDP is not set
# CONFIG_LLDPD_WITH_FDP is not set
# CONFIG_LLDPD_WITH_SONMP is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_getrandom=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
# CONFIG_PACKAGE_libip4tc is not set
# CONFIG_PACKAGE_libip6tc is not set
# CONFIG_PACKAGE_libxtables is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stintel, 2024-10-13:
Looks like I attached the wrong compile.txt when reporting this issue. That is now fixed.

I have also tried bumping to 1.4.3, which exhibits the same problem. I have also tried using the OpenWrt LLVM toolchain rather than the one installed on the host, by enabling CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y, but still getting the same error.

rockdrilla, 2024-10-14:
@stintel I've seen your [commit](https://github.com/openwrt/openwrt/commit/2d350fded62d559d5a037d0febc9c806ae4f1d6e). Does it build successfully (on qoriq)?

stintel, 2024-10-14:
It does not. But I found some other issues, e.g. It picks up my host's bpftool, v7.5.0 while we have v7.4.0 in OpenWrt. I tried adding PKG_BUILD_DEPENDS:=bpftool/host but it still picked up the one in /usr rather than the one in staging_dir. I'll do some more digging later.

closed 2024-10-28 by openwrt-bot

stintel, 2024-10-28:
For reference, #16787 did not fix this, but I suspect the combination of #16787 and 5fdd18243eecf405737939bde553e14b6eec0043 does.

PR #16787 (2024-10-28): xdp-tools: fix compilation when bpftool is installed
Currently, `xdp-tools` doesn't compile on build systems where `bpftool` is installed because additional tools and BPF programs will be compiled then, which results in build errors.

This commit disables the compilation of those `bpftool`-dependent tools.
Additionally, it drops a patch to re-add the `vlan_hdr` struct because it is required for building `xdp-filter`.

This should fix #16320.

It's currently not possible to always build the `bpftool`-dependent tools because the latest release version 7.4 of `bpftool` cannot open BPF objects tha [... cut]