openwrt/openwrt #16324: OpenWRT snapshot doesn't boot on Unifi Security gateway

trenchcoatjedi, 2024-09-04:
### Describe the bug

Current snapshot of OpenWRT won't boot on a Unifi Security Gateway. Tried with version 23.05.04 as well with identical results. USB drive setup exactly as described here: https://openwrt.org/toh/ubiquiti/unifi_security_gateway_3p

Here is the console output of the boot log.

```
Looking for valid bootloader image....
Jumping to start of image at address 0xbfc80000

U-Boot 1.1.1 (UBNT Build ID: 4674499-gfa58f5d) (Build time: Jun  9 2014 - 14:38)

BIST check passed.
UBNT_E120 r1:0, r2:16, f:8/135, serial #: 802aa88e5c8f
MPR 13-02044-16
Core clock: 500 MHz, DDR clock: 266 MHz (532 Mhz data rate)
DRAM:  512 MB
Clearing DRAM....... done
Flash:  8 MB
Net:   octeth0, octeth1, octeth2
                                                                                
USB:   (port 0) scanning bus for devices... 1 USB Devices found                 
       scanning bus for storage devices...                                      
  Device 0: Vendor:          Prod.: USB DISK 2.0     Rev: PMAP                  
            Type: Removable Hard Disk                                           
            Capacity: 3824.0 MB = 3.7 GB (7831552 x 512)                        
 0                                                                              
reading vmlinux.64                                                              
.....................................................                           
                                                                                
10756736 bytes read                                                             
argv[2]: coremask=0x3                                                           
argv[3]: root=/dev/sda2                                                         
argv[4]: rootdelay=15                                                           
argv[5]: rw                                                                     
argv[6]: rootsqimg=squashfs.img                                                 
argv[7]: rootsqwdir=w                                                           
argv[8]: mtdparts=phys_mapped_flash:512k(boot0),512k(boot1),64k@1024k(eeprom)   
ELF file is 64 bit                                                              
Allocating memory for ELF segment: addr: 0xffffffff81100000 (adjusted to: 0x1104
Allocated memory for ELF segment: addr: 0xffffffff81100000, size 0x1b930f4      
Processing PHDR 0                                                               
  Loading a40d4d bytes at ffffffff81100000                                      
  Clearing 11523a7 bytes at ffffffff81b40d4d                                    
## Loading Linux kernel with entry point: 0xffffffff818dd430 ...                
Bootloader: Done loading app on coremask: 0x3                                   
[    0.000000] Linux version 5.15.162 (builder@buildhost) (mips64-openwrt-linux4
[    0.000000] CVMSEG size: 0 cache lines (0 bytes)                             
[    0.000000] printk: bootconsole [early0] enabled                             
[    0.000000] CPU0 revision is: 000d0601 (Cavium Octeon+)                      
[    0.000000] Kernel sections are not in the memory maps                       
[    0.000000] Wasting 278528 bytes for tracking 4352 unused pages              
[    0.000000] Initrd not found or empty - disabling initrd                     
[    0.000000] Using appended Device Tree.                                      
[    0.000000] software IO TLB: mapped [mem 0x0000000002d09000-0x0000000006d090)
[    0.000000] Primary instruction cache 32kB, virtually tagged, 4 way, 64 sets.
[    0.000000] Primary data cache 16kB, 64-way, 2 sets, linesize 128 bytes.     
[    0.000000] Zone ranges:                                                     
[    0.000000]   DMA32    [mem 0x0000000001100000-0x00000000efffffff]           
[    0.000000]   Normal   [mem 0x00000000f0000000-0x000000041fbfffff]           
[    0.000000] Movable zone start for each node                                 
[    0.000000] Early memory node ranges                                         
[    0.000000]   node   0: [mem 0x0000000001100000-0x0000000002c92fff]          
[    0.000000]   node   0: [mem 0x0000000002d00000-0x00000000080fffff]          
[    0.000000]   node   0: [mem 0x0000000008200000-0x000000000fdfffff]          
[    0.000000]   node   0: [mem 0x0000000410000000-0x000000041fbfffff]          
[    0.000000] Initmem setup node 0 [mem 0x0000000001100000-0x000000041fbfffff] 
[    0.000000] On node 0, zone DMA32: 4352 pages in unavailable ranges          
[    0.000000] On node 0, zone DMA32: 109 pages in unavailable ranges           
[    0.000000] On node 0, zone DMA32: 256 pages in unavailable ranges           
[... 209 lines cut ...]
octeon/generic

### Device

Ubiquiti UniFi Security Gateway 3P (USG-3P)

### Image kind

Official downloaded image

### Steps to reproduce

Download image, follow directions on Wiki.

### Actual behaviour

rootfs doesn't load; see log

### Expected behaviour

system boots

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

trenchcoatjedi, 2024-09-04:
images downloaded here: https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=octeon%2Fgeneric&id=ubnt_unifi-usg

mrkiko, 2024-09-04:
Did you try with multiple USB sticks?

trenchcoatjedi, 2024-09-05:
USB compatibility issue unrelated to OpenWRT

closed 2024-09-05 by trenchcoatjedi