openwrt/openwrt #16333: ipq4019  squashfs error

zhangsd20, 2024-09-05:
### Describe the bug

My device works fine on openwrt19.07. But when I upgraded to 22.03/23.05 something abnormal happened.

Devices sometimes have problems after multiple restarts (19.07 Moyo has any problems, so there is no hardware problem)

When the device fails to access the web, the following information is displayed through the debugging serial port：

.....
SQUASHFS error: Unable to read fragment cache entry [253432]
SQUASHFS error: Unable to read page, block 253532, size 16968
SQUASHFS error: Unable to read fragment cache entry [253432]
SQUASHFS error: Unable to read page, block 253432, size 16968
.....

I have seen many people encounter this problem, but the methods they mentioned don't seem to be the real problem is there a way to solve it completely?

https://github.com/openwrt/openwrt/issues/9085
https://forum.openwrt.org/t/patch-squashfs-data-probably-corrupt/70480
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=540377010532d9840ebe0778c2af991bd4b67052

### OpenWrt version

r20196-1ec274a204

### OpenWrt release

22.03-SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Qxwlan E2600AC c1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-09-05:
Upgrade to 23.05.4
Is squasshfs on fixed flash blocksor ubi?
Narural flash wear, one bit flipped?

zhangsd20, 2024-09-06:
I tested a lot of versions 21.02.0 21.02.5 21.02.7 22.03.0  22.03.5 23.05.1 and 23.05.4,
they all have this problem. 

I am using 32M nor flash，this is my partitions：

[    0.979939] 0x000000000000-0x000000040000 : "0:SBL1"
[    0.985475] 0x000000040000-0x000000060000 : "0:MIBIB"
[    0.990480] 0x000000060000-0x0000000c0000 : "0:QSEE"
[    0.995501] 0x0000000c0000-0x0000000d0000 : "0:CDT"
[    1.000459] 0x0000000d0000-0x0000000e0000 : "0:DDRPARAMS"
[    1.005098] 0x0000000e0000-0x0000000f0000 : "0:APPSBLENV"
[    1.010636] 0x0000000f0000-0x000000170000 : "0:APPSBL"
[    1.016202] 0x000000170000-0x000000180000 : "0:ART"
[    1.025413] 0x000000180000-0x000002000000 : "firmware"
[    1.032818] 2 fit-fw partitions found on MTD device firmware
[    1.036884] Creating 2 MTD partitions on "firmware":
[    1.042735] 0x000000000000-0x0000004a0000 : "kernel"
[    1.048320] 0x00000049ad18-0x000001e80000 : "rootfs"
[    1.058188] mtd: setting mtd10 (rootfs) as root device
[    1.066891] 1 squashfs-split partitions found on MTD device rootfs
[    1.071724] 0x000000840000-0x000001e80000 : "rootfs_data"

Worked fine on 19.07 and 18.06, but had problems with kernel >= 5.x

mips171, 2024-10-06:
Seeing similar thing on snapshot at the moment